NorthernMan54/rtl_433_ESP #3: Typedef for Callback not compatible

AusGunno, 2021-03-19:
Firstly - Awesome project. Very interested to be part of it. I have several OOK and FSK devices I would like to integrate.

Prior to cloning this repo I just downloaded it and set up a new PlatformIO project. I am using the ELECHOUSE CPP and H files from LSatan.

When I complied it the following error occurs:
src\main.cpp: In function 'void setup()':
src\main.cpp:52:66: error: invalid conversion from 'void (*)(char*, char*, unsigned int)' to 'rtl_433_ESPCallBack {aka void (*)(char*)}' [-fpermissive]     
   rf.setCallback(rtl_433_Callback, messageBuffer, JSON_MSG_BUFFER);

Looking into the typedef in the rtl_433_ESP.H
/**
  * (char *name, char *message, unsigned int modulation)
   * name - name of device as defined in device signal decoder
   * message - JSON formated message from device
   * modulation - Type of signal modulation ( r_device.h )
   */

typedef void (*rtl_433_ESPCallBack)(char *message);

This will not work when called 
  rf.setCallback(rtl_433_Callback, messageBuffer, JSON_MSG_BUFFER);

in this case we are passing a function to the setCAllback but it is expecting the pointer to the message which you then want to store in the config structure.

Should it be more like the function for PulseTrainCallBack??? Or am I doing something wrong??

typedef std::function<void(const uint16_t *pulses, size_t length)>
    PulseTrainCallBack;

Cheers
Gunno

NorthernMan54, 2021-03-19:
Did you just opening the example directory in Platformio and building ? I updated it yesterday to fix a couple of bugs

NorthernMan54, 2021-03-19:
Also I went back and looked at the comments in rtl_433_ESP.H and realize I had never updated them when I had adjusted the signature late in the build cycle.

AusGunno, 2021-03-19:
OK - tried that but it does not build - cannot find the rtl_433_ESP package
Library Manager: Installing rtl_433_ESP
Warning! Could not find the package with 'rtl_433_ESP' requirements for your system 'windows_amd64'
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoLog> 1.0.3+sha.d13cd80
|-- <ArduinoJson> 5.13.4
Error: Nothing to build. Please put your source code files to 'C:\Users\terry\Documents\GitHub\rtl_433_ESP\example\src' folder

 I dont normally play with arduino projects in platformio. So I had copied the ino as main.cpp and moved it to the src folder with your cpp and header files as well as the ELECHOUSE files.

So do you import it as an arduino project??

AusGunno, 2021-03-20:
OK got the updated release - my bad.
Compiles now.

closed 2021-03-20 by AusGunno