NorthernMan54/rtl_433_ESP #8: FYI: Skylink HA-434TL - false positives

DigiH, 2021-04-12:
Hi @NorthernMan54,

just as an update. During the discussion with my pull request yesterday you mentioned that the Skylink HA-434TL decoder might produce too many false positives, as I kept seeing

> {"model":"Skylink HA-434TL motion sensor","motion":"false","id":"00000","raw":"00000","protocol":"Skylink HA-434TL motion sensor","rssi":-57,"duration":99281}

Running my OMG rtl_433_ESP gatway for several hours today, I didn't see one single such false positive message. This made me think what was different to my testing yesterday.

- I had the gateway connected to the PC for serial monitoring. Ok didn't do that again just now, but tried an OTA update with the following debugging build flags on

> '-DRTL_DEBUG=4'           ; rtl_433 verbose mode - ; Enable RTL_433 Verbose option ( 0=normal, 1=verbose, 2=verbose decoders, 3=debug decoders, 4=trace decoding. )
>   '-DLOG_LEVEL=LOG_LEVEL_TRACE'
>   '-DDEMOD_DEBUG=true'
>   '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages

and low and behold, the above phantom Skylink HA-434TL mqtt messages started appearing again in regular intervals.

Maybe you can reproduce this on your side, as

> "motion":"false","id":"00000","raw":"00000",

is always identical in these messages and possibly might not conflict with any of your proper Skylink HA-434TL messages.

Commenting out 

>   ;'-DLOG_LEVEL=LOG_LEVEL_TRACE'
>   ;'-DDEMOD_DEBUG=true'
>   ;'-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages

which should be default for most users, bringe everything back to normal for me, without any false positive Skylink HA-434TL messages at all.

Thanks

NorthernMan54, 2021-04-12:
I looked at my code for the device decoder, and it does not have a data quality or checksum.  So am thinking to just reject ID '00000' going forward.

NorthernMan54, 2023-04-03:
Current release reduces the number of false positives

closed 2023-04-03 by NorthernMan54