NorthernMan54/rtl_433_ESP #10: Support for configuration and use

joelsernamoreno, 2021-04-29:
Hi!

LSatan and I are trying to test your rtl_433 example, but we can't get it to work

I think the documentation is a bit sparse and causes confusion

Can you tell me the full process to make this work?

Thanks!

NorthernMan54, 2021-04-29:
I agree with you that the documentation is currently lacking and I will be improving over time.  I mostly just threw a README up and had concentrated on the integration with OpenMQTTGateway for the first release.

You should be able to clone the repo, and open a new empty platformio window and "add folder to workspace" the rtl_433_ESP/example/sample/ folder.  With Platformio you should then be able to build and deploy the sample to your ESP32.  Platformio will then pull all the required libraries in ( including an additional copy of this library ).  I had included my platformio.ini with all the settings needed to build and deploy.

With the current sample it uses the sample wiring from LSATAN, but with the next release with support for FSK Modulation I will be changing the wiring as usage of GPIO2 is currently problematic.

I was also thinking to add a example of this library to LSatan/SmartRC-CC1101-Driver-Lib once the FSK modulation release is complete.  Expect that in the near future.

Can you share more details about your issue ?

joelsernamoreno, 2021-04-29:
Hello!

I have downloaded your repository and modified sample (ESP_rtl_433.ino) with the following changes:
[image]

The changes are as follows:

* Badurate change: Serial.begin(115200);
* Pins change: ELECHOUSE_cc1101.setSpiPin(sck, miso, mosi, csn1); (This is a reference for my connections)
* RX pin change: rf.enableReceiver(4);

I have flashed the example to my device with pio run -t upload

I have opened the Arduino serial monitor and I get the following, it looks like everything is set up correctly:
[image]

What do I have to do now?

Thank you

NorthernMan54, 2021-04-29:
Nothing, just listen for ASK/OOK signals

This from my device (   '-DRTL_DEBUG=4' is very verbose )

```
Registering protocol [96] "WT0124 Pool Thermometer"
Registering protocol [98] "X10 Security"
N: ****** setup complete ******
 
rtl_433_ESP(7): demod(5) - Acurite 609TXC Temperature and Humidity Sensor
rtl_433_ESP(7): demod(6) - Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas
acurite_txr_decode: bitbuffer:: Number of rows: 25 
[00] { 2} 40        : 01
[01] { 1} 80        : 1
[02] { 1} 80        : 1
[03] { 1} 80        : 1
[04] { 1} 80        : 1
[05] { 1} 80        : 1
[06] { 1} 80        : 1
[07] { 1} 80        : 1
[08] { 1} 80        : 1
[09] { 1} 80        : 1
[10] { 1} 80        : 1
[11] { 1} 80        : 1
[12] { 1} 80        : 1
[13] { 1} 80        : 1
[14] { 1} 80        : 1
[15] { 1} 80        : 1
[16] { 1} 80        : 1
[17] { 1} 80        : 1
[18] { 1} 80        : 1
[19] { 1} 80        : 1
[20] { 1} 80        : 1
[21] { 1} 80        : 1
[22] { 1} 80        : 1
[23] { 1} 80        : 1
[24] { 1} 80        : 1
... Maximum number of rows reached. Message is likely truncated.
acurite_txr_decode: row 0 bits 2, bytes 1 
acurite_txr_decode: row 1 bits 1, bytes 1 
acurite_txr_decode: row 2 bits 1, bytes 1 
acurite_txr_decode: row 3 bits 1, bytes 1 
acurite_txr_decode: row 4 bits 1, bytes 1 
acurite_txr_decode: row 5 bits 1, bytes 1 
acurite_txr_decode: row 6 bits 1, bytes 1 
acurite_txr_decode: row 7 bits 1, bytes 1 
acurite_txr_decode: row 8 bits 1, bytes 1 
acurite_txr_decode: row 9 bits 1, bytes 1 
acurite_txr_decode: row 10 bits 1, bytes 1 
acurite_txr_decode: row 11 bits 1, bytes 1 
acurite_txr_decode: row 12 bits 1, bytes 1 
acurite_txr_decode: row 13 bits 1, bytes 1 
acurite_txr_decode: row 14 bits 1, bytes 1 
acurite_txr_decode: row 15 bits 1, bytes 1 
acurite_txr_decode: row 16 bits 1, bytes 1 
acurite_txr_decode: row 17 bits 1, bytes 1 
acurite_txr_decode: row 18 bits 1, bytes 1 
acurite_txr_decode: row 19 bits 1, bytes 1 
acurite_txr_decode: row 20 bits 1, bytes 1 
acurite_txr_decode: row 21 bits 1, bytes 1 
acurite_txr_decode: row 22 bits 1, bytes 1 
acurite_txr_decode: row 23 bits 1, bytes 1 
acurite_txr_decode: row 24 bits 1, bytes 1 
acurite_txr_decode: bitbuffer:: Number of rows: 25 
[00] { 1} 80        : 1
[01] { 1} 80        : 1
[02] { 1} 80        : 1
[03] { 1} 80        : 1
[04] { 1} 80        : 1
[05] { 1} 80        : 1
[... 1331 lines cut ...]
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
burnhardbbq_decode: bitbuffer:: Number of rows: 1 
[00] { 1} 80        : 1
```

NorthernMan54, 2021-04-29:
This is when '-DRTL_DEBUG=4' is commented out

I have several devices and they are chatty so I get multiple messages per minute.

```
N: ****** setup ******
N: ****** setup complete ******
 
 
rtl_433_ESP(6): Unparsed Signal length: 125553, Signal RSSI: -80, train: 1, messageCount: 1, pulses: 170
rtl_433_ESP(6): RAW (125553): +1424-152+73-1108+226-498+222-593+196-340+101-477+328-329+377-506+279-79+100-102+103-101+125-100+304-179+98-125+355-102+77-103+76-75+306-230+100-98+74-153+377-73+77-102+76-507+225-535+275-890+203-591+202-262+126-126+76-101+125-75+305-925+203-200+74-128+0-174+78-99+374-97+74-149+100-102+268-98+98-698+228-894+198-547+252-943+245-121+73-98+98-100+0-99+101-99+327-77+51-101+101-101+274-124+123-198+251-538+228-532+199-522+176-537+227-532+200-208+100-127+330-128+148-101+276-199+52-98+153-125+75-101+251-531+199-613+175-902+204-536+225-471+229-584+229-257+228-280+1609-879+98-199+174-125+1705-100+97-99+76-100+126-151+78-98+101-75+76-126+1828-917+123-101+75-102+1904-126+127-152+51-126+99-202+150-303+220-542+204-127+78-101+100-101+329-126+51-101+100-126+77-75+276-97+98-124+100-76+246-128+125-325+205-886+202-538+205-535+175-122+51-128+53-100+329-530+230-896+201-593+203-892+199-600+201-944+0-992+0-176+256-129+76-127+101-101+75-104+0-74+472-148+101-74+274-124+123-123+101-98+100-99+299-98+123-73+99-100+100-51+99-99+296-123+74-100+100-100+272-528+101-100+428-1612+177-558+177-540+207-535+201-519+204-512+228-535+172-529+222-518+229-505+199-123+97-97+77-73+297-99+97-175+377-102+151-100+75-77+300-76+51-75+100-100+103-74+127-101+0-636+101-100+76-153+301-98+102-103+101-126+150-101 
N: Received message : {"model":"unknown","protocol":"signal parsing failed","duration":125553,"signalRssi":-80,"pulses":170,"train":1,"messageCount":1,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-87,"minimumRssi":-82}
 
N: Received message : {"model":"Prologue-TH","subtype":5,"id":97,"channel":2,"battery":"OK","temperature_C":19.2,"humidity":17,"button":0,"protocol":"Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor","rssi":-62,"duration":903231}
```

NorthernMan54, 2021-04-29:
This is not a scenario I took into account when designing the library. And the library is currently not checking for a live cc1101 as well ;-(

Can you just edit the rtl_433_ESP library to try and set the values?

joelsernamoreno, 2021-04-29:
Do you mean changing the pins in the library?

NorthernMan54, 2021-04-29:
Yes change either the rtl_433_ESP or SmartRC-CC1101-Driver-Lib ?

joelsernamoreno, 2021-04-29:
ELECHOUSE_cc1101.setSpiPin (sck, miso, mosi, csn1); change in SmartRC library

Where do you specify the pins in your rtl_433_ESP library?

LSatan, 2021-04-29:
do you use the internal reception functions for reception or an interrupt that records the pulses?

NorthernMan54, 2021-04-29:
I’m watching the RSSI level in the arduino Loop function, then when RSSI is above MINRSSI a interrupt routine records pulse duration of GDO2.  Start and end of signal is determined based on RSSI

My next release with FSK modulation support will utilize a second interupt triggered by carrier sense on GDO0 to determine start and end of signal.

For receiving I’m not using the onboard packet decoders but am leveraging asynchronous serial mode.

joelsernamoreno, 2021-04-29:
@NorthernMan54 I am using the same connections that you indicate in the readme.md

sck = 18
miso = 19
mosi = 23
csn = 5
gdo0= 2
gdo2= 4

With these connections I do not receive any data on the serial monitor either

Thanks

NorthernMan54, 2021-04-30:
Humm...any thoughts on how to troubleshoot this further, is it possibly a wiring issue, signal modulation issue, background noise or ???

1 - Is the wiring to the device working correctly ?
- Do any of the other CC1101 examples ( which I presume use the same GPIO wiring ) work ?

2 - The compiler directives RAW_SIGNAL_DEBUG and DEMOD_DEBUG will log more details about any received signal

3 - Does your ESP32 board have an onboard LED on gpio2 ( mine does ), is it off, on or flashing ?   It is configured to turn on and off when a signal is received.

The start and end of signal is determined when a signal RSSI raises above and drops below MSSI.  If the LED is stuck on, you have a lot of background noise and need to set the MINRSSI to a higher value.  It defaults to -82, try 5 higher ( -78 ) to see if it starts and stops receiving a signal

If your LED is on a different GPIO, the compiler directive ONBOARD_LED allows you to change it.

4 - Do you know what the modulation scheme is of your devices ? ASK/OOK or ??? If it isn't ASK/OOK the current code base can't receive the signal ?

joelsernamoreno, 2021-04-30:
@NorthernMan54 Hello!

My device works correctly with other examples from the SmartRC library. The wiring is correct

About the leds, my device has 3 leds, but they are configured on pins io12, io13 and io15. This does not affect the device

About the compiler directives you indicate, I think they are commented. Only '-DRTL_DEBUG=4' is uncommented. I will try to uncomment the others you indicate

Where can I change MINRSSI?

About my remote controls they are all ASK/OOK.

Thanks!

NorthernMan54, 2021-04-30:
If you look at the header file you can see the additional compiler directives and their defaults.

https://github.com/NorthernMan54/rtl_433_ESP/blob/b4a1cc02753e949a9a3ff8ce43f3392b2e7836a0/src/rtl_433_ESP.h#L26

I would recommend enabling one of your LEDs and then seeing what is happening

another thought is to occasionally call this function in the loop, maybe once a minute to trigger display of receiver status and current Rssi level.  Calling it every loop would be too often.

https://github.com/NorthernMan54/rtl_433_ESP/blob/b4a1cc02753e949a9a3ff8ce43f3392b2e7836a0/src/rtl_433_ESP.cpp#L533

NorthernMan54, 2021-04-30:
Sorry skip the getstatus function as it currently calls a message received callback, which isn’t implemented in the sample.

mozgy, 2021-08-24:
Hello, with DoIT's NodeMCU (1st one with less pins) this works fine ->
```
build_flags = 
	'-DRF_EMITTER_GPIO=12' ; GPIO2 orig
	'-DRF_RECEIVER_GPIO=27' ; GPIO4 orig
	'-DLOG_LEVEL=LOG_LEVEL_ERROR'
```

mozgy, 2021-08-24:
> 
> 
> This is when '-DRTL_DEBUG=4' is commented out
>

This does not work for me, if I comment out that line in platformio.ini ESP32 board craps out and does not run code. Barfs something about falling back to internal interpreter.

NorthernMan54, 2021-08-24:
@mozgy Can you share the message, I'm thinking it may be related to resource constraints on your device.  RTL_DEBUG mode is resource intensive.

mozgy, 2021-08-25:
Something strange is going on here -
cleaned env, commented out, recompiled and when trying to upload it barfed on MD5 mismatch,
(clicked to fast on terminal win, so no log), changed back to define, compiled and uploaded fine.
Commented out again, clean, build, upload, now it works ..
```
RAM:   [=         ]  11.3% (used 37096 bytes from 327680 bytes)
Flash: [===       ]  27.6% (used 361858 bytes from 1310720 bytes)
```
Will test more later ..

skiphansen, 2021-11-20:
This is a great idea, but I just can't get it to work.   

I'm pretty sure the basic connections are correct.  

I've connected my cc1101 board to RF test equipment and verified that it basically works using TI's Smartrf studio and a CC-programmer.  It's definitely a 433 Mhz board and not a 866 Mhz board (power output on 425 Mhz is .66 milliwatt, but only .022 milliwatt on 853Mhz).  The measured transmit frequency on 433.92 Mhz is about 20Khz high.  I've tried recompiling the code for 433.90 and 433.94 but that didn't help any.

I have a Acurite 00592TXR temperature sensor outside of the house which is picked up fine by the Acurite base station in the next room.  

Ages ago I played with an actual RTL dongle and using the same external antenna I was able to decode several devices from my neighbors as well as my own.

 I'm building from the master branch of the repo, hopefully that's correct.

Here's my [log](https://github.com/NorthernMan54/rtl_433_ESP/files/7573741/minicom.txt).

Can you give me some guidance on what to try? 

Thanks!
Skip

NorthernMan54, 2021-11-20:
One of my findings with the board is receiver sensitivity, I find that the range that it receives signals is about 50% of a usb-sdr or the accurate receiver.

skiphansen, 2021-11-20:
What bandwidth do you set the receiver to?  Do you think off frequency is the issue?  20Khz doesn't seem all that bad if the receiver is wide band.

I'll try bring in my temperature sensor and see if i works in the same room.

skiphansen, 2021-11-20:
Well that was the problem.  With the sensor in the same room it works fine!  Kind of disappointing, but at least it works!
````{"model":"Acurite-Tower",
"id":10200,
"channel":"A",
"battery_ok":1,
"temperature_C":18.0,
"humidity":83,
"mic":"CHECKSUM",
"protocol":"Acurite 592TXR Temp/Humidity,5n1 Weather Station,6045 Lightning,3N1,Atlas",
"rssi":-58,
"duration":267224
}
````
Thanks for the fast response!
Skip

closed 2023-04-03 by NorthernMan54