NorthernMan54/rtl_433_ESP #11: Support for RX127X Transceiver Chipset

NorthernMan54, 2021-05-04:
I finally received my parts today and am ready to start adding support for this chipset

For parts I have received the 433Mhz and 868Mhz modules https://heltec.org/project/wifi-lora-32/

[image]
[image]

Stay tuned over the next few weeks for this.

FYI @1technophile

1technophile, 2021-05-04:
really nice, @sensorsiot feel free to subscribe if you want to follow

SensorsIot, 2021-05-04:
Done. I am interested in the results...

NorthernMan54, 2021-05-09:
Research notes for the build

1 - Searching thru GitHub found a few libraries and packages supporting the chipset but on first pass did not find one that used ask/ook/fsk mode everything appeared to be geared towards LORA.  May need to craft my own library from scratch unfortunately, but could use the cc1101 library as a base.

2 - Need to use continuous mode ( this is similar to asynchronis mode on the cc1101 )

```
4.2.9.2. Data Operation Modes
The SX1276/77/78/79 has two different data operation modes selectable by the user:
Continuous mode: each bit transmitted or received is accessed in real time at the DIO2/DATA pin. This mode may be used if adequate external signal processing is available.
```

3 - Signal RSSI is available here

```
RegRssiValu e(0x1B) | 7-0 | Rssi | r | n/a | Current RSSI value (dBm)
```

4 - Digital IO PIN Mapping

```
Table 29 DIO Mapping, Continuous Mode ( Page 69 )
```

cpainchaud, 2021-06-30:
If I am not mistaken this model has DIO2 connected to an OUTPUT only pin of the ESP32 which makes it unsuitable for our use

NorthernMan54, 2021-06-30:
Good question

Looking at this diagram - https://resource.heltec.cn/download/WiFi_LoRa_32/V2 it appears that the RX127X DIO pins are wired like this

DIO0 - GPIO26
DIO1 - GPIO35 - This is input only
DIO2 - GPIO34 - This is input only

I snagged the ESP32 GPIO details from here - https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

What materials are you looking at ?

PS Looking at my personal calendar, this will be on hold to the fall, the need to carve out time to write the library put a damper on quickly checking out the feasibility with a POC code.  Need to enjoy the Canadian Summer

cpainchaud, 2021-06-30:
My bad, it's an Input only! I remember now I had to discard this model because I also needed to output signal in my project!

nikil511, 2021-07-14:
We (in my company) have created an open source IoT node that listens to weather station RF packets and forwards over WiFi/GSM. We use RFM95 lora modules on ESP32 to sniff the packets, so @NorthernMan54 you are on the right track :-) 

https://github.com/exmgr/exm-iot-node-firmware/blob/main/src/fo_sniffer.cpp
https://github.com/jgromes/RadioLib

We are very much interested in to extending our firmware to support other weather stations too, via rtl_433_ESP, so will keep an eye on this and try to help if possible.

garudaonekh, 2021-12-31:
@NorthernMan54 How is this chipset integration? I have similar chipset from Heltect(without LCD) which I also want to try.

NorthernMan54, 2021-12-31:
It is still sitting on the shelf, as I did a quick scan for a library that supported a raw signal mode that handled ask modulation and couldn’t find one.  Rtl_433_esp needs to input the raw signal, as the signal decoding is done in software and not on the receiver chipset itself.  When I read the chipset data sheet I thought I saw the proper mode listed.

garudaonekh, 2021-12-31:
It is beyond my understanding.

NorthernMan54, 2022-06-13:
Well spent some time this week on this and tested out using https://github.com/jgromes/RadioLib instead of https://github.com/LSatan/SmartRC-CC1101-Driver-Lib to interface with a CC1101 module, and was successful.

To use the RadioLib library the most noticeable changes to rtl_433_ESP where

1 - getRssi is not supported in direct mode in RadioLib, so needed to provide my own function instead

2 - With SmartRC-CC1101-Driver-Lib it uses GDO2 to receive the raw signal, while RadioLib uses GDO0.  Need to adjust this

3 - And for CC1101 deaf workaround, I needed to directly access the module using SPI to Idle and start Receive again.

My test/sample code is here https://github.com/NorthernMan54/rtl_433_ESP/tree/4537d15d968e4ef48043ff0ba066d2ca91e8e4c1 in the RadioLib branch.  Please note that I short circuit the platformio lib function and symlink to the repo, hence the RadioLib and rtf_433_ESP library is commented out.

Now for the next step, try it out with a RX127X module.

NorthernMan54, 2022-10-26:
This has been released to production

closed 2022-10-26 by NorthernMan54