NorthernMan54/rtl_433_ESP #13: Is it possible to add support for Inkbird ITH-20R

gregb79, 2021-05-20:
Hi,

is it possible to add support for an Inkbird ITH-20R  for esp32 with cc1101 i am not sure how to add additional decoders. a link for it working with rlt_433 is , https://github.com/ehagan/rtl_433

hopefully you can help as i want to log this sensor in home assistant,

thank you in advance.

NorthernMan54, 2021-05-20:
A couple of comments

1 - My next release will be based on the FSK branch, so any changes will need to be applied their first 

2 - The library currently supports only OOK/ASK Modulation do you know what modulation scheme it is using?  You can find this in the device decoder file for the device.

3 - If you can submit a pull request against the FSK branch with the device decoder file placed in the contrib directory, I can run the build script that updates the code base for the additional file.  My draft update script is here

https://github.com/NorthernMan54/rtl_433_ESP/blob/fsk/tools/update_rtl_433_devices.sh

If your curious as to what changes are required to add support for additional devices you can look at the update script

4 - As I don't have the physical device I will not be able to perform any testing, so will need your sign off on it working correctly.

5 - What framework are you using with this ?  ie OpenMQTTGateway or ?

gregb79, 2021-05-21:
`

`

gregb79, 2021-05-21:
HI i tried to attach a copy of the decoder file, 

[inkbird_ith20r.c.txt](https://github.com/NorthernMan54/rtl_433_ESP/files/6519988/inkbird_ith20r.c.txt)

Thank you.

NorthernMan54, 2021-05-21:
If you look at the decoder file, it is fsk and not ook/ask.  At the present time only ook/ask is fully supported, and I don’t have fsk working.  And the road map fsk has a fair bit of effort to go, so may not be launched for a few months.

gregb79, 2021-05-22:
HI , 

 

Thank you for the update, I appreciate all you help.

 

I look forward to this feature being added, I am more than happy to help test with the inkbird ITH-20r device, 

 

Kind Regards

Greg.

 

This email has been checked for viruses by AVG.
https://www.avg.com

NorthernMan54, 2023-03-30:
@riegelbrau FSK is still a future feature.

riegelbrau, 2023-03-30:
I read that, but wanted to give some "pressure" 🙈.

seb821, 2023-08-03:
Hello,

I have got a Inkbird IBS-P01R that uses the same protocol as the Inkbird ITH-20R.

Now that FSK has been implemented, I thought I would give it a try. I have Lilygo board 433 MHz that works well in OOK.

After changing OOK_MODULATION to false, here is what I was able to catch.

```
----- SX127x Status -----
RegOpMode: 0x0d
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x07
RegBitrateLsb: 0x40
RegRxBw: 0x12
RegAfcBw: 0x02
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----------- FSK --------------
FDEV_MSB: 0x02
FDEV_LSB: 0x8f
----- SX127x Status -----
Analyzing pulses...
Total count:   48,  width: 20.94 ms             (20942 S)
Pulse width distribution:
 [ 0] count:   19,  width:  120 us [113;128]    ( 120 S)
 [ 1] count:    8,  width:  177 us [161;186]    ( 177 S)
 [ 2] count:    6,  width:  275 us [242;296]    ( 275 S)
 [ 3] count:    9,  width:   54 us [44;67]      (  54 S)
 [ 4] count:    5,  width:  462 us [416;530]    ( 462 S)
 [ 5] count:    1,  width:  724 us [724;724]    ( 724 S)
Gap width distribution:
 [ 0] count:    9,  width:  275 us [234;300]    ( 275 S)
 [ 1] count:   16,  width:  116 us [109;124]    ( 116 S)
 [ 2] count:    3,  width:  169 us [168;172]    ( 169 S)
 [ 3] count:    2,  width:  416 us [413;420]    ( 416 S)
 [ 4] count:    2,  width: 1185 us [1057;1313]  (1185 S)
 [ 5] count:    4,  width:  723 us [702;757]    ( 723 S)
 [ 6] count:    1,  width:  545 us [545;545]    ( 545 S)
 [ 7] count:   10,  width:   55 us [44;66]      (  55 S)
Pulse period distribution:
 [ 0] count:   10,  width:  400 us [355;479]    ( 400 S)
 [ 1] count:   12,  width:  266 us [223;299]    ( 266 S)
 [ 2] count:    8,  width:  587 us [537;703]    ( 587 S)
 [ 3] count:    1,  width: 1788 us [1788;1788]  (1788 S)
 [ 4] count:    5,  width:  863 us [757;1017]   ( 863 S)
 [ 5] count:    1,  width: 1178 us [1178;1178]  (1178 S)
 [ 6] count:    8,  width:  175 us [168;183]    ( 175 S)
 [ 7] count:    2,  width:  117 us [113;121]    ( 117 S)
Pulse timing distribution:
 [ 0] count:   36,  width:  118 us [109;128]    ( 118 S)
 [ 1] count:   11,  width:  175 us [161;186]    ( 175 S)
 [ 2] count:   15,  width:  275 us [234;300]    ( 275 S)
 [ 3] count:   19,  width:   54 us [44;67]      (  54 S)
 [ 4] count:    8,  width:  461 us [413;545]    ( 461 S)
 [ 5] count:    5,  width:  723 us [702;757]    ( 723 S)
 [ 6] count:    2,  width: 1185 us [1057;1313]  (1185 S)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB0110701007600AF0113003601CD02D304A18255+AAB0150701007600AF0113003601CD02D304A18090A080B255+AAB0130701007600AF0113003601CD02D304A19081A255+AAB0110701007600AF0113003601CD02D304A1B255+AAB0110701007600AF0113003601CD02D304A18455+AAB0110701007600AF0113003601CD02D304A1C655+AAB0110701007600AF0113003601CD02D304A19555+AAB0110701007600AF0113003601CD02D304A18655+AAB0120701007600AF0113003601CD02D304A1C0B455+AAB0130701007600AF0113003601CD02D304A1A0B0A455+AAB0120701007600AF0113003601CD02D304A1908555+AAB0110701007600AF0113003601CD02D304A18255+AAB0120701007600AF0113003601CD02D304A180B555+AAB0120701007600AF0113003601CD02D304A1838255+AAB0120701007600AF0113003601CD02D304A180B555+AAB0130701007600AF0113003601CD02D304A183A1D255+AAB0120701007600AF0113003601CD02D304A1838255+AAB0160701007600AF0113003601CD02D304A1B3C083C0909255+AAB0190701007600AF0113003601CD02D304A1B08391C08393B3A38055

rtl_433_ESP(6): Unparsed Signal length: 1457000, Signal RSSI: -40, pulses: 48
rtl_433_ESP(6): RAW (1457000): +113-278(-39)+114-109(-39)+161-117(-39)+282-113(-39)+114-113(-39)+44-234(-39)+172-117(-40)+121-168(-40)+296-241(-45)+55-300(-45)+125-420(-45)+475-1313(-41)+183-717(-39)+121-1057(-41)+420-121(-41)+55-545(-41)+292-114(-42)+47-124(-42)+290-413(-45)+175-124(-45)+125-757(-40)+117-296(-40)+121-113(-40)+55-702(-40)+128-51(-39)+125-300(-39)+124-121(-39)+44-717(-40)+117-58(-40)+249-168(-41)+724-293(-44)+124-44(-44)+124-238(-44)+67-54(-44)+416-121(-102)+117-59(-102)+471-121(-92)+179-114(-92)+179-300(-92)+65-114(-92)+128-55(-94)+186-172(-94)+530-114(-100)+120-56(-100)+182-66(-100)+55-58(-100)+242-55(-100)+118-112(-44)
N: Received message : {"model":"undecoded signal","protocol":"signal parsing failed","duration":1457000,"rssi":-40,"pulses":48}
```
Unfortunately it does not decode properly.

Any idea of parameters I could change or tests I could perform to go forward ?

riegelbrau, 2023-08-03:
Please see my command line from inside my Docker Container running on my Synology NAS:[image]

seb821, 2023-08-05:
> Please see my command line from inside my Docker Container running rtl_433 native on my Synology NAS:!

I am not sure I understand. What is providing the rtl_tcp server ? An esp32 ?

riegelbrau, 2023-08-05:
You asked for parameters. Even I don't know how to modify parameters inside the ESP32, I wanted to show, with which parameters rtl_433 running on Linux decodes the sensor quite well. May be that helps, may be not.

NorthernMan54, 2023-08-07:
@riegelbrau @seb821 The current release includes support for FSK, so this should be able to push forward.  Unfortunately I do not have access to an Inkbird device, so my ability to troubleshoot this further is very limited.

seb821, 2023-08-10:
I was hoping that with raw or unparsed signals something could be done. But I completely understand it is very difficult to troubleshoot remotely.
Thanks for the feedback.

riegelbrau, 2023-10-17:
Today I've got managed to compile the image for the board "ttgo-lora32-v21" with option -DOOK_MODULATION=false to test my FSK modulated devices. For the Inkbird ITH-20R I did not recognize any mqtt messages, while the sensor did successfully send messages to the native rtl_433 instance. 
The Lilygo did successfully send configuration messages for two seperate sensor ids of type "Ford TPMS" and messages from my "Jansite Solar TPMS" that I use to control the carbonization of my home brewed beer during the secondary fermentation .
I only used the console messages and the mqtt messages. I have no clue how to see other logs of the rtl_433 instance on the Lilygo board. If I get some hint i will provide some log output or debug infos.

Regards 
Christoph

NorthernMan54, 2024-06-08:
One thing people have found, is that in FSK mode the frequency setting needs to be more exact.  Can you lookup the exact frequency with your rtl_sdr then try using that frequency ?

riegelbrau, 2024-06-09:
I am running rtl_433 with 433.92 MHz. The MQTT messages show two frequencies:
[image]
One is very near to 433.92 and the other is lower. I tried both with OMG V1.7 without success. Are there any other parameters that should or might be adjusted?

NorthernMan54, 2024-06-10:
There are other FSK related settings that are internal to the code base- https://github.com/NorthernMan54/rtl_433_ESP/blob/276bc2cc6cef73037aaf3325b69be99ccd139742/src/rtl_433_ESP.cpp#L267

If you look at the OOK_Receiver example https://github.com/NorthernMan54/rtl_433_ESP/blob/276bc2cc6cef73037aaf3325b69be99ccd139742/example/OOK_Receiver/OOK_Receiver.ino#L75

I had added code to 'twist the knobs' and see if I could get better results during testing.   I had tuned the values based on the signal from my FineOffSet WH51.  You could try using that against your device to find a better setting.

When I was doing the testing, I let it run for about a day for each `knob`, logged the serial output to a file, then did some statistical analysis to figure out the best setting for each knob.  I let it run for a day, as the sensor only sent a signal every 60 seconds, and wanted to test each setting a few times.

NorthernMan54, 2024-06-12:
The MY_DEVICES directive is in a couple of spots, and you can use it to only include a subset of decoders to use.  When I did my testing I did not use a subset but the full list of decoders.

I then did some statistical analysis on this log line.  Log.notice(F(CR "Finished %s: %s, count: %d" CR), TEST, stepPrint, count);

riegelbrau, 2024-06-19:
I've got managed to compile the OOK_Receiver example only with the compiler settings for setBitrate and -DsetRxBW. 
With -DsetFrequencyDeviation I get compiler errors:
```
Compiling .pio\build\esp32_lilygo\libb82\RadioLib\modules\RFM9x\RFM96.cpp.o
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/../../protocols/PhysicalLayer/PhysicalLayer.h:225:21: note: in expansion of macro 'setFrequencyDeviation'
     virtual int16_t setFrequencyDeviation(float freqDev);
                     ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/CC1101.h:737:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/LLCC68/../SX126x/SX126x.h:774:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/nRF24/nRF24.h:429:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/RF69/RF69.h:773:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/RFM2x/../Si443x/Si443x.h:730:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/RFM9x/../SX127x/SX127x.h:920:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/SX128x/SX128x.h:670:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\esp32_lilygo\src\OOK_Receiver.ino.cpp.o] Error 1
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/../../protocols/PhysicalLayer/PhysicalLayer.h:225:21: note: in expansion of macro 'setFrequencyDeviation'
     virtual int16_t setFrequencyDeviation(float freqDev);
                     ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/CC1101.h:737:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/CC1101.cpp: In member function 'int16_t CC1101::begin(float, float, float, float, int8_t, uint8_t)':
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/CC1101.cpp:68:40: error: expression cannot be used as a function
   state = setFrequencyDeviation(freqDev);
                                        ^
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/CC1101.cpp: At global scope:
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/CC1101/CC1101.cpp:502:17: note: in expansion of macro 'setFrequencyDeviation'
 int16_t CC1101::setFrequencyDeviation(float freqDev) {
                 ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/LLCC68/../SX126x/../../protocols/PhysicalLayer/PhysicalLayer.h:225:21: note: in expansion of macro 'setFrequencyDeviation'
     virtual int16_t setFrequencyDeviation(float freqDev);
                     ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/LLCC68/../SX126x/SX126x.h:774:13: note: in expansion of macro 'setFrequencyDeviation'
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\esp32_lilygo\libb82\RadioLib\modules\CC1101\CC1101.cpp.o] Error 1
*** [.pio\build\esp32_lilygo\libb82\RadioLib\modules\LLCC68\LLCC68.cpp.o] Error 1
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/RF69/../../protocols/PhysicalLayer/PhysicalLayer.h:225:21: note: in expansion of macro 'setFrequencyDeviation'
     virtual int16_t setFrequencyDeviation(float freqDev);
                     ^~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected unqualified-id before numeric constant
.pio/libdeps/esp32_lilygo/RadioLib/src/modules/RF69/RF69.h:773:13: note: in expansion of macro 'setFrequencyDeviation'
[... 11 lines cut ...]
<command-line>: error: expected unqualified-id before numeric constant
[... cut ...]
     int16_t setFrequencyDeviation(float freqDev) override;
             ^~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\esp32_lilygo\libb82\RadioLib\modules\RFM9x\RFM96.cpp.o] Error 1
=================================================================================== [FAILED] Took 28.43 seconds ===================================================================================
Environment    Status    Duration
-------------  --------  ------------
esp32_lilygo   FAILED    00:00:28.428
```
Can you help with this?

riegelbrau, 2024-07-27:
I don't get any steps further. Can anyone help? For details, see above.

riegelbrau, 2024-09-20:
Hello @gregb79 , did you get the ITH-20R sensor running? Do you have any tipps how to get further with this device?

NorthernMan54, 2024-11-30:
I refreshed the device decoders from rtl_433 24.10, maybe this is resolved?

riegelbrau, 2024-12-09:
Now I tried the same tests again. I compiled and uploaded the OOK_Receiver package from GitHub Desktop using the platformio extension of VSCode. I see the serial output, but no success for protocol no 35. The compilation using the knobs mentioned above finishes with the same results. I can compile with -DsetBitrate and -DsetRxBW but not with setFreqDev:
[image]
Furthermore I get readable serial output in the terminal, but when I use PuTTY the output is unreadable. These puTTY settings run well, when I use them with an image of OMG compiled with platformio on gitpod.io. My be I need some special settings?

I need some more hints please, how to go on.

Regards, Christoph

NorthernMan54, 2024-12-13:
A couple of comments

1 - That issue is triggered by a bad compiler directive in platformio.ini -> change '-DsetFrequencyDeviation' to  '-DsetFreqDev'

2 - Those options can only be used one at a time

3 - Check the baud settings in putty, they need to match platformio / the compiled code

jaymemaurice, 2025-07-28:
> I refreshed the device decoders from rtl_433 24.10, maybe this is resolved?

I too tried latest build with OpenMqttGateway in FSK - no success with my ITH-20R :(

```
gitpod /workspace/OpenMQTTGateway (development) $ grep -i rtl_433_ESP platformio.ini
rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#64f0566
```

```
[env:lilygo-rtl_433-fsk]`
platform = espressif32@6.1.0
board = ttgo-lora32-v21
; ~/.platformio/packages/framework-arduinoespressif32/variants/.../pins_arduino.h
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.ssd1306}
  ${libraries.rtl_433_ESP}
build_flags =
  ${com-esp32.build_flags}
; *** OpenMQTTGateway Config ***
  ;'-UZmqttDiscovery'          ; disables MQTT Discovery
  '-DvalueAsATopic=true'    ; MQTT topic includes model and device
  '-DGateway_Name="OMG_lilygo_rtl_433_ESP_FSK"'
  '-DOOK_MODULATION=false' ; FSK modulation activated
; *** OpenMQTTGateway Modules ***
  '-DZgatewayRTL_433="rtl_433"'
  '-DZradioSX127x="SX127x"'
  '-DRTL_DEBUG=true'
  '-DPUBLISH_UNPARSED=true'
  '-DDEMOD_DEBUG=true'
; *** ssd1306 Display Options ***
  '-DZdisplaySSD1306="LilyGo_SSD1306"'
custom_description = For ESP32, Gateway using RTL_433_ESP and RadioLib with FSK modulation (beta)
custom_hardware = ESP32 LILYGO LoRa32 V2.1
```

Not able to get anything in the console when pushing the Tx button on the device. I don't have any FSK devices to test.

jaymemaurice, 2025-07-28:
I was finally able to get something finally when setting 

'-DsetFreqDev=80'

```
N: [ OMG->MQTT ] topic: home/rtl433a/RTL_433toMQTT/undecoded signal msg: {"model":"undecoded signal","protocol":"signal parsing failed","duration":1463996,"rssi":-53,"pulses":38} 

N: [ OMG->MQTT ] topic: home/rtl433a/RTL_433toMQTT/undecoded signal msg: {"model":"undecoded signal","protocol":"signal parsing failed","duration":1464000,"rssi":-50,"pulses":38} 

N: [ OMG->MQTT ] topic: home/rtl433a/RTL_433toMQTT/undecoded signal msg: {"model":"undecoded signal","protocol":"signal parsing failed","duration":1465182,"rssi":-55,"pulses":41} 

N: [ OMG->MQTT ] topic: home/rtl433a/RTL_433toMQTT/undecoded signal msg: {"model":"undecoded signal","protocol":"signal parsing failed","duration":5913,"rssi":-67,"pulses":29} 

N: [ OMG->MQTT ] topic: home/rtl433a/RTL_433toMQTT/undecoded signal msg: {"model":"undecoded signal","protocol":"signal parsing failed","duration":6242,"rssi":-100,"pulses":31} 

N: [ OMG->MQTT ] topic: home/rtl433a/RTL_433toMQTT/undecoded signal msg: {"model":"undecoded signal","protocol":"signal parsing failed","duration":1463996,"rssi":-57,"pulses":39} 
```

jaymemaurice, 2025-07-28:
I tried to increase PD_MAX_PULSES and BITBUF_COLS and got the not unexpected compile error which I assume is the device probably doesn't have enough memory to do that...

```
Linking .pio/build/lilygo-rtl_433-fsk/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/lilygo-rtl_433-fsk/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/gitpod/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/gitpod/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/gitpod/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 43232 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/lilygo-rtl_433-fsk/firmware.elf] Error 1
======================================================= [FAILED] Took 36.37 seconds =======================================================

```

jaymemaurice, 2025-07-28:
I'm going to put these back on the shelf to which they came for now unless someone has specific recommendations...
I have other OOK sensors which work well enough - The Thermpro Prologue protocol and unknown Chinese Ali Express LaCrosse TX141THBv2 based fridge sensors have been working very well with the OOK firmware.

Clearly these are working with the full rtl_433 - but I speculate now that this is just a matter of efficiency of the rx logic which may not work in the full scope of rtl_433 on such a limited embedded device.