NorthernMan54/rtl_433_ESP #17: Can't get to work in Arduino IDE with DOIT ESP32 DEVKIT1

davoau, 2021-11-28:
Hi,  Has anyone managed to get rtl_433_ESP to compile and load on a DOIT ESP32 DEVKIT1. 
I get the following errors:

C:\Users\DEFAUL~2\AppData\Local\Temp\arduino_build_72779\libraries\rtl_433_ESP\rtl_433_ESP.cpp.o:(.literal._ZN11rtl_433_ESP8rtlSetupEP5r_cfg+0x4): undefined reference to `acurite_rain_896'
C:\Users\DEFAUL~2\AppData\Local\Temp\arduino_build_72779\libraries\rtl_433_ESP\rtl_433_ESP.cpp.o:(.literal._ZN11rtl_433_ESP8rtlSetupEP5r_cfg+0x8): undefined reference to `acurite_th'
C:\Users\DEFAUL~2\AppData\Local\Temp\arduino_build_72779\libraries\rtl_433_ESP\rtl_433_ESP.cpp.o:(.literal._ZN11rtl_433_ESP8rtlSetupEP5r_cfg+0xc): undefined reference to `acurite_txr'
......
I have had lots of errors with the example ESP_rtl_433.ino - managed to fix most of them but cant get the above to go.

Maybe I have put the downloaded library files in wrong place - where should they go ?.

I did try the compiled version on openmqttgateway, it uploaded to the ESP32, but didn't work, and when I tried to re-upload to ESP32 again it had been bricked.. I could no longer upload anything to it - so don't want to try that again.

I have a CC1101 module connected to the ESP32, and it works with a SmartRC-CC1101-Driver example thats in the Arduino IDE. I can receive code from a 433mhz remote and send to homeassistant via mqtt publisher - eg {"id":31,"unit":15,"state":"down"}, {"id":31,"unit":15,"state":"up"}, "{"systemcode":480,"unitcode":0,"state":"on"}. So I am guessing I have wired up everything ok.

Anyway if anyone has example code they can share for the DOIT ESP32 that works with a fine offset weather station, that would be great.

Thanks

NorthernMan54, 2021-11-28:
What development environment are you using ?  I have been using Visual Studio with PlatformIO.

davoau, 2021-11-29:
Hi NorthernMan - thanks for the quick reply..

I am using Arduino IDE 18.1.5. 
Shouldn't it be able to compile with Arduino IDE anyways ?

So I have installed PlatformIO and VS code etc.. and now getting the below errors when compiling - should I worry about them as it says [SUCCESS] Took 36.53 seconds, so I am guessing its compiled ok ?

In file included from .pio\libdeps\esp32\rtl_433_ESP\src\rtl_433_ESP.cpp:22:0:
.pio\libdeps\esp32\rtl_433_ESP\src\rtl_433_ESP.cpp: In member function 'void rtl_433_ESP::loop()':
.pio\libdeps\esp32\rtl_433_ESP\include/log.h:35:80: warning: format '%d' expects argument of type 'int', but argument 2 has 
type 'long unsigned int' [-Wformat=]
 #define logprintf(prio, args...) {printf("rtl_433_ESP(%d): ", prio);printf(args);}
                                                                                ^
Compiling .pio\build\esp32\FrameworkArduino\FunctionalInterrupt.cpp.o
.pio\libdeps\esp32\rtl_433_ESP\src\rtl_433_ESP.cpp:451:9: note: in expansion of macro 'logprintf'
         logprintf(LOG_INFO, "RAW (%d): ", rtl_pulses->signalDuration);
         ^
.pio\libdeps\esp32\rtl_433_ESP\include/log.h:35:80: warning: format '%d' expects argument of type 'int', but argument 2 has 
type 'long unsigned int' [-Wformat=]
 #define logprintf(prio, args...) {printf("rtl_433_ESP(%d): ", prio);printf(args);}
Archiving .pio\build\esp32\lib24b\librtl_433_ESP.a
                                                                                ^
.pio\libdeps\esp32\rtl_433_ESP\src\rtl_433_ESP.cpp:451:9: note: in expansion of macro 'logprintf'
         logprintf(LOG_INFO, "RAW (%d): ", rtl_pulses->signalDuration);
         ^
.pio\libdeps\esp32\rtl_433_ESP\src\rtl_433_ESP.cpp:419:21: warning: unused variable 'signalProcessingStart' [-Wunused-variable]
       unsigned long signalProcessingStart = micros();
                     ^
Archiving .pio\build\esp32\lib909\libSmartRC-CC1101-Driver-Lib@2.5.5.a
.pio\libdeps\esp32\SmartRC-CC1101-Driver-Lib@2.5.5\ELECHOUSE_CC1101_SRC_DRV.cpp: In member function 'void ELECHOUSE_CC1101::Split_MDMCFG1()':
Compiling .pio\build\esp32\FrameworkArduino\HardwareSerial.cpp.o
.pio\libdeps\esp32\SmartRC-CC1101-Driver-Lib@2.5.5\ELECHOUSE_CC1101_SRC_DRV.cpp:888:5: warning: unused variable 's2' [-Wunused-variable]
 int s2 = 0;
     ^
Compiling .pio\build\esp32\FrameworkArduino\IPAddress.cpp.o

C:\Users\DefaultUser\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\DefaultUser\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\DefaultUser\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

davoau, 2021-11-29:
Hi again:
I have managed to get rtl_433_ESP compiled and installed onto the ESP32.
I tried a 4 button remote which gave me the output below on a SDR USB TV dongle connected to a PI Zero, and to the ESP32 running rtl_433 using the example ESP_rtl_433.ino.

FROM rtl_433_ESP
15:05 - esprf {"model":"Smoke-GS558","id":30679,"unit":31,"learn":0,"code":"1efaff","protocol":"Wireless Smoke and Heat Detector GS 558","rssi":-53,"duration":313692}

FROM RTL-SDR
15:05 - rtl_433/1046007 {"time":"2021-11-29 15:05:56","model":"Akhan-100F14","id":1046007,"data":"0x8 (Alarm)"}

So it all seems to work kind of... or its picking up a RF signal.

However the rt;_433_ESP doesnt pick up the weather station which is close (no more than 10M away), but is picked up by the SDR dongle which is aprox 15M away from weather station.

This is the output from SDR_RTL dongle:
14:56 - rtl_433/150 {"time":"2021-11-29 14:56:35","model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.2,"humidity":63,"wind_dir_deg":90,"wind_avg_km_h":1.224,"wind_max_km_h":3.672,"rain_mm":297.30002,"mic":"CRC"}

Is there something I can to do find out why ?

davoau, 2021-11-30:
Hi,  I moved the ESP closer to the weather station, and it still didn't receive any data.
I guessed that all fine offset weather stations would use the same pulse code modulation.

What PCM do the below use in rtl_433_ESP ?
Registering protocol [34] "Fine Offset WH1050 Weather Station"
Registering protocol [35] "Fine Offset Electronics WH1080/WH3080 Weather Station"

Is it possible to get rtl_433_ESP to support FSK_PULSE_PCM, as in copy files from SDR_RTL code ? 

I have this weather station [http://www.foshk.com/Weather_Professional/WN5300.html](url) - WN5300

On the wxforum is stays: The sensor array which comes with FineOffset WN5300, WN5350, WN2080 is NOT a WN67. It is a WN14 and uses a different modulation method for transmitting the sensor readings. WN14 uses ASK=amplitude shift keying and the WN67 uses FSK=frequency shift keying. So they are not interchangeable and cannot be received by the consoles in our matrix. The sensor arrays of Aercus 2085 and Aercus 3085 also use ASK.

So it looks like it uses ASK. Would it be possible to add that to rtl_433_ESP ?

Thanks for your help, and sorry for being a pain in the arse...  I would really, really, really like to get this working....

NorthernMan54, 2021-11-30:
One way to find out for sure is use rtl_433 and it’s signal analysis modes, I believe one of the will indicate the modulation scheme.

davoau, 2021-11-30:
I got this doing rtl_433 -A

Detected OOK package    2021-11-30 13:58:59
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-11-30 13:58:59
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 150
Battery   : 1            Temperature: 31.6 C       Humidity  : 68 %          Wind Direction: 113       Wind avg speed: 2.45      Wind gust : 7.34
Total rainfall: 329.4    Integrity : CRC
Analyzing pulses...
Total count:   88,  width: 180.48 ms            (45121 S)
Pulse width distribution:
 [ 0] count:    1,  width:  328 us [328;328]    (  82 S)
 [ 1] count:   33,  width:  508 us [500;512]    ( 127 S)
 [ 2] count:   54,  width: 1484 us [1480;1492]  ( 371 S)
Gap width distribution:
 [ 0] count:   87,  width:  952 us [948;964]    ( 238 S)
Pulse period distribution:
 [ 0] count:   33,  width: 1456 us [1284;1468]  ( 364 S)
 [ 1] count:   54,  width: 2440 us [2436;2448]  ( 610 S)
Pulse timing distribution:
 [ 0] count:    1,  width:  328 us [328;328]    (  82 S)
 [ 1] count:   33,  width:  508 us [500;512]    ( 127 S)
 [ 2] count:   54,  width: 1484 us [1480;1492]  ( 371 S)
 [ 3] count:   87,  width:  952 us [948;964]    ( 238 S)
 [ 4] count:    1,  width: 14924 us [14924;14924]       (3731 S)
Level estimates [high, low]:  16027,    618
RSSI: -0.1 dB SNR: 14.1 dB Noise: -14.2 dB
Frequency offsets [F1, F2]:   -3969,      0     (-15.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB105014801FC05CC03B83A4C839393939393939393A393A393A3A393A39393A3A3A393A39393A3A39393A3A3A393A3A3A393A3A3A3A3A3A3A3A393A3A3A3A3A3A39393A3A3A3A3A3A393A3A3A393A3A393A393A3A3A3A3A3A393A393A3A393A3A393939455
Attempting demodulation... short_width: 508, long_width: 1484, reset_limit: 968, sync_width: 328
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=508,l=1484,r=968,g=0,t=0,y=328'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {87} ff 52 c5 98 88 04 0c 08 94 0a 4e

NorthernMan54, 2021-12-02:
Sorry for the delay in getting back to you but that does look like OOK which is supported with the current package.  The only other issue could be range, which is know to be limited.

davoau, 2021-12-03:
Ok, I will put is very close to weather station and let you know what happens..

davoau, 2021-12-03:
I put the ESP32 within 3M of the weather station, I then also used the 4 button remote 5M from ESP32. It would pick up the signal from remote, and send mqtt message to home assistant. However there was no signal from weather station. The SDR_RTL was receiving data from the weather station during this, so weather station was transmitting.
Is there anything I can try ?

NorthernMan54, 2021-12-04:
Are you seeing any messages on the serial console ?

Have enabled publishing of unparsed messages ?

PUBLISH_UNPARSED=true

The serial console will log more details of signals received if you enable demod debug

DEMOD_DEBUG=true

davoau, 2021-12-04:
To enable publishing of unparsed messages and demod debug do I add the following to build flags (which I did) ?
  '-DDEMOD_DEBUG=true' ;
  '-DPUBLISH_UNPARSED=true' ;

An example of message in serial monitor terminal is as below... 
 �D0D @`r��K����? A�F�B��~��������������������
                                                                                      �������<������>��E�?���� ��<  �
                                                                                                                    ���

���
   ���=
       �L8!
           <
            ��
              ����OAN�P'�P'������&A^��K����`�1����{(^%d|$x$�D�Ko|�>�0��1�e(�1�L�!
                                                                                 >�|P���PO�P�
                                                                                             ~�|@���>�P0�
                                                                                                         >0
�p��o*K�e�Z��%���$�CT�
                      ��K�(�     �K�z��^yx����
                                              1�C~�!.
                                                     1�K�Z" �KA^r��H\B���%I�v5/���E}�%i
(�+�&�K"X
         �KE��/@'�*ȗ�/R1�K��PH^���#^� �%���KBA_�! �K�H<^�_"y�z�yy$���R
.H*�$A �L�v.�j��+�
                  )�

Which appears in MQTT as:
13:37:03 - esprf {"model":"unknown","protocol":"signal parsing failed","duration":274935,"signalRssi":-38,"pulses":92,"train":0,"messageCount":4,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-94,"minimumRssi":-82}

NorthernMan54, 2021-12-04:
That block of gobblygook should be in English, you have the incorrect baud rate for the serial monitor.

davoau, 2021-12-05:
NorthernMan, thanks for pointing out baud rate - I can't believe I am such a numb nuts.... anyways I fixed the speed so both platformIO.ini and ESP_rtl_433.ino are 115200. So now I get data in serial terminal.

rtl_433_ESP is getting the signal/data from weather station, and shows up in serial terminal.
Here is whats in JSONmessageBuffer
{"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery":"OK","temperature_C":33.1,"humidity":53,"wind_dir_deg":158,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":330.0,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-29,"duration":491802}

I think the problem is when the message is published.

If I send the below, it doesn't publish to MQTT:
{"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery":"OK","temperature_C":33.3,"humidity":56,"wind_dir_deg":158,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":330.0,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-40,"duration":497037}

But if I send the cut down version it does:
{"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery":"OK","temperature_C":33.3,"humidity":56,"wind_dir_deg":158,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":330.0,"mic":"CRC"}

I noticed that SDR_RTL only sends:
rtl_433/150 {"time":"2021-12-05 14:58:14","model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":33.1,"humidity":53,"wind_dir_deg":158,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":330.0,"mic":"CRC"}

So i though maybe MQTT has a message length max, and have changed as message length is are 280.
#define MQTT_MAX_PACKET_SIZE 256
 to
#define MQTT_MAX_PACKET_SIZE 328

Which hasnt made any difference - the massage still isn't being published.

Is there a way to suppress that last part of message below ?
,"protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-40,"duration":497037

And is it possible to only listen for my weather station messages.

NorthernMan54, 2021-12-05:
```
#define MQTT_MAX_PACKET_SIZE 256
```

Is that in the pubsubclient library ?

Googling shows this - https://github.com/knolleary/pubsubclient/issues/782

I would say go for the maximum value and work your way backwards.  The library is used with a lot of Arduino boards with limited resources, so going to 512 or 1024 should be fine on a ESP32

```
Is there a way to suppress that last part of message below ?
,"protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-40,"duration":497037
```

That is in r_api.c, around line 840

```
And is it possible to only listen for my weather station messages.

```

Nope sorry that is not a feature.  With my setup under OpenMqttGateway, I use the setting  '-DvalueAsASubject=true' , which includes the model and device id as part of the topic. ie home/9C9C1FC6F56C/RTL_433toMQTT/Acurite-Tower/2043

The first hex string is the openmqttgateway device macid.  ( I run multiple devices due to the range issue )

And the subscribe to the particular device messages.

davoau, 2021-12-06:
Yes I used the pubsubclient library.
I did try 512 for MQTT_MAX_PACKET_SIZE, but didnt make any difference, will try 1024.

I saw you had settings for gateway/wifi
  '-Dota_password=""'
  '-DMQTT_SERVER="192.168.1.12"'
;  '-DMDNS_SD=true'
  '-DESPWifiManualSetup=true'
  '-DGateway_Name="cc1101-ec20dc"'    ; 9e0770  ; ec20dc

I have wifi access point I connect to, SSID is sensor_ap, with password, so where would I put those settings, or do I just use my normal wifi library #include <WiFi.h> ?

NorthernMan54, 2021-12-06:
Personally I the codebase inside of the OpenMqttGateway firmware package, as it does all this and more.  Those settings were from openmqttgateway.

davoau, 2021-12-06:
I did try the pre-compiled version of openmqttgateway a few weeks ago (this was before I had touched arduino IDE or C++ and was trying the easy way), but my esp32 stopped working after upload from net onto it, and couldn't re-program it again. I will try the code and compile myself, and hopefully that will work..
Anyways thanks for all your help.. and now I know the RF transceiver works, and distance the signal goes etc, so at least I'm not driving blind anymore....

NorthernMan54, 2021-12-06:
Glad to hear you have it all sorted now.  And for myself right now my outdoor temperature sensors are only reporting 0C, and we have about 6 inches of white stuff on the ground.  But with that said, I really dislike how Covid has stopped people from traveling as I'm majorly missing the warmer climate white stuff.

It is about 6-9 months away from +34C here.

closed 2021-12-06 by NorthernMan54

Cezar8421, 2022-04-20:
Hello @ NorthernMan54 I am not a programmer but an electronics enthusiast. Could you, at my hot request, compile the bin file for OMG esp32dev-rtl_433 with device support:
`DECL (emos_e6016) \`
I will be very grateful for your help, I am impressed with your skills.

NorthernMan54, 2022-04-20:
Just looking at this, and will need to do a refresh of the rtl_433_ESP device decoders from rtl_433 to pick this up, then submit it to OMG for inclusion.  It will take a bit of time, so please stay tuned.

Cezar8421, 2022-04-21:
Thank you @NorthernMan54  very much and I am waiting impatiently. I am starting to learn PlatformIO, waiting for the current decoders.

NorthernMan54, 2022-04-22:
@Cezar8421 pls follow issue #31