NorthernMan54/rtl_433_ESP #19: TFA Dostmann 30.3221.02 T/H Outdoor Sensor not parsed

Dattel, 2021-12-17:
Hi,
since my old sensor got flooded during the last rain if've bourght a "TFA Dostmann 30.3221.02 T/H Outdoor Sensor".

Is there anything i need to know to enable parsing? My debug-console only reports "unparsed data"
I'm using rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP#e03f689
in OpenMQTTGateway

here are some captures from the console if i press the "TX" button on TFA-Sensor

```
rtl_433_ESP(6): Unparsed Signal length: 272560, Signal RSSI: -18, train: 0, messageCount: 2, pulses: 201
rtl_433_ESP(6): RAW (272560): +501-860+817-864+820-868+809-859+209-533+205-528+198-534+455-271+450-287+206-526+209-517+224-517+205-530+461-261+207-528+459-276+196-550+193-534+439-295+205-510+468-277+457-271+450-287+206-526+210-512+224-518+205-528+460-260+209-525+208-538+445-284+209-528+206-512+469-278+431-298+198-532+206-528+206-518+197-541+209-530+204-512+231-500+459-278+470-259+206-579+806-864+820-869+807-862+804-880+220-518+206-529+208-516+464-277+459-269+200-532+195-534+209-519+226-503+469-272+209-517+471-283+207-521+200-530+447-271+210-538+444-285+451-268+459-286+200-537+194-536+210-516+197-539+441-294+209-517+197-539+440-294+212-514+199-538+440-293+461-285+178-540+207-529+207-513+222-511+217-510+235-517+196-536+441-269+460-284+448-332+794-870+802-884+805-866+794-893+206-512+205-526+209-520+468-284+449-270+206-538+197-540+207-508+229-510+446-271+236-518+446-284+208-530+205-513+444-301+210-512+447-286+447-271+459-285+202-540+195-534+211-511+222-516+465-271+210-515+223-516+441-295+208-519+196-539+440-295+437-285+201-542+206-530+206-514+204-526+209-540+197-519+220-534+440-272+452-289+206-575+800-865+802-875+812-863+823-869+202-533+205-526+211-515+469-283+442-268+232-513+196-540+207-530+207-509+468-276+192-536+458-260+218-538+207-506+455-286+195-534+456-275+449-285+466-273+195-534+208-536+196-512+210-533+454-288+205-526+210-513+444-306+208-519+201-528+440-299+433-278+219-529+204-526+210-516+199-541+205-529+206-514+203-526+460-274+445-287+232-78913+75-506+0-549+0-524+124-101+0-303+74-100+100-101+350-125+176-74+150-74+0-125+324-227+125-101+125-327+250-76+75-100+0-76+0-227+200-101+297-428+125-279
 

rtl_433_ESP(6): Unparsed Signal length: 142499, Signal RSSI: -22, train: 1, messageCount: 3, pulses: 37
rtl_433_ESP(6): RAW (142499): +286-241+207-528+207-519+221-517+221-509+463-278+195-523+222-534+439-276+219-526+197-534+439-297+456-264+199-538+205-528+206-514+226-508+208-545+201-530+207-524+439-286+446-279+454-80337+76-99+74-375+209-125+198-527+175-402+176-126+226-278+350-351+425-175+127-125+427-127+173-279+195-75+151-101
 

rtl_433_ESP(6): Unparsed Signal length: 172251, Signal RSSI: -67, train: 0, messageCount: 4, pulses: 79
rtl_433_ESP(6): RAW (172251): +118-419+250-511+204-526+459-278+195-546+196-535+433-275+472-287+445-271+459-278+442-286+206-529+460-287+445-280+456-270+200-535+442-275+459-322+810-885+801-877+795-873+812-858+221-534+191-537+208-510+469-277+433-297+199-533+208-525+208-512+223-517+207-529+206-539+442-278+212-516+199-528+469-271+209-517+468-284+450-269+456-286+202-536+193-535+210-517+442-280+208-533+203-530+205-527+460-277+194-522+222-534+431-277+470-287+445-272+461-278+443-284+206-531+433-286+471-275+432-303+194-532+449-293+212-76795+447-504+124-478+444-175+51-304+124-428+126-126+102-126+210-99+126-328+441-127+99-453+174-380+123-457+124-327+199-100+124-399
 
rtl_433_ESP(6): Unparsed Signal length: 268762, Signal RSSI: -22, train: 1, messageCount: 5, pulses: 190
rtl_433_ESP(6): RAW (268762): +430-868+781-901+783-889+797-868+229-508+221-512+210-542+450-289+419-295+208-516+199-540+195-534+213-513+470-285+206-506+457-286+207-526+208-517+444-307+208-508+455-284+445-274+457-293+198-532+208-524+208-516+445-307+209-518+200-531+206-525+437-285+202-541+208-531+430-288+444-299+433-275+222-530+205-526+435-311+423-281+454-294+457-263+205-526+464-278+442-337+794-870+810-889+803-875+795-873+209-518+221-517+196-535+462-278+443-284+207-530+204-539+198-536+194-536+434-285+201-540+440-294+209-517+199-538+450-270+204-541+443-279+454-293+432-285+207-526+208-517+222-518+465-271+208-517+222-517+195-534+460-277+195-554+191-534+434-298+447-288+445-272+208-537+198-519+466-269+461-275+444-286+450-292+208-516+444-309+179-571+811-859+828-875+798-873+808-864+206-526+208-541+198-537+440-270+461-283+202-537+195-539+205-516+198-539+440-297+207-516+444-282+208-548+200-528+446-271+209-543+444-284+450-270+457-286+202-535+194-535+211-514+444-280+208-545+204-528+200-508+458-276+223-527+206-527+436-282+449-306+425-295+203-539+197-513+454-272+477-262+471-275+458-272+198-532+446-298+211-572+779-880+811-861+820-869+808-860+207-548+196-530+206-528+433-287+448-279+208-547+199-528+205-527+209-516+444-283+206-545+426-311+179-529+207-540+446-276+209-534+427-311+448-269+459-284+201-520+194-533+209-544+444-285+206-506+230-510+205-526+461-275+196-526+217-536+439-295+430-286+445-300+210-518+199-546+441-272+458-277+443-285+447-296+208-517+443-282+208-74328+382-1103+353-574+404-504+100-353+97-253+177-75+474-325+510-855+399-75+200-77 
 
rtl_433_ESP(6): Unparsed Signal length: 147803, Signal RSSI: -24, train: 0, messageCount: 6, pulses: 49
rtl_433_ESP(6): RAW (147803): +118-242+202-540+195-535+207-518+224-516+462-271+209-517+468-258+235-521+199-505+471-271+209-520+468-284+451-268+456-288+199-513+235-506+205-534+445-272+235-518+199-518+219-511+459-275+223-526+205-528+436-284+447-282+452-272+226-513+223-511+465-274+451-285+446-276+456-268+221-530+444-277+206-74499+100-426+128-401+151-403+100-277+300-202+1183-75+210-150+398-125+224-175+376-175+205-200+426-150

rtl_433_ESP(6): Unparsed Signal length: 275968, Signal RSSI: -52, train: 0, messageCount: 12, pulses: 198
rtl_433_ESP(6): RAW (275968): +694-862+814-868+806-883+814-852+220-519+234-501+236-486+469-276+463-271+228-487+221-511+236-504+231-510+465-275+193-533+460-263+235-500+237-493+470-260+215-509+489-253+465-286+463-244+234-519+199-510+477-268+231-492+223-512+232-506+229-511+467-253+213-536+206-516+467-278+466-250+220-505+233-526+461-261+447-281+236-497+471-263+474-244+234-520+220-514+464-293+841-843+824-853+810-867+821-869+232-485+230-526+211-518+468-257+456-270+226-513+235-496+237-488+220-518+464-272+235-494+468-258+235-510+225-510+471-246+216-534+460-265+464-259+476-268+232-490+222-509+463-269+235-515+197-512+218-532+211-515+468-284+208-506+230-510+445-272+485-249+222-529+205-504+487-259+448-282+235-496+475-262+473-247+233-518+196-537+196-580+815-845+826-855+834-863+817-842+231-516+229-499+240-484+470-286+450-268+232-491+226-508+234-506+230-511+443-275+236-495+471-283+209-504+232-510+470-248+236-510+447-290+445-269+462-288+201-538+207-506+459-288+207-503+234-514+223-512+218-509+461-275+219-501+233-502+462-285+446-280+210-521+224-501+471-271+461-280+217-498+465-271+461-276+219-528+205-501+239-540+825-855+811-868+822-844+827-880+210-509+228-502+226-506+486-249+470-264+233-502+234-488+223-537+206-522+452-261+218-509+459-275+221-530+195-533+456-273+223-509+465-277+464-248+481-257+217-535+211-505+456-290+206-528+211-519+198-538+205-505+458-288+237-498+209-519+467-259+475-247+227-514+223-511+463-271+457-266+219-519+464-274+460-274+221-498+234-501+234-82018+99-1001+224-201+174-399+150-75+203-76+477-177+251-78+99-100+176-101+127-176+126-101+602-77+74-126+328-150+200-299+77-73+250-125+452-77
 

rtl_433_ESP(6): Unparsed Signal length: 143350, Signal RSSI: -19, train: 1, messageCount: 13, pulses: 29
rtl_433_ESP(6): RAW (143350): +202-219+235-498+213-512+221-517+220-507+463-270+223-511+232-498+460-278+468-262+234-502+208-515+467-279+465-274+198-533+445-271+461-275+220-503+231-502+235-83467+100-479+75-452+175-100+484-305+125-101+275-101+176-152+428-76+75-99

```

Thanks :-D

NorthernMan54, 2021-12-17:
If you add this definition to your build, it will enable verbose mode in the rtl_433 device decoders.  It may give a clue as to why the decode is failing.

```
  '-DRTL_DEBUG=4'            ; rtl_433 verbose mode
```

I did check and the device should be supported as I assume it uses this device decoder [tfa_30_3221.c](https://github.com/NorthernMan54/rtl_433_ESP/blob/master/src/rtl_433/devices/tfa_30_3221.c) which is `OOK_PULSE_PWM`

When I looked at the debug output, what was surprising was that the #of pulses received was not consistent which may give a clue into why it is failing.

Dattel, 2021-12-17:
Wow.. that was a fast response. :-D

i tried to capture exactly one "TX" - Click... 
```
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: v0.9.8-development
N: WiFi ok with manual config credentials
Registering protocol [2] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [3] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas"
Registering protocol [4] "Acurite 986 Refrigerator / Freezer Thermometer"
Registering protocol [5] "Acurite 606TX Temperature Sensor"
Registering protocol [6] "Acurite 00275rm,00276rm Temp/Humidity with optional probe"
Registering protocol [7] "Acurite 590TX Temperature with optional Humidity"
Registering protocol [8] "Akhan 100F14 remote keyless entry"
Registering protocol [9] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [10] "Ambient Weather TX-8300 Temperature/Humidity Sensor"
Registering protocol [11] "Auriol AFW2A1 temperature/humidity sensor"
Registering protocol [12] "Auriol HG02832, HG05124A-DCF, Rubicson 48957 temperature/humidity sensor"
Registering protocol [13] "BlueLine Power Monitor"
Registering protocol [14] "Blyss DC5-UK-WH"
Registering protocol [16] "Bresser Thermo-/Hygro-Sensor 3CH"
Registering protocol [18] "Burnhard BBQ thermometer"        
Registering protocol [19] "Calibeur RF-104 Sensor"
Registering protocol [20] "Cardin S466-TX2"
Registering protocol [21] "Chuango Security Technology"        
Registering protocol [22] "Companion WTR001 Temperature Sensor"
Registering protocol [25] "Ecowitt Wireless Outdoor Thermometer WH53/WH0280/WH0281A"
Registering protocol [26] "Eurochron EFTH-800 temperature and humidity sensor"      
Registering protocol [30] "Esperanza EWS"
Registering protocol [32] "Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor"
Registering protocol [33] "Fine Offset Electronics, WH0530 Temperature/Rain Sensor"
Registering protocol [34] "Fine Offset WH1050 Weather Station"
Registering protocol [35] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [37] "FT-004-B Temperature Sensor"
Registering protocol [38] "Generic wireless motion sensor"
Registering protocol [39] "Generic Remote SC226x EV1527"  
Registering protocol [40] "Generic temperature sensor 1"  
Registering protocol [41] "Globaltronics QUIGG GT-TMBBQ-05"
Registering protocol [42] "Globaltronics GT-WT-02 Sensor"  
Registering protocol [43] "Globaltronics GT-WT-03 Sensor"  
Registering protocol [44] "Microchip HCS200 KeeLoq Hopping Encoder based remotes"
Registering protocol [45] "Honeywell ActivLink, Wireless Doorbell"
Registering protocol [46] "HT680 Remote control"
Registering protocol [47] "inFactory, nor-tec, FreeTec NC-3982-913 temperature humidity sensor"
Registering protocol [49] "Interlogix GE UTC Security Devices"
Registering protocol [51] "Kedsum Temperature & Humidity Sensor, Pearl NC-7415"
Registering protocol [52] "Kerui PIR / Contact Sensor"
Registering protocol [53] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [54] "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor"
Registering protocol [55] "LaCrosse/ELV/Conrad WS7000/WS2500 weather sensors"
Registering protocol [56] "LaCrosse WS-2310 / WS-3600 Weather Station"       
Registering protocol [58] "Maverick et73"
Registering protocol [60] "Missil ML0757 weather station"
Registering protocol [64] "Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor"
Registering protocol [66] "Opus/Imagintronix XT300 Soil Moisture"
Registering protocol [67] "Oregon Scientific SL109H Remote Thermal Hygro Sensor"
Registering protocol [69] "Philips outdoor temperature sensor (type AJ3650)"
Registering protocol [70] "Philips outdoor temperature sensor (type AJ7010)"
Registering protocol [71] "Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor"
Registering protocol [73] "Quhwa"
Registering protocol [75] "Rubicson Temperature Sensor"
Registering protocol [76] "Rubicson 48659 Thermometer"
Registering protocol [77] "Conrad S3318P, FreeTec NC-5849-913 temperature humidity sensor"
Registering protocol [78] "Silvercrest Remote Control"
Registering protocol [79] "Skylink HA-434TL motion sensor"
Registering protocol [80] "Wireless Smoke and Heat Detector GS 558"
Registering protocol [81] "Solight TE44/TE66, EMOS E0107T, NX-6876-917"
Registering protocol [82] "Springfield Temperature and Soil Moisture"
Registering protocol [83] "TFA Dostmann 30.3221.02 T/H Outdoor Sensor"
Registering protocol [84] "TFA Drop Rain Gauge 30.3233.01"
[... 283 lines cut ...]
rtl_433_ESP(7): demod(6) - Silvercrest Remote Control
rtl_433_ESP(7): demod(5) - Skylink HA-434TL motion sensor
skylink_motion_callback: rows 1 bits 17
rtl_433_ESP(7): demod(6) - Wireless Smoke and Heat Detector GS 558
rtl_433_ESP(7): demod(5) - Solight TE44/TE66, EMOS E0107T, NX-6876-917
rtl_433_ESP(7): demod(5) - Springfield Temperature and Soil Moisture
rtl_433_ESP(7): demod(6) - TFA Dostmann 30.3221.02 T/H Outdoor Sensor
rtl_433_ESP(7): demod(6) - TFA Drop Rain Gauge 30.3233.01
rtl_433_ESP(7): demod(5) - TFA pool temperature sensor
rtl_433_ESP(7): demod(5) - TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999
rtl_433_ESP(7): demod(5) - Thermopro TP11 Thermometer
rtl_433_ESP(7): demod(5) - Thermopro TP08/TP12/TP20 thermometer
rtl_433_ESP(7): demod(5) - TS-FT002 Wireless Ultrasonic Tank Liquid Level Meter With Temperature Sensor
[... cut ...] +443-308+181-546+203-528+440-301+433-273+470-287+207-529+435-284+201-538+210-503+230-510+444-301+431-278+219-529+198-79965+101-1100+125-498+150-400+199-226+249-99+100-100+75-52+123-425+100-199+51-226+98-174+101-198+125-228+0-204+97-299+0-176+174-301 
N: Send on /RTL_433toMQTT/unknown/null msg {"model":"unknown","protocol":"signal parsing failed","duration":273769,"signalRssi":-18,"pulses":197,"train":1,"messageCount":1,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-87,"minimumRssi":-82}
```

Dattel, 2021-12-17:
one more
```
rtl_433_ESP(7): demod(5) - Acurite 609TXC Temperature and Humidity Sensor
rtl_433_ESP(7): demod(6) - Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas
acurite_txr_decode: bitbuffer:: Number of rows: 25
[00] { 1} 80        : 1
[01] { 1} 00        : 0
[02] { 6} f8        : 111110
[03] { 1} 00        : 0
[04] { 1} 00        : 0
[05] { 2} 80        : 10
[06] { 2} 80        : 10
[07] { 1} 00        : 0
[08] { 2} 80        : 10
[09] { 6} f8        : 111110
[10] { 3} c0        : 110
[11] { 1} 00        : 0
[12] { 2} 80        : 10
[13] { 1} 00        : 0
[14] { 2} 80        : 10
[15] { 2} 80        : 10
[16] { 4} e0        : 1110
[17] { 1} 00        : 0
[18] { 2} 80        : 10
[19] { 1} 00        : 0
[20] { 1} 00        : 0
[21] { 6} f8        : 111110
[22] { 1} 00        : 0
[23] { 3} 40        : 010
[24] { 1} 76        : 0
... Maximum number of rows reached. Message is likely truncated.
acurite_txr_decode: row 0 bits 1, bytes 1
acurite_txr_decode: row 1 bits 1, bytes 1
acurite_txr_decode: row 2 bits 6, bytes 1 
acurite_txr_decode: row 3 bits 1, bytes 1
acurite_txr_decode: row 4 bits 1, bytes 1
acurite_txr_decode: row 5 bits 2, bytes 1
acurite_txr_decode: row 6 bits 2, bytes 1 
acurite_txr_decode: row 7 bits 1, bytes 1
acurite_txr_decode: row 8 bits 2, bytes 1
acurite_txr_decode: row 9 bits 6, bytes 1
acurite_txr_decode: row 10 bits 3, bytes 1 
acurite_txr_decode: row 11 bits 1, bytes 1
acurite_txr_decode: row 12 bits 2, bytes 1
acurite_txr_decode: row 13 bits 1, bytes 1
acurite_txr_decode: row 14 bits 2, bytes 1 
acurite_txr_decode: row 15 bits 2, bytes 1
acurite_txr_decode: row 16 bits 4, bytes 1
acurite_txr_decode: row 17 bits 1, bytes 1
acurite_txr_decode: row 18 bits 2, bytes 1 
acurite_txr_decode: row 19 bits 1, bytes 1
acurite_txr_decode: row 20 bits 1, bytes 1
acurite_txr_decode: row 21 bits 6, bytes 1
acurite_txr_decode: row 22 bits 1, bytes 1 
acurite_txr_decode: row 23 bits 3, bytes 1
acurite_txr_decode: row 24 bits 1, bytes 1
acurite_txr_decode: bitbuffer:: Number of rows: 1
[00] {11} 00 00     : 00000000 000
acurite_txr_decode: row 0 bits 11, bytes 2 
rtl_433_ESP(7): demod(5) - Acurite 986 Refrigerator / Freezer Thermometer
acurite_986_decode: skipping wrong len
rtl_433_ESP(7): demod(5) - Acurite 606TX Temperature Sensor
rtl_433_ESP(7): demod(6) - Acurite 00275rm,00276rm Temp/Humidity with optional probe
acurite_00275rm_decode: bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
acurite_00275rm_decode: bitbuffer:: Number of rows: 1
[00] { 0}           :
acurite_00275rm_decode: bitbuffer:: Number of rows: 1
[00] { 0}           :
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[... 436 lines cut ...]
rtl_433_ESP(7): demod(5) - Conrad S3318P, FreeTec NC-5849-913 temperature humidity sensor
rtl_433_ESP(7): demod(6) - Silvercrest Remote Control
rtl_433_ESP(7): demod(5) - Skylink HA-434TL motion sensor
rtl_433_ESP(7): demod(6) - Wireless Smoke and Heat Detector GS 558
rtl_433_ESP(7): demod(5) - Solight TE44/TE66, EMOS E0107T, NX-6876-917
rtl_433_ESP(7): demod(5) - Springfield Temperature and Soil Moisture
rtl_433_ESP(7): demod(6) - TFA Dostmann 30.3221.02 T/H Outdoor Sensor
rtl_433_ESP(7): demod(6) - TFA Drop Rain Gauge 30.3233.01
rtl_433_ESP(7): demod(5) - TFA pool temperature sensor
rtl_433_ESP(7): demod(5) - TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999
rtl_433_ESP(7): demod(5) - Thermopro TP11 Thermometer
rtl_433_ESP(7): demod(5) - Thermopro TP08/TP12/TP20 thermometer
rtl_433_ESP(7): demod(5) - TS-FT002 Wireless Ultrasonic Tank Liquid Level Meter With Temperature Sensor
rtl_433_ESP(7): demod(6) - Visonic powercode
rtl_433_ESP(7): demod(6) - Waveman Switch Transmitter
rtl_433_ESP(7): demod(6) - WG-PB12V1 Temperature Sensor
rtl_433_ESP(7): demod(6) - WS2032 weather station
rtl_433_ESP(7): demod(5) - Hyundai WS SENZOR Remote Temperature Sensor
rtl_433_ESP(7): demod(6) - WT0124 Pool Thermometer
rtl_433_ESP(7): demod(5) - X10 Security

rtl_433_ESP(6): Unparsed Signal length: 274254, Signal RSSI: -18, train: 1, messageCount: 33, pulses: 194
rtl_433_ESP(6): RAW (274254): +649-877+810-887+794-870+805-882+210-519+450-285+448-271+459-284+449-280+453-273+225-511+221-512+209-534+461-263+203-526+460-277+220-528+204-527+435-286+202-539+450-269+458-289+443-280+453-269+455-286+207-526+462-275+443-285+207-529+207-514+468-277+211-519+198-548+445-271+210-516+470-285+206-506+456-291+441-274+458-280+219-528+205-525+435-284+203-538+440-319+816-873+797-877+808-868+818-867+202-535+444-276+458-275+470-288+445-271+459-278+219-500+233-500+235-518+440-280+209-520+450-286+207-530+206-514+468-280+191-537+459-282+439-281+463-270+458-281+442-302+195-511+458-281+466-281+194-535+210-515+447-285+196-532+209-518+470-284+208-519+455-287+206-502+461-287+445-282+453-272+226-511+204-525+461-276+197-550+194-559+814-868+804-882+806-857+819-869+208-541+442-280+455-271+452-287+445-301+432-275+221-530+205-528+208-517+446-278+235-495+475-266+202-529+211-537+442-280+207-520+451-283+448-272+458-285+450-305+425-297+202-513+468-276+457-272+224-512+203-526+460-279+193-546+196-534+439-296+205-510+471-273+211-518+448-285+451-268+459-285+201-540+193-534+439-285+203-539+194-583+790-889+801-855+816-865+822-869+199-534+444-300+444-266+454-288+444-301+440-270+227-512+232-499+209-538+447-283+207-531+433-286+206-527+209-520+466-284+208-520+452-289+443-270+460-277+444-287+448-272+231-511+443-284+452-271+227-513+223-511+453-296+203-513+223-514+461-271+204-541+445-281+205-520+450-285+450-294+437-286+201-539+194-534+438-286+201-539+195-81578+507-552+526-504+151-301+276-254+274-101+302-250+77-125+99-479+100-454+99-354+74-75+201-302+176-76+125-151
N: Send on /RTL_433toMQTT/unknown/null msg {"model":"unknown","protocol":"signal parsing failed","duration":274254,"signalRssi":-18,"pulses":194,"train":1,"messageCount":33,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-87,"minimumRssi":-82}
```

NorthernMan54, 2021-12-17:
Well that device decoder doesn't have any debugging in verbose mode ;-(

It should have had some lines logged here, like some of the others do.

rtl_433_ESP(7): demod(6) - TFA Dostmann 30.3221.02 T/H Outdoor Sensor
rtl_433_ESP(7): demod(6) - TFA Drop Rain Gauge 30.3233.01

Well will need to debug this the hard way ( take the raw signal and walk it thru the device decoder code )

Dattel, 2021-12-17:
the sender itself seems to work fine..
i just restarted my old FHEM-Instance and attached my signalesp with CC1101 and FHEM can read the temp/hum

i'm not sure, if that will help you:

```
LASTDMSG: W89#7C42F026FF
RAWMSG: MU;P0=-305;P1=428;P2=180;P3=-543;P4=790;P5=-886;D=0101010101010102345454545231010101010232323102323232310231010101023232323232310232310102310101010101010102;CP=2;R=47;
```
batteryState | ok | 2021-12-17 17:59:48
channel | 1 | 2021-12-17 17:59:48
humidity | 38 | 2021-12-17 17:59:48
sendmode | manual | 2021-12-17 17:59:48
temperature | 25.2 | 2021-12-17 17:59:48
type | TFA 30.3221.02

Dattel, 2021-12-18:
the code from tfa_30_3221.c exits realy quickly at 

```
  // Device send 4 row, checking for two repeated
  row = bitbuffer_find_repeated_row(bitbuffer, (bitbuffer->num_rows > 4) ? 4 : 2, 40);
  if (row < 0)
      return DECODE_ABORT_EARLY;
```

i added some debug outputs to print out the number of rows before calling bitbuffer_find_repeated_row and an indication for the abort.

Here are some outputs:

```
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 2
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 2
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 2
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY

rtl_433_ESP(6): Unparsed Signal length: 264692, Signal RSSI: -64, train: 0, messageCount: 14, pulses: 198
rtl_433_ESP(6): RAW (264692): +101-724892+697-883+780-914+769-894+808-862+205-546+451-287+445-275+459-288+420-308+427-295+204-514+222-513+218-534+438-287+199-538+194-535+209-513+197-541+440-293+210-512+469-286+447-271+456-286+445-279+209-519+450-285+208-529+458-264+220-518+208-527+458-287+200-512+464-269+231-516+197-537+197-534+436-286+447-281+454-273+452-288+444-300+440-272+227-514+468-274+210-570+810-857+814-876+795-873+814-859+234-500+460-285+446-280+438-295+458-282+442-278+218-512+231-512+198-537+440-297+206-516+198-539+195-534+209-514+470-282+210-521+450-288+446-270+461-279+441-286+206-530+436-285+233-502+459-276+219-500+233-501+460-285+201-540+450-271+204-540+196-539+208-505+458-287+447-272+460-275+469-287+421-295+464-277+195-520+464-272+460-325+807-859+804-876+819-873+788-887+206-529+459-261+471-278+439-275+471-285+448-269+211-516+221-519+205-529+460-262+225-511+217-510+236-519+194-535+441-270+233-514+441-281+455-293+459-258+471-278+209-518+448-284+209-532+433-285+206-525+212-541+445-285+206-506+456-287+209-527+207-517+198-539+439-294+435-284+476-285+448-270+456-261+471-277+212-518+447-286+206-555+809-890+796-866+808-882+808-865+206-526+457-292+449-262+472-272+460-286+449-282+209-507+202-536+204-526+462-285+199-518+193-534+209-538+204-515+445-294+209-517+469-258+453-273+477-261+471-277+209-519+451-283+208-531+457-262+207-525+210-539+445-286+206-506+456-287+210-525+207-517+222-516+465-273+456-269+444-285+447-274+459-285+449-280+237-492+451-286+206-71617+126-424+52-73+101-481+100-500+100-279+223-403+176-232+297-177+376-228+151-226+125-403+174-101+531-153+485-174+125-126+101-227+174-380
N: Send on /RTL_433toMQTT/unknown/null msg {"model":"unknown","protocol":"signal parsing failed","duration":264692,"signalRssi":-64,"pulses":198,"train":0,"messageCount":14,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-86,"minimumRssi":-82}

tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 2
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
[... 41 lines cut ...]
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 2
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 2
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY
tfa_303221_callback: num Rows: 1
tfa_303221_callback: DECODE_ABORT_EARLY

rtl_433_ESP(6): Unparsed Signal length: 252699, Signal RSSI: -17, train: 1, messageCount: 19, pulses: 199
rtl_433_ESP(6): RAW (252699): +249-869+816-870+810-860+825-867+196-523+464-273+465-277+445-284+448-273+484-260+202-538+196-532+210-519+462-284+209-519+203-530+206-526+211-516+443-284+205-522+474-262+472-272+437-286+447-282+208-522+450-286+465-270+210-518+222-519+206-530+461-259+233-500+460-276+219-527+206-526+210-517+197-537+195-533+464-260+474-256+237-493+475-264+470-270+461-277+198-567+817-871+810-866+798-865+824-854+235-494+ [... cut ...] 53-296+204-513+222-510+221-506+235-518+197-536+442-267+459-287+202-539+464-269+461-261+471-258+234-60035+279-174+124-174+170-150+152-125+174-329+275-278+223-255+71-75+174-75+101-227+177-204+150-77+124-51+179-99+301-349+149-450+124-51+101-51+351-303
N: Send on /RTL_433toMQTT/unknown/null msg {"model":"unknown","protocol":"signal parsing failed","duration":252699,"signalRssi":-17,"pulses":199,"train":1,"messageCount":19,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-85,"minimumRssi":-82}

```

Dattel, 2021-12-18:
with the current rtl_433-refresh branch, the TFA is recognized as LaCrosse-TX141THBv2

```
{
  "model": "LaCrosse-TX141THBv2",
  "id": 124,
  "channel": 0,
  "battery_ok": 1,
  "temperature_C": 24.1,
  "humidity": 41,
  "test": "No",
  "mic": "CRC",
  "protocol": "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor",
  "rssi": -28,
  "duration": 265677
}
```

The test-button for manually transmitting is not recognized - ther terminal still counts a lot of unparsed signals.. but it's going in the right direction :-D

NorthernMan54, 2021-12-18:
So I took this debug line you shared earlier

```
rtl_433_ESP(6): Unparsed Signal length: 273763, Signal RSSI: -41, train: 0, messageCount: 18, pulses: 198
rtl_433_ESP(6): RAW (273763): +502-865+822-870+806-879+791-889+204-527+459-259+477-255+452-297+455-261+471-277+211-517+201-522+218-535+433-278+220-525+197-534+209-509+226-528+445-271+209-518+470-285+450-269+458-286+446-279+209-512+473-262+448-297+208-516+198-540+207-529+434-289+231-499+460-276+220-527+205-501+235-516+199-536+194-536+459-285+448-286+180-547+451-286+446-271+462-286+199-567+794-868+811-887+803-872+797-872+210-518+468-286+447-269+457-288+444-279+455-271+226-509+232-498+210-518+471-282+207-521+202-532+205-526+210-516+444-282+207-521+475-262+471-272+461-275+443-286+207-529+459-261+471-276+209-518+199-520+222-507+459-278+219-528+440-274+236-518+197-520+219-510+209-544+198-521+464-269+459-280+193-524+471-270+461-276+442-288+446-319+805-889+797-867+811-862+830-850+223-523+444-296+459-257+474-284+425-294+457-265+225-513+220-534+183-520+470-283+209-530+205-511+205-527+210-540+442-286+205-530+457-264+444-274+459-275+470-263+231-501+461-285+447-279+210-520+222-503+231-502+462-286+201-538+450-269+206-539+198-539+205-505+231-512+203-525+458-268+471-260+233-527+436-285+447-277+465-273+197-584+796-874+811-864+800-872+825-852+236-518+446-286+450-267+459-286+446-280+464-271+203-537+197-539+208-519+452-286+207-502+235-515+198-539+207-506+456-288+206-527+435-284+449-309+426-297+455-260+225-517+447-293+455-261+207-526+210-543+198-518+467-268+209-518+470-282+208-520+203-530+205-526+210-516+199-540+438-295+435-283+209-533+439-296+462-257+450-310+177-79133+630-528+99-1025+175-498+173-450+124-353+275-99+224-328+124-250+277-102+525-201+150-124+150-100+250-75+125-76+78-151+123-325+374-227+197-201
N: Send on /RTL_433toMQTT/unknown/null msg {"model":"unknown","protocol":"signal parsing failed","duration":273763,"signalRssi":-41,"pulses":198,"train":0,"messageCount":18,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-85,"minimumRssi":-82}
```

And after playing around with the data, it looks like this

```
01111100010000101111011000101000001101110
01111100010000101111011000101000001101111
01111100010000101111011000101000001101110
01111100010000101111011000101000001101110100000000100000010
```

And looking at the device decoder, the temperature bits would be approx 12 to 23

aka `0010 1111 0110` which is 758 and in the decoder they -500 and / 10 which is 25.8 which looks reasonable

PS Workflow was this

```
 cat issue#19.log | sed -e 's/+/,/g' | sed -e 's/-/,-/g'
```

Then grab the output into excel and run this against each number

```
=if(B2 < 0, ,if(B2 < 350, 0, if(B2 < 650, 1, "R")))
```
That gave the bit pattern

Dattel, 2021-12-18:
That means, the transmission itself and raw-receive component seems to work - the problem is part of the parsing?

NorthernMan54, 2021-12-18:
Do you have a rtl_sdr that you can run rtl_433 against to ensure that the decoder works consistently?

It would help show where the challenge is.  My code that passes the data stream to the device decoder or the decoder logic.

It would be interesting if rtl_433 had a similar challenge.

Dattel, 2021-12-18:
sadly no... 
My ESP32-OMG with CC1101 is running here to capture 3 Temperature-Sensors. All of a different type. Till that date the messurements work's very reliable. The rain busted one of them so i needed to buy a replacement. (TFA)

Since the update from today with the 'rtl_433-refresh' branch there are a lots of foreign devices (i had some of them before, but now they appear really quick after booting) ... either thats caused by a changed rssi tolerance or it's faulty created by background noise?

[image]

in the attached picture you can see the traffic sended to mqtt- currently i have disabled "unparsed signal reporting" to mqtt.
As you can see, Nexus-T and Prologue-TH seems to be parsed stable. 

The new TFA is parsed as LaCrosse device - but i found out, that this is a bit unreliable. Sometimes when i press the test button, the signal get's translated correct (even with the "Test":"Yes"-Property) but i think it's 1 out of 5.
Earlier i actually saw a TFA-Device on MQTT, but that was a single parsing-success.

Dattel, 2021-12-18:
...... omg

who can read has a clear advantage
this is an extract from tfa_30_3221.c

**This is the same as LaCrosse-TX141THBv2 and should be merged.**

NorthernMan54, 2021-12-19:
Is this the same device ?

https://www.canadiantire.ca/en/pdp/la-crosse-wireless-temperature-station-1427129p.html?rrec=true#spc

Manual says the sensor is `Insert 2-AA batteries (not included) into the TX141-Bv3 outdoor sensor`

Price was cheap, so was thinking of grabbing one to add to my fleet of sensors

Dattel, 2021-12-19:
Looks [terrifyingly](https://www.amazon.de/TFA-30-3221-02-Kombi-Sensor-Funk-433MHz/dp/B01NAW72SD) similar for me.
Just rebranded. Surprisingly there is a separate decoder in the RTL-Lib.

So i think, whatever you changed in these branch, it did the trick - device is parsed...

[image]

Sporadic dropouts when forcing manually transmitting with the "test" button are definitly acceptable for me.
Thanks for that.

NorthernMan54, 2021-12-21:
@Dattel I think we can close this as resolved with the rtl_433-refresh branch.  Also I think I resolved the 'dropouts'

Dattel, 2021-12-21:
very brilliant - thanks a lot :-D
i will test that out by tomorrow

NorthernMan54, 2021-12-23:
Resolved with #18

closed 2021-12-23 by NorthernMan54