NorthernMan54/rtl_433_ESP #20: Backlog for rtl_433-refresh branch aka next release

NorthernMan54, 2021-12-18:
This is the  rtl_433-refresh branch

1 - Random reboot

Serial monitor has this, No memory leaks or stack leaks visible

```
T: [ OMG->MQTT ] topic: home/240AC4EC20DC/RTL_433toMQTT/Acurite-Tower/9899 msg: {"model":"Acurite-Tower","id":9899,"channel":"C","battery_ok":1,"temperature_C":7.3,"humidity":62,"mic":"CHECKSUM","protocol":"Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas","rssi":-43,"duration":272394}
�������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������N:
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: esp32dev_rtl_433-v0.9.8-40-g0070967[development]
N: OTA Hostname: OMG_240AC4EC20DC.local
```

2 - Flex decoder needs more work or write a contrib device decoder for sonoff PIR device
    [ ] - Enable flex decoder
    [ ] -  A definition option for flex decoder ( single pattern )

3 - Try signal strength detection method ( this was initially investigated as part of FSK feature )

4 - Further investigation on the DEAF issue
    aka issue #16 

5 - Update of CC1101 driver version, needs to be part of openMQTTGateway's development branch
    2.5.7 is the latest

6 - Add to openMQTTGateway development branch

@DigiH - anything else to add to the backlog ?

@Dattel @mozgy - anything else ?

Dattel, 2021-12-18:
Currently not.
how do i include that branch properly in OMG to make a test build?

I have one more posible protocol confusion - Silvercrest remotes are recognized as SmokeDetectors :-D
Not sure, if that is intended. Currently i'm working with another OMG with RBX08 receiver and RF lib to receive these messages. It works, but the sensitivity is not very goot - so when you're already correcting things ...

Example from MQTT: 

home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/Smoke-GS558/9143
```
{
  "model": "Smoke-GS558",
  "id": 9143,
  "unit": 18,
  "learn": 0,
  "code": "0476f2",
  "protocol": "Wireless Smoke and Heat Detector GS 558",
  "rssi": -58,
  "duration": 647615
}
```

DigiH, 2021-12-18:
@NorthernMan54 - Thanks for the work, I haven't tested with my 868Mhz FSK weather station recently, but will test the new branch soon. Will also test with my currently running 434Mhz setup.

DigiH, 2021-12-19:
@NorthernMan54 - Just am a bit confused when going through the commits of the rtl_433-refresh branch. With the changes I made for the SUPPRESS_UNPARSED option, which is still in the ReadMe, and the connected PUBLISH_UNPARSED in the rtl_433_ESP.cpp, you deleted 

`#ifndef SUPPRESS_UNPARSED
#define PUBLISH_UNPARSED
#endif`

in rtl_433_ESP.h with https://github.com/NorthernMan54/rtl_433_ESP/commit/e03f689d8836bc48b3db628b33f414a67633b3f3

Making the described SUPPRESS_UNPARSED flag non-working.

Will you change the ReadMe to define the new optional flag as PUBLISH_UNPARSED, or was this an accidental deletion?

NorthernMan54, 2021-12-19:
@DigiH - I think I had changed that when I was making the changes for OpenMQTTGateway, I will adjust the README

Also added #7 to the backlog, realized that I had missed the other code components from rtl_433

NorthernMan54, 2021-12-21:
@Dattel I spent some time digging further into your comment about unparsed messages happening at a higher frequency when the test button was pressed, and found that the device decoder was always reporting issues with the first bit of the received signal.  And after comparing it with rtl_433 came to the conclusion that the first bit is just noise and can be ignored.   

@DigiH This applies to all signals, and I did do regression testing ( directly compared a rtl_433 with OMG running the latest version ) for the afternoon with my fleet of devices.  No issues identified.

From rtl_433_ESP
```
bitbuffer:: Number of rows: 5 
[00] { 1} 00             : 0
[01] {33} 5f 6d 35 ab 00 : 01011111 01101101 00110101 10101011 0
[02] {33} 5f 6d 35 ab 00 : 01011111 01101101 00110101 10101011 0
[03] {33} 5f 6d 35 ab 00 : 01011111 01101101 00110101 10101011 0
[04] {33} 5f 6d 35 ab 00 : 01011111 01101101 00110101 10101011 0
```

From rtl_433
```
bitbuffer:: Number of rows: 4 
[00] {33} 96 6d 46 f3 80 : 10010110 01101101 01000110 11110011 1
[01] {33} 96 6d 46 f3 80 : 10010110 01101101 01000110 11110011 1
[02] {33} 96 6d 46 f3 80 : 10010110 01101101 01000110 11110011 1
[03] {33} 96 6d 46 f3 80 : 10010110 01101101 01000110 11110011 1
```

Dattel, 2021-12-22:
So shall i test the  "manuall transmission" again with the latest build?

NorthernMan54, 2021-12-22:
Definitely

Dattel, 2021-12-22:
Current used libs

```
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <ArduinoJson> 6.18.3
|-- <ArduinoLog> 1.0.3+sha.d13cd80 
|-- <WifiManager> 1.0.0+sha.c3ff582
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <rtl_433_ESP> 0.0.3+sha.068514e
|   |-- <SmartRC-CC1101-Driver-Lib> 2.5.7
|   |   |-- <SPI> 1.0
|-- <rc-switch> 2.6.2+sha.36dfb45
|-- <EEPROM> 1.0.3
|-- <ESP32 BLE Arduino> 1.0.1
|-- <Wire> 1.0.1
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <SmartRC-CC1101-Driver-Lib> 2.5.7
|   |-- <SPI> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Preferences> 1.0
|-- <SPI> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Update> 1.0

```

Each transmission (by button or automatically) results in one "parsed signal" followed by an "unparsed signal"

```
N: Send on /RTL_433toMQTT/LaCrosse-TX141THBv2/124 msg {"model":"LaCrosse-TX141THBv2","id":124,"channel":2,"battery_ok":1,"temperature_C":18.6,"humidity":71,"test":"Yes","mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-19,"duration":275011}
 
 
rtl_433_ESP(6): Unparsed Signal length: 260116, Signal RSSI: -20, train: 1, messageCount: 117, pulses: 163
rtl_433_ESP(6): RAW (260116): +201-512+464-269+233-518+441-281+234-506+456-259+465-275+463-252+244-499+231-503+462-288+199-515+235-504+230-510+468-252+461-276+469-263+467-270+230-489+224-540+206-498+473-263+473-271+462-259+474-258+211-570+813-861+821-851+817-869+835-856+197-533+459-278+466-261+450-269+488-257+449-279+236-506+200-534+206-502+491-258+450-280+238-504+202-532+206-526+466-252+216-519+466-269+230-490+467-280+234-482+477-263+462-272+466-252+243-498+232-503+462-285+201-517+233-494+225-506+494-250+465-273+466-262+473-246+234-518+196-538+219-509+461-276+466-263+448-273+462-284+200-562+816-867+812-859+816-872+824-853+236-492+469-285+465-248+477-263+471-251+463-268+226-511+221-512+236-505+455-286+462-250+216-535+206-511+221-509+466-272+226-514+462-278+191-534+459-266+219-521+464-270+457-260+485-256+210-520+225-502+471-273+238-489+221-517+217-509+467-251+489-262+469-243+480-261+226-511+217-510+235-517+442-282+468-272+452-287+467-247+233-543+822-854+836-834+825-870+830-836+235-511+455-287+464-250+462-268+479-262+471-252+239-511+206-514+230-500+484-242+471-289+207-503+235-518+197-511+493-244+231-511+468-276+211-521+444-286+207-506+483-264+445-274+466-276+222-502+230-504+464-287+201-513+235-495+230-499+469-276+465-260+473-259+477-246+229-514+223-512+219-506+465-286+448-280+470-246+472-290+445-86101+508-1841+511-1272 
```

```
N: Send on /RTL_433toMQTT/LaCrosse-TX141THBv2/124 msg {"model":"LaCrosse-TX141THBv2","id":124,"channel":2,"battery_ok":1,"temperature_C":19.2,"humidity":70,"test":"Yes","mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-18,"duration":276040}
 

rtl_433_ESP(6): Unparsed Signal length: 258127, Signal RSSI: -19, train: 0, messageCount: 136, pulses: 159
rtl_433_ESP(6): RAW (258127): +210-519+470-256+479-245+229-514+467-280+210-495+247-500+234-501+460-289+202-510+216-509+235-513+467-255+478-243+227-517+222-511+234-506+202-534+232-501+211-519+221-515+208-531+459-263+235-525+842-839+822-855+841-834+820-867+230-511+468-252+460-276+469-263+472-272+461-265+221-503+221-507+237-508+474-261+474-244+234-512+221-516+236-499+460-262+236-500+461-276+218-523+464-251+460-277+220-501+469-274+237-494+221-518+218-509+467-259+227-516+220-506+238-486+469-285+451-268+231-492+223-510+243-509+211-518+222-516+218-510+210-519+223-514+463-271+237-540+803-858+816-860+824-866+806-884+210-510+480-259+461-278+464-245+480-261+460-281+217-509+234-513+222-490+488-247+459-288+201-514+218-531+237-484+471-255+236-495+476-288+208-503+458-287+446-279+210-509+478-259+237-500+234-492+221-510+463-271+231-511+223-512+218-510+462-279+465-263+235-478+230-515+222-512+210-519+226-502+231-505+233-495+242-490+464-269+233-541+828-855+817-866+825-846+825-854+237-511+453-285+437-273+460-269+470-287+446-273+236-487+221-520+220-508+492-250+469-261+236-500+208-518+222-512+462-271+233-488+493-255+210-521+472-263+471-248+233-521+464-256+211-509+225-531+205-502+464-287+224-493+216-533+237-492+468-257+480-247+226-516+229-499+236-509+199-523+234-501+234-492+224-512+236-512+464-257+473-85562 
```

```
N: Send on /RTL_433toMQTT/LaCrosse-TX141THBv2/124 msg {"model":"LaCrosse-TX141THBv2","id":124,"channel":2,"battery_ok":1,"temperature_C":19.8,"humidity":68,"test":"No","mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-35,"duration":258574}
 

rtl_433_ESP(6): Unparsed Signal length: 246389, Signal RSSI: -33, train: 1, messageCount: 163, pulses: 183
rtl_433_ESP(6): RAW (246389): +787-888+799-897+800-854+209-544+446-282+451-268+460-284+446-280+465-272+198-533+204-526+210-518+223-516+449-271+232-516+196-540+193-534+461-277+195-520+465-273+209-518+471-283+451-269+457-266+219-520+463-272+211-516+222-517+465-269+209-513+222-519+217-510+459-278+219-527+206-502+461-286+446-281+209-522+223-503+470-271+461-276+444-285+449-294+437-337+804-880+790-863+822-869+803-881+210-520+448-286+447-270+459-286+448-281+452-272+2 [... cut ...] +465-246+232-517+470-255 
 

rtl_433_ESP(6): Unparsed Signal length: 151914, Signal RSSI: -24, train: 0, messageCount: 178, pulses: 41
rtl_433_ESP(6): RAW (151914): +445-273+460-276+445-288+444-296+210-514+198-541+206-530+434-287+446-303+207-512+198-548+195-534+433-298+198-531+443-277+234-510+446-287+448-269+459-284+449-280+464-273+202-533+203-527+460-283+203-517+206-529+210-537+197-514+206-535+201-532+444-273+218-536+208-509+207-526+207-540+445-283+209-529+432-288+444-79647+542-1876+97-605 
```

NorthernMan54, 2021-12-22:
@Dattel would you call that an improvement ?

Looking at the raw signals, their is a tiny bit of noise triggering the unparsed

Manually parsing one, it looks like this

```
01111100001000101011101001000100110011111
01111100001000101011101001000100110011110
01111100001000101011101001000100110011111
01111100001000101011101001000100110011110
1
111
```

That is marked as unparsed, as the device decoder has some funky rules about # of rows and repeats

If the signal has more than 4 rows, then 5 rows need to match, then only 3 need to match. ( lacrosse_tx141.c line 131 )

Dattel, 2021-12-22:
it's definitiv much better than before 👍 
I think we all can live with this - or not? 

One remark - i think @DigiH already mentioned it - unparsed signales are still printed to terminal - independently to the compiler switch.

DigiH, 2021-12-22:
Also much better results here for the reboots, running fine here for the last 3 days. 👍

@Dattel - the implementation for suppressing (now activating) unparsed signals was only really meant for MQTT messages (not)being published for these. The unparsed signals still showing in the serial terminal, I think, is fine for debugging, and would need a separate change if also wanting to be suppresed in terminal.

Dattel, 2021-12-22:
not it's okay... 
just thought the terminal should also suppress unparsed messages with these settings :-) 
I'm not sure, if writing to the terminal (even if nothing is connected) eats up cpu time/ressources - especially due to the mem-allocs/deallocs used by the print/formatstring functions.

NorthernMan54, 2021-12-22:
Well, in the latest commit I stopped logging of unparsed messages to the serial output unless PUBLISH_UNPARSED is set.

I also added the ability to enable device decoder verbose mode for a single device decoder

```
RTL_VERBOSE=##        ; Enable RTL_433 device decoder verbose mode, ## is the decoder # from the appropriate memcpy line in rtl_433_ESP.cpp
```

NorthernMan54, 2021-12-22:
From the list at the top of this issue, I think I'm ready to publish this as the final.  I'm going to park item 3 and 4 for now.

Dattel, 2021-12-23:
> I also added the ability to enable device decoder verbose mode for a single device decoder

i think, that is very helpfull for the future 👍 
i just upgraded to the latest version and terminal looks much cleaner now.. 

```
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: v0.9.9-development
N: WiFi ok with manual config credentials
N: RF_EMITTER_GPIO: 4
N: RF_RECEIVER_GPIO: 15
N: ZgatewayRTL_433 setup done
N: Switching to RTL_433 Receiver: 433.92Mhz
N: OpenMQTTGateway modules: ["RF","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
N: Connected to broker
N: Send on /SYStoMQTT msg {"uptime":6,"version":"v0.9.9-development","freemem":122072,"mqttport":"1883","mqttsecure":false,"freestack":4792,"rssi":-61,"SSID":"XXXX","ip":"192.168.178.77","mac":"XXXX","actRec":3,"mhz":433.92,"RTLminRssi":-82,"RTLRssi":0,"RTLCnt":0,"modules":["RF","rtl_433"]}

N: Send on /RTL_433toMQTT/Nexus-T/167 msg {"model":"Nexus-T","id":167,"channel":1,"battery_ok":1,"temperature_C":22.6,"protocol":"Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor","rssi":-72,"duration":966815}

N: Send on /RTL_433toMQTT/Nexus-T/167 msg {"model":"Nexus-T","id":167,"channel":1,"battery_ok":1,"temperature_C":22.6,"protocol":"Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor","rssi":-72,"duration":967109}

N: Send on /RTL_433toMQTT/Prologue-TH/5 msg {"model":"Prologue-TH","subtype":9,"id":5,"channel":1,"battery_ok":1,"temperature_C":19.7,"humidity":0,"button":0,"protocol":"Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor","rssi":-57,"duration":898715}

N: Send on /RTL_433toMQTT/LaCrosse-TX141THBv2/124 msg {"model":"LaCrosse-TX141THBv2","id":124,"channel":2,"battery_ok":1,"temperature_C":-2.7,"humidity":88,"test":"No","mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-52,"duration":211585}
```

NorthernMan54, 2021-12-23:
Tks everyone for their assistance with this release, I have now merged the branch into the master via #18

Now to get openMQTTGateway updated as well

NorthernMan54, 2021-12-23:
https://github.com/1technophile/OpenMQTTGateway/pull/1122

NorthernMan54, 2023-04-03:
This was released this morning