NorthernMan54/rtl_433_ESP #22: HELP with decoding Fineoffset-WHx080

garudaonekh, 2022-01-05:
Hi,
I try to decode my weather station with this library. I set my devices to Fine Offset Electronics WH1080/WH3080 Weather Station. Do I miss something?

Please note that, I use RTL_SDR(rtl_433 on RasberryPI) which can decode this data successfully.

Here's the log I got on my ESP32:
```
rtl_433_ESP(6): RAW (1025187): +475-512+470-483+495-512+474-498+478-491+516-448+502-487+534-423+527-492+501-435+524-488+490-497+502-528+514-942+1003-967+995-962+997-146+0-788+1029-486+517-941+512-456+523-462+519-471+520-465+496-489+977-286+0-387+0-308+985-482+517-927+524-494+520-430+524-477+519-441+1007-489+522-944+524-438+546-438+520-472+514-479+504-469+1011-456+518-454+527-951+995-472+516-934+1030-930+532-498+987-964+500-464+522-455+523-475+491-476+506-489+536-499+491-487+510-445+508-492+528-449+502-487+521-393+1053-935+1028-489+495-463+521-938+529-465+501-489+520-461+519-462+501-472+1007-940+532-466+542-403+524-489+492-504+479-501+482-475+502-501+479-484+496-487+499-484+500-481+502-471+990-961+534-450+1008-932+1023-942+1029-921+1029-948+1001-961+522-491+496-464+673-742+30187-724+0-423+0-798+0-400+0-399+0-351+174-251+99-300+150-75+324735-1945+0-797+0-374+100-301+97-323+75-279+0-77341+0-778+75-351+0-377+99-278+198-176+201-203+72710-449+506-491+524-430+498-515+489-444+529-473+510-487+506-470+507-485+474-487+497-488+523-457+495-493+515-453+503-489+539-937+1017-941+994-960+985-512+0-434+1006-496+516-962+595-331+551-488+504-445+535-447+514-463+1006-985+972-499+511-937+505-498+509-499+512-465+534-425+1030-466+506-974+544-403+550-459+556-396+530-469+516-462+1008-484+554-403+528-956+997-492+498-940+1035-951+505-490+971-984+538-403+548-447+539-423+530-467+541-417+557-469+490-468+529-471+508-462+521-458+523-464+524-457+1010-872+1081-470+505-470+517-926+523-490+524-447+529-452+516-458+530-469+1060-888+520-467+516-463+498-205+0-282+551-485+498-491+500-453+527-476+536-425+506-491+523-403+549-466+573-401+1004-964+509-469+1043-944+1004-937+1018-944+1023-934+1009-882+580-491+522-429+747-701+0-108583+75-354+0-755+124-277+124-276+148-224+201-203+247-151 
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
 
rtl_433_ESP(6): RAW (159590): +496-486+496-477+502-478+494-487+506-476+504-490+522-461+493-488+549-374+593-487+523-456+497-469+560-376+551-492+506-942+1018-890+1070-944+1008-934+1026-470+514-982+496-427+556-451+541-445+507-321+0-150+516-466+1005-974+972-495+515-940+520-464+546-406+547-473+539-397+1028-490+493-973+505-458+522-465+503-485+500-482+510-467+982-510+502-443+531-950+997-296+0-174+512-931+1030-944+527-492+1003-349+0-588+522-449+566-491+495-454+513-477+512-466+510-494+494-460+525-449+526-485+496-463+522-463+552-379+1056-947+1005-473+534-348+0-102+511-945+532-473+487-486+502-473+505-480+518-430+1022-964+518-461+529-428+528-475+562-376+551-490+501-460+522-463+520-455+528-453+510-487+507-471+510-461+994-964+518-460+994-527+0-369+1077-921+1032-938+1023-947+1030-942+508-493+485-473+728-917+0-1588+0-1958+0-3162 
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
 
rtl_433_ESP(6): RAW (196156): +480-1014+466-976+477-988+482-994+489-975+494-957+1478-957+1485-953+514-954+497-965+521-947+1488-966+1461-967+507-971+498-958+498-945+1505-964+1488-956+1468-971+1464-964+517-968+1459-972+506-955+1467-993+490-954+1504-959+1460-953+1491-965+1465-967+494-998+1459-970+1463-977+1461-973+487-977+1007-2231+74-127+492-999+1451-1002+473-960+1478-965+1484-956+1494-949+1482-966+1465-981+1472-944+1509-963+1485-956+489-942+528-966+491-954+509-958+503-981+491-968+503-962+486-970+509-970+1467-979+496-956+1461-985+513-945+1470-969+504-958+508-974+1474-945+520-965+1454-990+483-969+514-946+1495-971+478-964+513-13190+532-2363+542-1969+524-418+75-297+99-720+482-795+502-397+75-348 
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
 
rtl_433_ESP(6): RAW (523455): +488-973+478-997+468-988+481-1004+457-993+466-985+501-965+1463-990+485-998+1446-973+494-988+1434-992+1458-989+489-972+467-988+1482-965+489-996+464-995+1449-978+1452-1012+459-989+1442-994+468-1010+1455-963+508-964+484-995+470-980+1451-988+1484-964+1462-984+1458-966+1467-994+469-987+480-990+486-995+1447-973+1469-984+1457-989+1435-990+1467-974+1468-990+1443-983+1448-1012+1435-989+1463-990+1463-967+1451-993+1471-960+1449-1013+1461-967+1460-985+1459-974+1455-993+1466-982+1444-986+1460-989+1461-989+1435-997+1451-993+1468-983+1430-1011+1435-991+1461-989+1442-992+1444-1001+1447-989+1458-990+1460-990+1441-994+1447-997+472-988+1461-988+1441-990+1466-978+472-989+485-974+1457-995+469-984+501-964+1462-981+483-991+466-1008+1453-965+482-991+1462-975+478-30991+489-1011+453-1030+459-994+468-1006+448-1008+458-991+490-980+473-1009+1436-981+490-990+1446-975+494-965+1486-982+1436-1003+464-994+493-963+1456-1009+461-998+464-999+1448-988+1456-990+483-975+1465-971+496-979+1446-986+482-990+489-974+493-977+1448-986+1456-985+1462-988+1466-968+1446-1001+474-987+485-975+488-973+1472-987+1451-989+1457-990+1440-990+1469-974+1447-987+1457-992+1459-965+1461-990+1462-975+1454-992+1467-981+1446-988+1458-990+1461-991+1438-978+1461-996+1467-973+1443-1008+1454-965+1459-988+1465-980+1462-974+1455-998+1443-1000+1447-987+1458-987+1462-964+1462-979+1460-1000+3896-992+1435-984+1463-991+1461-975+489-973+1468-987+1456-987+1460-966+486-974+502-975+1444-983+500-965+486-997+1452-971+493-986+478-990+1462-965+484-993+1466-979+471-70114+0-3936+0-12618+0-401+76-300+100-326+0-406+0-402+75-326+0-401 
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
```

garudaonekh, 2022-01-05:
Here's the output I got with the rtl_433 on Raspberry PI with SDR_RTL:

 ```
rtl_433 -R 32:vv
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/chamroeun/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 207 device decoding protocols [ 32 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Fine Offset WH1080 data {80} 01 fd 4d 96 51 a0 00 00 00 00 
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
bitbuffer:: Number of rows: 1 
[00] {93} 01 fd 4d 96 51 a0 00 00 00 00 3d f0 
Fine Offset WH1080 data {80} ff a9 b2 ca 34 00 00 00 00 07 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-05 16:37:18
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 155
Battery   : 1            Temperature: 31.4 C       Humidity  : 52 %          Wind Direction: 158       Wind avg speed: 0.00      Wind gust : 0.00          Total rainfall: 0.0
Integrity : CRC
```

garudaonekh, 2022-01-24:
Any suggestion, please?

NorthernMan54, 2022-01-24:
Sorry, missed this earlier

It appears that the device decoder is decoding and returning a decoded message - I'm guessing this is the case as this line is appearing in the log.  Is something wrong with your callback ?

https://github.com/NorthernMan54/rtl_433_ESP/blob/08966ffa3e9d90bbe0b450ffa20a58d2da20c83a/src/rtl_433/pulse_demod.c#L51

Can you try with the sample in the #issue #23 branch, and just change the decoder to your device?

garudaonekh, 2022-01-25:
I tried this branch #23 but still can't decode the message:

Here's what I get when I print raw data:

`> Executing task: platformio device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/tty.usbserial-0001  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

load:�bb���10124
load:0x40080400,len:5828
entry 0x400806a8
Registering protocol [1] "Fine Offset Electronics WH1080/WH3080 Weather Station"
 
rtl_433_ESP(6): RAW (45192646): +51-380+51-305+76-20033+75-406+76-276+127-25099+73-351+100-332+100-306+51-9848+176-5874+152-85918+100-379+51-2184+101-402+150-256+147-1090+52-329+126-256+152-235+76-11925+75-23155+75-379+75-350+72-84292+75-1550+101-79942+75-35031+74-4725+99-546+99-299+124-12128+76-699+125-377+125-373+150-329+72-350+125-278+127-1055+149-252+154-30767+74-327+126-278+149-678+76-305+149-253+126-278+101-1100+75-7054+74-24337+75-321+149-273+124-30649+52727-375+75-377+76-680+99-400+72-378+126-252+123-38944+99-6991+74-7891+73-421+100-273+100-12120+73-719+125-299+122-7376+76-22402+101-403+127-727+149-747+125-400+76-328+126-277+76-12544+75-723+100-1938+75-253+51-76+72-303+149-3420+124-28778+74-727+99-1507+75-18879+126-22036+101-3454+101-349+102-28691+76-39073+2006-3156+101-1498+121-447+75-324+75-1046+75-456+74-302+125-251+124-27249+100-20361+148-1121+75-323+73-348+74-344+51-797+102-7905+75-792+99-1464+125-39305+100-1235+152-325+74-1613+148-349+173-17632+76-4630+73-399+149-274+75-44932+151-331+126-330+125-1515+98-404+51-328+77-329+74-29208+101-1124+122-23564+76-297+99-371+127-772+100-276+100-299+124-299+74-1108+76-20799+71-302+172-250+121-323+150-404+76-305+151-32138+149-277+101-301+101-6615+51-6253+100-323+100-251+146-1471+74-55569+75-5195+76-2381+122-2337+75-1165+75-371+100-375+99-23586+661-298+125-695+75-3435+74-323+100-296+173-35366+75-1490+73-5109+72-1126+145-426+99-278+127-681+101-278+196-51936+76-11330+74-1636+100-250+125-274+100-2359+101-308+150-279+124-23947+0-382+103-686+126-43632+51-733+152-256+51-18473+97-2744+74-299+175-651+75-328+174-254+61396-759+76-400+101-276+126-1860+51-1530+51-398+145-10511+74-295+151-8235+100-700+127-2250+74-726+126-255+123-707+101-402+102-497+126-277+198-21745+51-7866+74-299+100-33159+51-1086+124-380+150-277+127-17621+3527-419+99-726+101-757+75-331+75-381+152-302+100-1052+74-1616+101-396+126-394+100-9417+126-828+151-21667+125-10617+102-21191+74-742+54-11350+76-432+101-28659+51-7427+75-6628+51-355+127-257+126-2254+124-404+123-10998+101-283+101-21248+100-381+100-735+51-1163+75-9863+101-372+125-272+96-28675+75-395+74-48764+73-1147+126-275+125-1074+51-10181+100-355+76-277+151-276+76-303+74-6318+125-790+51-23940+73-35918+76-3082+76-26036+100-35393+677-9790+74-1918+75-329+152-251+125-1076+126-35363+100-384+74-380+100-277+101-73168+51-381+76-349+72-348+148-424+75-381+174-50294+100-407+76-307+126-636+76-1147+74-423+99-201+123-85138+75-307+101-34992+152-381+100-687+99-31434+76-383+51-279+126-1490+75-304+76-11459+75-356+75-2642+100-3950+174-401+101-327+101-255+98-12717+75-25524+99-444+74-297+99-297+75-371+74-29550+100-92976+77-332+101-1063+76-47263+75-276+125-10109+127-24310+98-280+147-1152+76-278+148-329+72-376+152-7372+99-72390+51-406+100-297+76-33113+75-722+100-8974+153-4684+75-401+153-205+203-231+125-1116+74-505+178-280+150-38095+125-813+76-12964+151-567+100-277+122-298+483-400+125-301+75-700+73-305+74-424+126-301+100-656+71-8301+51-1520+54-2274+99-325+147-250+101-327+76-274+75-473+124-246+98-10260+76-733+76-279+73-755+126-8178+74-346+74-298+76-2012+76-4613+75-20973+100-373+99-720+75-3128+449-347+99-346+74-13345+102-400+74-729+127-8930+51-429+101-248+177-299+51-35835+75-794+75-1972+125-37391+101-325+101-23183+75-1130+98-377+149-279+126-250+100-452+75-351+200-251+51-13352+102-18514+120-296+98-30268+74-1228+74-11389+74-8243+75-720+490-771+74-31484+122-422+76-23644+75-251+147-295+74-1904+75-19252+76-777+100-849+101-4359+51-304+103-688+76-357+51-11404+74-330+101-282+177-254+127-257+99-305+75-706+171-296+99-249+97-326+99-19807+150-804+51-65658+51-402+151-278+126-25287+75-700+128-1064+51-1149+124-660+98-328+100-58351+76-396+74-4229+99-24775+51-3464+98-1098+98-431+76-382+127-280+76-22000+101-297+125-617+123-353+124-22033+122-449+75-296+74-2298+102-379+126-303+127-687+77-17671+420-2778+484-326+71-370+75-322+123-672+125-1070+53-8283+76-380+125-757+121-18512+472-403+101-305+51-3125+74-9396+74-8593+74-18193+75-1957+51-46061+642-272+99-1121+51-803+74-327+150-199+124-346+102-799+75-5926+487-1516+123-80757+1445-419+74-64128+126-250+151-249+123-27462+73-370+73-347+1461-32673+75-27905+75-1907+127-279+1475-811+127-277+75-304+127-34185+74-344+75-85400+75-302+99-402+125-2256+75-350+99-30294+75-692+99-1863+51-1556+126-395+126-307+494-44087+75-802+151-247+126-250+74-373+483-402+124-296+74-699+73-9028+74-320+99-296+75-32243+830-325+148-274+122-298+51-1177+75-376+75-324+76-68606+98-45286+76-303+96-322+75-25153+99-321+75-2332+100-1139+78-375+99-296+149-33741+51-357+151-295+150-13444+74-4257+75-11756+74-346+74-301+97-45226+76-332+102-309+101-687+77-709+51-353+102-281+125-24741+100-1768+125-674+98-2282+51-380+148-273+74-11769+99-9035+124-348+74-51581+51-307+102-308+76-2296+151-426+51-353+101-277+102-49558+101-282+76-2333+51-329+74-5147+75-1589+101-24802+51-355+126-273+150-228+101-2693+126-420+75-320+149-224+96-41366+75-729+74-1551+73-2308+75-344+75-301+12 [... cut ...] 2-12565+503-33493+510-13932+503-4316+494-26927+513-20173+511-742+99-3849+518-806+75-297+124-11418+495-324+124-3826+98-47011+75-321+74-33871+126-3835 
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
 
rtl_433_ESP(6): RAW (346829): +75-44632+51-375+99-1152+74-31224+73-730+97-16921+57301-372+99-298+51-29592+76-12559+99-3562+75-18494+76-726+75-36133+123-1868+124-13793+123-1241+102-348 
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
`

NorthernMan54, 2022-01-25:
Taking a look at this, this is not from the sample in `rtl_433_ESP/example/sample` it has a couple of different compiler definitions and I'm not seeing them in the log.  Also my analysis is showing that the code is working in your environment, and the issue is likely outside of the rtl_433_ESP code.  Hence the desire to simplify the code base temporarily to remove other factors.

With that said, looking at the raw data supplied, your environment looks very noisy with alot of RF noise being picked up as a potential signal.  Please keep in mind that the CC1101 is not as able to pick out signals as good as a rtl_sdr, and I find that it typically needs to be 50% closer to the sensor than a rtl_sdr.

In the data dump you supplied, only one signal was detected as being from an wh1080, everything else was interpreted as just noise.

This was a successful decode, but your callback is not being fired.
```
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
Fine Offset WH1080 data {80} ff a3 02 9a 50 00 01 00 00 0a
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
bitbuffer:: Number of rows: 1
[00] {87} ff 46 05 34 a0 00 02 00 00 15 40
Fine Offset WH1080 data {80} ff a3 02 9a 50 00 01 00 00 0a
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
bitbuffer:: Number of rows: 1
[00] {87} ff 46 05 34 a0 00 02 00 00 15 40
```

The last 3 lines of log are from this portion of the code

https://github.com/NorthernMan54/rtl_433_ESP/blob/08966ffa3e9d90bbe0b450ffa20a58d2da20c83a/src/rtl_433/pulse_demod.c#L51

Which is logging the details of a successful decode as I mentioned before.  I'm thinking the issue is potentially with your code around the library.

NorthernMan54, 2022-01-25:
This is from the rtl_433_ESP sample, with a single decoder enabled (TX141W)

In this log there are 2 signals received, the first is not decoded by the `TX141W` decoder and the second is decoded by the `TX141W` decoder.  I did make a minor change to code base, and added logging of the return code from a failed decode.  My first signal had a -1 return code, which is `DECODE_ABORT_LENGTH`

If you refresh your issue #23 branch it will include this.

```
N: ****** setup ******
rtl_433_ESP(6): CC1101 gpio receive pin: 4
rtl_433_ESP(6): CC1101 receive frequency: 433.920013
rtl_433_ESP(6): # of device(s) configured 1
rtl_433_ESP(6): ssizeof(r_device): 108
rtl_433_ESP(6): cfg->devices size: 108
rtl_433_ESP(6): # of device(s) enabled 1
Registering protocol [1] "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor"
rtl_433_ESP(6): CC1101 minumum rssi: -82
N: ****** setup complete ******
 
rtl_433_ESP(6): Signal length: 208159, Gap length: 47454, Signal RSSI: -58, train: 0, messageCount: 62, pulses: 150
rtl_433_ESP(7): demod(6) - LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor Status: -1
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor Status: -1
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor Status: -1
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor Status: -1
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 207839, Gap length: 30536758, Signal RSSI: -45, train: 1, messageCount: 63, pulses: 146
rtl_433_ESP(7): demod(6) - LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
rtl_433_ESP(6): data_output {"model":"LaCrosse-TX141Bv3","id":10,"channel":1,"battery_ok":1,"temperature_C":20.2,"test":"No","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-45,"duration":208002}
N: Received message : {"model":"LaCrosse-TX141Bv3","id":10,"channel":1,"battery_ok":1,"temperature_C":20.2,"test":"No","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-45,"duration":208002}
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
bitbuffer:: Number of rows: 4 
[00] {33} 0a 92 be a1 80 : 00001010 10010010 10111110 10100001 1
[01] {33} 0a 92 be a1 80 : 00001010 10010010 10111110 10100001 1
[02] {33} 0a 92 be a1 80 : 00001010 10010010 10111110 10100001 1
[03] {32} 0a 92 be a1    : 00001010 10010010 10111110 10100001 
rtl_433_ESP(6): # of messages decoded 1
```

garudaonekh, 2022-01-26:
Hi, Here's what I get after applying your latest update.

```
+101-1502+99-403+125-681+124-19386+76-2710+1183-651+75-471+101-272+127-254+124-729+151-730+200-20929+100-8284+76-11394+51-1489+126-1463+101-278+126-303+74-1126+75-1127+74-7037+126-397+75-24323+101-733+101-282+126-255+75-355+150-280+51-733+100-355+100-355+11030-722+74-350+101-746+51-1078+76-701+174-679+127-11438+75-406+74-376+51-1063+123-1527+74-324+125-12760+75-8667+51-1132+51-751+74-353+151-353+75-1034+76-23477+51-710+128-1157+125-2360+99-780+76-253+75-324+75-21021+76-308+76-28350+75-4986+75-1231+100-454+75-31787+120-272+126-1097+125-8636+150-748+99-18794+173-1957+125-323+75-724+102-278+125-503+101-301+125-275+74-376+51-708+100-279+175-201+51-406+122-325+76-15821+101-607+99-2288+98-345+75-348+125-10553+75-844+100-248+75-373+125-297+148-302+151-252+101-21517+101-1925+74-351+100-276+73-750+125-225+175-14529+100-775+174-253+124-1101+51-1459+125-12188+150-981+126-1444+76-776+151-249+125-322+125-378+375-731+100-11942+149-785+76-1236+76-354+125-609+148-961+151-1173+126-14949+75-326+74-299+176-254+152-251+53-120+101-305+101-17326+74-1134+75-732+102-1080+99-330+101-304+126-27030+75-3059+125-2651+74-324+200-227+128-707+73-402+99-302+124-24954+75-1533+126-277+100-676+196-252+127-652+75-20583+101-8284+75-3877+97-298+76-327+98-301+151-301+98-222+227-229+74-23597+73-825+123-1125+149-1010+75-351+74-350+124-727+125-229+127-27219+100-250+99-273+126-321+75-322+100-398+149-223+99-321+100-795+75-9006+151-11839+75-373+101-1454+51-11054+126-100+99-176+76-332+51-3858+101-21104+75-353+51-328+76-303+75-374+201-356+99-330+75-400+126-323+98-124+51-277+75-329+76-42022+73-426+76-278+123-402+148-455+125-301+125-326+100-228+51-77+101-732+127-10981+100-1981+75-251+100-11513+101-150+75-125+99-1107+76-1505+75-18234+125-2736+75-350+74-299+124-325+51-749+125-37575+101-352+98-9804+74-1925+75-733+76-1929+101-252+125-676+126-54410+99-1496+99-298+75-747+74-1529+99-278+75-347+75-18745+76-4262+73-748+73-347+125-10158+98-426+101-252+51-302+75-381+101-325+149-375+125-275+150-225+75-1570+150-352+149-32133+102-1579+125-845+74-300+75-723+125-1049+76-331+99-304+100-330+51-20438+152-1030+100-680+175-251+151-675+75-702+74-374+75-378+75-21449+76-301+76-1089+73-352+101-377+101-18095+125-1068+75-11411+147-406+101-304+125-279+98-325+100-251+123-349+75-274+125-348+123-1095+101-673+170-16568+102-2627+51-1137+72-332+125-283+98-13387+100-199+123-326+75-398+125-225+98-372+71-1156+75-330+151-734+77-700+75-17838+51-300+74-301+73-327+151-278+99-349+75-1130+488-893+99-349+98-1099+488-399+149-275+482-378+99-1097+75-401+75-277+101-249+124-8989+75-733+126-706+492-1129+101-301+126-276+149-274+123-1052+1059-446+76-303+75-40112+76-3440+75-755+51-354+76-707+496-379+75-329+100-281+150-279+494-403+126-10107+488-2394+1468-452+150-227+75-329+51-352+487-151+76-579+153-254+101-304+52-328+1458-20694+126-709+75-680+76-9814+495-836+1486-1538+51-1163+100-305+76-9009+75-1863+99-322+123-843+1463-1412+1463-7599+51-680+1473-3922+76-357+127-277+1485-422+1465-149+74-149+74-324+125-276+96-22820+99-371+1485-1471+99-347+99-298+75-1173+1465-379+124-649+124-8265+1464-4669+489-807+51-829+75-200+100-299+73-25039+485-3578+1475-1132+495-475+75-1436+125-254+472-452+100-326+101-227+445-11082+457-425+101-1132+51-298+474-2013+51-402+489-24513+51-275+76-1928+1491-9873+51-735+101-272+75-10567+123-302+145-546+74-1231+1470-449+100-273+124-273+100-247+100-373+75-18153+492-383+71-395+74-276+100-350+467-352+76-29897+75-1166+488-1541+1451-449+75-246+751-378+100-325+100-10546+126-277+472-430+100-250+99-1519+75-1119+74-325+150-206+1473-11131+482-1608+101-252+150-300+1470-380+51-803+1461-475+1461-277+126-251+125-11745+497-451+1483-350+75-705+104-294+1466-9930+1448-328+99-728+1477-777+76-328+126-1839+1468-424+75-324+1467-15071+75-729+1486-377+1465-51+1468-1554+74-1084+149-299+1468-37102+51-1132+76-327+488-173+1493-1025+102-1009+1450-902+1486-754+127-277+127-37825+75-1927+494-2355+1484-372+125-74+1465-1003+75-355+100-397+125-678+176-378+226-33478+76-402+99-751+75-9423+98-19624+749-823+75-324+75-2285+100-404+51-355+224-602+148-12850+125-401+101-275+124-300+100-682+100-325+448-350+151-251+100-301+125-48958+73-299+76-1648+75-780+198-454+124-376+75-300+99-278+73-11422+76-727+126-278+100-3456+125-747+51-18758+101-328+76-425+73-776+102-348+100-749+101-1057+126-326+150-330+125-299+124-277+122-15015+74-699+100-373+125-707+75-327+75-329+125-424+99-1496+150-13085+75-5063+76-355+75-662+101-126+51-10487+99-381+101-707+76-282+125-450+75-274+124-722+101-783+100-7074+75-3395+74-498+75-750+75-322+125-1471+99-101+74-624+51-447+125-721+124-24128+123-583+76-377+75-1127+153-776+75-397+150-1451+101-424+399-100+75-275+125-32155+75-2760+74-2296+101-283+152-227+153-1515+125-378+77-126+51-100+76-45324+76-348+99-298+75-298+74-738+97-306+126-1106+75-14342+75-2385+51-756+125-202+73-374+200-249+75-675+75-9405+100-729+151-428+76-2988+98-10601+698-10284+75-1248+75-1501+75-350+74-404+76-327+101-252+99-326+102-806+100-304+149-253+98-277+74-353+76-19078+101-304+150-280+125-322+74-749+101-1252+124-722+126-17311+74-326+151-1058+101-327+51-307+149-327+75-732+100-41820+74-329+151-250+148-349+75-731+75-752+99-1123+74-15485+73-3465+148-254+99-328+148-249+99-15421+76-1056+127-300+75-1054+74-423+76-278+151-655+100-350+99-426+100-326+152-15717+100-745+75-350+126-778+101-303+101-279+76-1558+74-404+101-22511+129-432+101-277+72-1104+76-357+101-304+76-328+101-1497+124-15054+75-1980+99-1855+101-305+152-659+100-305+100-682+102-329+100-1077+100-41299+72-348+125-649+96-1525+101-398+101-280+434- [... cut ...] 446-964+1479-986+1461-980+1466-971+1465-974+494-959+1484-958+1487-967+1466-967+1470-997+1469-969+1470-950+1469-988+479-982+1462-972+489-976+1474-960+1478-983+1460-969+1464-991+1463-963+1480-965+1470-981+1449-987+1454-991+1461-967+1467-983+1466-976+1464-966+1466-1002+1448-987+1455-986+1463-961+1486-952+1484-971+1467-972+493-986+1461-960+490-976+1477-967+494-954+1478-985+1462-980+1465-966+489-982+475-990+486-975 
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station

```

NorthernMan54, 2022-01-26:
Is this the sample code, with the same compiler definitions ?  It should be logging why the decode failed.

garudaonekh, 2022-01-26:
I use your latest branch#23. However, I enable this to show raw data. -DRAW_SIGNAL_DEBUG=true

NorthernMan54, 2022-01-26:
Something isn't right with your build then, as it should be logging lines that start with this for each signal

`pulse_demod_pwm()`

garudaonekh, 2022-01-26:
Sorry, I forget to enable debug flag.

Here, it's 
```

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
N:  
N: ****** setup ******
rtl_433_ESP(6): Pre initReceiver: 349036
rtl_433_ESP(6): sizeof(*cfg->demod) 16
rtl_433_ESP(6): sizeof(*cfg->demod) 16
rtl_433_ESP(6): size of bitbuffer: 6604
rtl_433_ESP(6): size of pulse_data: 9620
Registering protocol [1] "Fine Offset Electronics WH1080/WH3080 Weather Station"
rtl_433_ESP(6): Post initReceiver: 348264
N: ****** setup complete ******
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 519
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 268
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 334
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 322
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 378
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 277
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 267
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 333
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 386
rtl_433_ESP(6): Post run_ook_demods memory 338576
 
rtl_433_ESP(6): Pre run_ook_demods: 338576
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
rtl_433_ESP(6): Signal processing time: 306
rtl_433_ESP(6): Post run_ook_demods memory 338576
```

NorthernMan54, 2022-01-26:
Its this flag

  '-DRTL_DEBUG=4'           ; rtl_433 verbose mode

garudaonekh, 2022-01-26:
SOrry, i forget to set this flag.

Here, it's
```
mode:DIO, clock dC!�+��0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
N:  
N: ****** setup ******
rtl_433_ESP(6): CC1101 gpio receive pin: 27
rtl_433_ESP(6): CC1101 receive frequency: 433.920013
rtl_433_ESP(6): # of device(s) configured 1
rtl_433_ESP(6): ssizeof(r_device): 108
rtl_433_ESP(6): cfg->devices size: 108
rtl_433_ESP(6): # of device(s) enabled 1
Registering protocol [1] "Fine Offset Electronics WH1080/WH3080 Weather Station"
rtl_433_ESP(6): CC1101 minumum rssi: -82
N: ****** setup complete ******
 
rtl_433_ESP(6): Signal length: 208043, Gap length: 6594936, Signal RSSI: -59, train: 0, messageCount: 0, pulses: 69
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 453385, Gap length: 22487025, Signal RSSI: -52, train: 1, messageCount: 1, pulses: 171
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 1420094, Gap length: 3019715, Signal RSSI: -56, train: 0, messageCount: 2, pulses: 278
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 378167, Gap length: 20410703, Signal RSSI: -64, train: 1, messageCount: 3, pulses: 140
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 242568, Gap length: 22317537, Signal RSSI: -54, train: 0, messageCount: 4, pulses: 85
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 212207, Gap length: 25061088, Signal RSSI: -65, train: 1, messageCount: 5, pulses: 69
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 456576, Gap length: 22481978, Signal RSSI: -70, train: 0, messageCount: 6, pulses: 171
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 381823, Gap length: 24848690, Signal RSSI: -58, train: 1, messageCount: 7, pulses: 140
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 1444265, Gap length: 109271, Signal RSSI: -58, train: 0, messageCount: 8, pulses: 278
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 248057, Gap length: 20757361, Signal RSSI: -63, train: 1, messageCount: 9, pulses: 85
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 212711, Gap length: 25058307, Signal RSSI: -51, train: 0, messageCount: 10, pulses: 69
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 1419234, Gap length: 19076541, Signal RSSI: -57, train: 1, messageCount: 11, pulses: 278
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
Fine Offset WH1080 data {80} 7f ff ff ff ff ff ff ff ff ff 
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -4
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 459923, Gap length: 1985091, Signal RSSI: -70, train: 0, messageCount: 12, pulses: 172
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 381520, Gap length: 24845456, Signal RSSI: -64, train: 1, messageCount: 13, pulses: 140
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 471704, Gap length: 21003481, Signal RSSI: -56, train: 0, messageCount: 76, pulses: 170
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
```

NorthernMan54, 2022-01-26:
How far away is your weather station ?  I’m not seeing any successful signal decodes.   The -1 on this line is the reason code for the signal decode failure.  Earlier you where getting signal decodes.

> On Jan 25, 2022, at 10:32 PM, GarudaOne ***@***.***> wrote:
> 
> pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1

garudaonekh, 2022-01-26:
It was around 10 meters(outside the room). Now I put it inside the room and 1 meter away.

Here is what I get:

```
rtl_433_ESP(6): Signal length: 4699777, Gap length: 26123452, Signal RSSI: -48, train: 1, messageCount: 113, pulses: 87
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
Fine Offset WH1080 data {80} ff d1 a1 5f 9e 80 00 00 00 47 
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -3
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 514118, Gap length: 5744986, Signal RSSI: -52, train: 0, messageCount: 114, pulses: 172
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 1432655, Gap length: 27568860, Signal RSSI: -64, train: 1, messageCount: 115, pulses: 283
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
Fine Offset WH1080 data {80} e7 ff ff ff ff ff ff ff ff ff 
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -4
rtl_433_ESP(6): # of messages decoded 0
 
rtl_433_ESP(6): Signal length: 265468, Gap length: 18483290, Signal RSSI: -58, train: 0, messageCount: 116, pulses: 85
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
rtl_433_ESP(6): # of messages decoded 0
```

garudaonekh, 2022-01-26:
Now, it looks like i got the decoded message:

rtl_433_ESP(6): Signal length: 485045, Gap length: 47747484, Signal RSSI: -58, train: 1, messageCount: 123, pulses: 173
rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station Status: -1
Fine Offset WH1080 data {80} ff a3 42 b9 3e 00 00 00 00 8e 
rtl_433_ESP(6): data_output {"model":"Fineoffset-WHx080","subtype":0,"id":52,"battery_ok":1,"temperature_C":29.7,"humidity":62,"wind_dir_deg":315,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":0,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-58,"duration":485218}
N: Received message : {"model":"Fineoffset-WHx080","subtype":0,"id":52,"battery_ok":1,"temperature_C":29.7,"humidity":62,"wind_dir_deg":315,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":0,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-58,"duration":485218}
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
bitbuffer:: Number of rows: 1 
[00] {87} ff 46 85 72 7c 00 00 00 01 1d 2a 
rtl_433_ESP(6): # of messages decoded 1

NorthernMan54, 2022-01-26:
Success, the call back fired

`N: Received message : {"model":"Fineoffset-WHx080","subtype":0,"id":52,"battery_ok":1,"temperature_C":29.7,"humidity":62,"wind_dir_deg":315,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":0,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-58,"duration":485218}`

garudaonekh, 2022-01-26:
SO it mean the master branch is ok? It's just my CC1101 receiver?

NorthernMan54, 2022-01-26:
Yes, no difference in the receiver / decoder logic in the master.  Just a simplified use case for debugging

garudaonekh, 2022-01-26:
I don't why I can get only 1 Received message after nearly 1 hour. While using the RTL-SDR with RTL_433(rtl_433) i get decoded message after 1 or 2 minutes.

NorthernMan54, 2022-01-26:
Your likely seeing the limited range of the cc1101 receiver.

garudaonekh, 2022-01-30:
I do not think so. I put it 1 meter away from each other. I try RTL-SDR and it keeps getting messages every 1 minute

NorthernMan54, 2022-02-01:
I'm wondering if the level of RF Noise is higher in your environment than mine and that is causing the receiver to be confused

The signal receiver works by watching the RSSI level being reported by the CC1101 and if it is greater than MINRSSI ( https://github.com/NorthernMan54/rtl_433_ESP/blob/77707f473414e05c0af26d1c61d1a2b0fa7735a7/src/rtl_433_ESP.h#L31 ) it starts recording the signal.  And once the RSSI level drops below MINRSSI it thinks the signal is complete and sends it for decoding.

If the level of noise is high in your environment, then maybe MINRSSI needs to be lowered slightly.  I did notice that in the raw signal data there is a lot of large values which indicate something odd is occurring.  With my devices the signal length is pretty consistent, and with yours they jump around alot.

Also are you doing any additional work as part of the Arduino loop in your code ?  If your doing something intensive it could impact start or end of signal detection.

garudaonekh, 2022-02-01:
My code does not have anything else to run. I use chip from Ebytes.com.

NorthernMan54, 2022-02-01:
If you turn the debug flags on again, and look at this line in particular it should give you an indication of the signal quality etc

`rtl_433_ESP(6): Signal length: 208159, Gap length: 47454, Signal RSSI: -58, train: 0, messageCount: 62, pulses: 150`

If it is cleanly receiving signals, the length, RSSI and pulses should be consistent.  You could try increasing MINRSSI to be closer to the Signal RSSI to see what impact that has.  When I picked -82, I just found the noise floor in my setup.  Yours may be different.  You can call getStatus to have the device report the noise level rssi when it is not receiving signals

See my sample from here - https://github.com/NorthernMan54/rtl_433_ESP/issues/22#issuecomment-1021721026

garudaonekh, 2022-02-01:
> #22 (comment)

> SOrry, i forget to set this flag.
>
> Here, it's
>
> ```
> mode:DIO, clock dC!�+��0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:10124
> load:0x40080400,len:5828
> entry 0x400806a8
> N:
> N: ****** [...]

Based on this pattern, which MINRSSI should I pick?

NorthernMan54, 2022-02-02:
Lowest I saw was a -70, so you could try -75

garudaonekh, 2022-04-19:
> Lowest I saw was a -70, so you could try -75

Sorry, I just have time to continue on this weather station project. I change according to your suggestion -75. But it's still the same. It looks like it can only successfully parse the data once. My wild guess is that the sender maybe very low power and the signal is very weak. Most of the time, it's Fine Offset WH1080 data {80} 7f ff ff ff ff ff ff ff ff ff

garudaonekh, 2023-02-15:
Hi, just to update on this.
I try using RXB7 with this Weather station and the range improve a lot. I think it can parse 90% of what RTL_SDR do.

NorthernMan54, 2023-02-15:
Could you share what your setup is ?

garudaonekh, 2023-02-16:
> Could you share what your setup is ?

Sorry, RXB7 is not based on this library.

closed 2023-02-16 by garudaonekh

garudaonekh, 2023-02-28:
I tries to use CC1101 based on this repo(https://github.com/LSatan/SmartRC-CC1101-Driver-Lib) to decode WH1080. Its range is nearly the same as superheterodyne and success rate is good. I don't know if this library use the same method(RSSI) as yours or not.

Your rf receiver code was also based on this SmartRC-CC1101 before you switched to RadioLib, correct me if I am wrong.