NorthernMan54/rtl_433_ESP #23: Get 2 sets of data

davoau, 2022-01-18:
Hi, I am getting 2 sets of data coming from the rtl_433_Callback function - or the function is being called twice.
This has only started happening since I updated to the new version of rtl_433_ESP.

The following modifications is all I have done in relation to rtl_433_ESP library/code:

rtl_433_ESP\include\rtl_433_devices.h  - line 120
#define DEVICES           \
  DECL(fineoffset_wh1080) \
  /* Add new personal decoders here. */
  #define NUMOFDEVICES 1
#endif

rtl_433_ESP\src\rtl_433_ESP.cpp  - line 206
#else
  memcpy(&cfg->devices[0], &fineoffset_wh1080, sizeof(r_device));
#endif

rtl_433_ESP\src\rtl_433\r_api.c  - line 852
/*
  data_append(data,
              "protocol", "", DATA_STRING, r_dev->name,
              "rssi", "RSSI", DATA_INT, cfg->demod->pulse_data->signalRssi,
              "duration", "", DATA_INT, cfg->demod->pulse_data->signalDuration,
              NULL);
*/

void rtl_433_Callback(char* message) {
  Log.notice(F("Received message : %s" CR), message);

  DynamicJsonBuffer jsonBuffer2(JSON_MSG_BUFFER);
  JsonObject& RFrtl_433_ESPdata = jsonBuffer2.parseObject(message);

  String mqttSubTopic;
  String model = RFrtl_433_ESPdata["model"];
  String id = RFrtl_433_ESPdata["id"];
  if (model != 0) {
    mqttSubTopic += "/" + model;
    if (id != 0) {
      mqttSubTopic += "/" + id;
    }
  }

  RFrtl_433_ESPdata["espid"] = espID;

  char JSONmessageBuffer[RFrtl_433_ESPdata.measureLength() + 1];
  RFrtl_433_ESPdata.printTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));

  mqttPublish(mqttSubTopic, JSONmessageBuffer );
}

void loop()
{
  rf.loop();
}

I also have BME280 sensor connected to the ESP, which is run in another loop on cpu core 1, and rf.loop is run on core 2 (default for ESP32 loop function).

It doesnt happen everytime, probably 2/3 of the time.
I also have SDR_RTL_433 running on a rpi0, and I only get the one set of data every 136 secs.

The following output shows both the weather station data and BME280 - BME280 sensor is read every 60secs, and the weather station sends out a message every 136 secs.

10:05:52 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.8,"humidity":69,"wind_dir_deg":315,"wind_avg_km_h":0,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","espid":"2115"}
10:05:52 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.8,"humidity":69,"wind_dir_deg":315,"wind_avg_km_h":0,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","espid":"2115"}
10:06:20 - rtl_433_ESP/SENSORS {"espid":"2115","temperature":30.84,"humidity":75.58,"abs_humidity":24.00,"dewpoint":26.02,"heatindex":38.88,"pressure":1007.50,"altitude":21.64,"sealevel":1008.41}
10:07:20 - rtl_433_ESP/SENSORS {"espid":"2115","temperature":30.84,"humidity":75.58,"abs_humidity":24.00,"dewpoint":26.02,"heatindex":38.88,"pressure":1007.50,"altitude":21.64,"sealevel":1008.41}
10:07:28 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.7,"humidity":69,"wind_dir_deg":315,"wind_avg_km_h":1.224,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","espid":"2115"}
10:08:20 - rtl_433_ESP/SENSORS {"espid":"2115","temperature":30.84,"humidity":75.58,"abs_humidity":24.00,"dewpoint":26.02,"heatindex":38.88,"pressure":1007.50,"altitude":21.64,"sealevel":1008.41}
10:09:04 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.8,"humidity":70,"wind_dir_deg":315,"wind_avg_km_h":1.224,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","espid":"2115"}
10:09:04 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.8,"humidity":70,"wind_dir_deg":315,"wind_avg_km_h":1.224,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","espid":"2115"}
10:09:20 - rtl_433_ESP/SENSORS {"espid":"2115","temperature":30.84,"humidity":75.58,"abs_humidity":24.00,"dewpoint":26.02,"heatindex":38.88,"pressure":1007.50,"altitude":21.64,"sealevel":1008.41}

The function that publishes the MQTT message is used for both rtl_433_ESP and BME280, so I dont think its from there or else it would show the same problem for BME280.

Here is the data from arduino terminal:

10:18:45.700 -> rtl_433_ESP(6): Signal length: 471335, Gap length: 47771067, Signal RSSI: -74, train: 0, messageCount: 4, pulses: 189
10:18:45.747 -> rtl_433_ESP(6): RAW (471517): +531-963+516-951+498-960+509-950+503-943+545-941+532-947+499-963+1493-931+551-939+1515-900+527-957+1494-932+1498-964+528-923+1518-929+520-952+499-942+1521-925+1505-950+1496-942+504-939+1509-937+532-938+542-904+1531-959+1453-975+478-1010+483-956+1474-988+1477-938+1481-977+505-966+1468-971+1472-954+1502-957+507-941+541-930+524-946+1485-932+1515-933+1509-936+1503-993+1443-988+1449-966+1478-965+532-932+1488-949+1480-979+1473-954+1495-957+1485-918+1532-928+529-958+517-958+1486-945+1483-960+1489-952+1508-925+524-950+1497-941+1478-968+507-966+1490-957+527-920+1521-914+524-917+1536-863+1605-923+494-951+518-953+1482-966+1488-952+1479-943+1513-944+1547-903+521-932+1488-915+1535-937+529-960+1474-954+520-956+1482-964+536-926+504-921+548-942+1509-6437+1484-822+548-24756+1483-398+555-939+519-924+547-937+540-896+549-965+530-924+529-936+545-915+1558-855+556-950+1465-1003+449-1001+1455-964+1480-968+503-977+1491-924+499-988+473-1021+1436-982+1489-950+1504-918+517-958+1478-968+556-903+505-956+1493-949+1492-961+551-911+507-963+1492-944+1496-940+1484-968+510-931+1516-946+1487-940+1506-960+541-914+533-941+509-948+1529-909+1528-902+1526-902+1528-919+1534-927+1481-988+1464-982+511-935+1502-928+1527-938+1513-905+1538-910+1511-933+1508-945+520-929+548-919+1519-926+1526-930+1543-883+1530-932+507-955+1491-957+1466-958+537-935+1501-938+525-936+1508-938+511-955+1481-964+1494-942+528-914+554-949+1488-942+1495-941+1503-945+1511-926+1505-947+514-936+1501-925+1522-926+551-941+1488-964+520-911+1536-914+529-962+541-913+524-933+1511-985+148-195+199-2320+0-3267+0-818+0-35000+100-299+74-376+0-378+75-357+75-381+0-811+0-1214 
10:18:45.887 -> rtl_433_ESP(6): Pre run_ook_demods: 227464
10:18:45.887 -> rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
10:18:45.887 -> Fine Offset WH1080 data {80} ff a9 62 cc 47 01 03 09 53 04 
10:18:45.887 -> rtl_433_ESP(6): data_output {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":31.6,"humidity":71,"wind_dir_deg":90,"wind_avg_km_h":1.224,"wind_max_km_h":3.672,"rain_mm":716.10004,"mic":"CRC"}
10:18:45.887 -> N: Received message : {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":31.6,"humidity":71,"wind_dir_deg":90,"wind_avg_km_h":1.224,"wind_max_km_h":3.672,"rain_mm":716.10004,"mic":"CRC"}
10:18:45.934 -> 
10:18:45.934 -> rtl_433_Callback running on core 1
10:18:45.934 -> mqttSubTopic: /Fineoffset-WHx080/150
10:18:45.934 -> 
10:18:45.934 -> mqttPublish running on core 1
10:18:45.934 -> mqttTopic:   rtl_433_ESP/Fineoffset-WHx080/150
10:18:45.934 -> jsonMessage: {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":31.6,"humidity":71,"wind_dir_deg":90,"wind_avg_km_h":1.224,"wind_max_km_h":3.672,"rain_mm":716.10004,"mic":"CRC","espid":"2115"}
10:18:45.934 -> pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
10:18:45.981 -> bitbuffer:: Number of rows: 1 
10:18:45.981 -> [00] {87} ff 52 c5 98 8e 02 06 12 a6 09 5c 
10:18:45.981 -> Fine Offset WH1080 data {80} 7f a9 62 cc 47 01 03 09 53 04 
10:18:45.981 -> rtl_433_ESP(6): # of messages decoded 1
10:18:45.981 -> rtl_433_ESP(6): Signal processing time: 251183
10:18:45.981 -> rtl_433_ESP(6): Post run_ook_demods memory 227464

When I connect the ESP32 to my computer, and monitor on the terminal it stops sending the RF data twice, and the blue LED on the ESP flickers continually - never goes out. If not connected to PC there is a clear flashing of the blue LED.

Anyways any help would be great.
Thanks

NorthernMan54, 2022-01-18:
This is the master branch and not Carrier Sense ?

Looking at the last bit of log, my working theory around what is occurring is that the receiver portion of the code is thinking that the signal is complete and sending it for processing, but then the signal continues and a second signal is captured and sent for processing.

The receiver watches the signal strength ( rssi ), and if it drops for a period of time the receiver considers signal is complete.  And these devices typically repeat the same message numerous times during a single transmission so this is a potential cause.  Looking at number of pulses and signal duration can assist in determining if this is the cause.

FYI - this line here provides details about the signal to assist in debugging this - rtl_433_ESP\src\rtl_433\r_api.c - line 852
You can look at the number of pulses received and signal duration and compare it to rtl_433 to prove the theory.

For your issue with the LED continually flickering ( signal being received ), this is likely caused by noise being received by the receiver.  The receiver code is coded to consider rssi over -82 ( MINRSSI ) as a valid signal.  As this is a potential issue, I'm working on new receiver logic using CC1101 Carrier Sense to resolve this issue.  This is in the Carrier Sense branch, and just needs to have the examples updated before publishing.

Question for you, I have been using the OpenMQTTGateway package to provide the production framework for this, and it also supports have a CC1101 and BME280 on the same device, have you looked at it ?

davoau, 2022-01-18:
Hi, I tried the openmqttgateway a few months back, and it bricked my ESP32, so wasn't game to try again.
I also want to use this for other sensors - UV, lux etc and stick outside in a stevenson screen, and run from battery that is charged via solar, so will be monitoring the battery too and sending to mqtt broker.
The PCB I am using is below - everything is working well except the double message from RF.

[image]
.

I removed the extra data in  rtl_433_ESP\src\rtl_433\r_api.c - line 852 as mqtt wouldn't send the message with the extra data. I have tried #define MQTT_MAX_PACKET_SIZE 512  but that didn't work even if I bumped it up to 1024, its as if it wouldn't change. However I have since read recently that other people were having the same problem and that if I set client.setBufferSize(512) it might fix the problem - I will give it a go and maybe be able to see rest of message and get back to you.

I didn't realise the weather station would send more than one message.. but now that you have pointed it out, it makes sense. 

Anyways thanks for your help/info..

davoau, 2022-01-18:
So I set the mqtt buffer size using set client.setBufferSize(512) and it worked. He is the output, interestingly it is not doing 2 x messages since I added back to rtl_433_ESP\src\rtl_433\r_api.c - line 852 so that the extra data gets sent.

15:27:29 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.0,"humidity":66,"wind_dir_deg":113,"wind_avg_km_h":3.672,"wind_max_km_h":7.344,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-60,"duration":464037,"espid":"2115"}

15:30:41 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.5,"humidity":67,"wind_dir_deg":338,"wind_avg_km_h":0,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-60,"duration":452733,"espid":"2115"}

15:33:04 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.7,"humidity":68,"wind_dir_deg":338,"wind_avg_km_h":0,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-76,"duration":248528,"espid":"2115"}

15:33:53 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.9,"humidity":66,"wind_dir_deg":338,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-70,"duration":460084,"espid":"2115"}

15:38:41 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":33.4,"humidity":62,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-58,"duration":454641,"espid":"2115"}

15:40:17 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":33.4,"humidity":62,"wind_dir_deg":338,"wind_avg_km_h":1.224,"wind_max_km_h":3.672,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-70,"duration":436907,"espid":"2115"}

15:41:53 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":33.3,"humidity":63,"wind_dir_deg":338,"wind_avg_km_h":1.224,"wind_max_km_h":2.448,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-70,"duration":436269,"espid":"2115"}

15:43:29 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.1,"humidity":66,"wind_dir_deg":113,"wind_avg_km_h":6.12,"wind_max_km_h":7.344,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-70,"duration":451817,"espid":"2115"}

15:45:05 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.5,"humidity":66,"wind_dir_deg":68,"wind_avg_km_h":1.224,"wind_max_km_h":3.672,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-62,"duration":448624,"espid":"2115"}

davoau, 2022-01-18:
I notice after 20 minutes it was dropping messages.. so I moved the esp32 over a bit, and I was getting the 2 messages again. At that stage it was approx 15M from the weather station. I have moved it to less than 10M from the weather station think it would get a better signal, but the problem is the same.

20:47:29 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.1,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-54,"duration":464109,"espid":"2115"}
20:47:29 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.1,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-54,"duration":464109,"espid":"2115"}
20:49:05 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.0,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-64,"duration":470426,"espid":"2115"}
20:49:05 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.0,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-64,"duration":470426,"espid":"2115"}
20:50:41 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.1,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-52,"duration":464614,"espid":"2115"}
20:50:41 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.1,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-52,"duration":464614,"espid":"2115"}
20:52:17 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.1,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-64,"duration":467482,"espid":"2115"}
20:52:17 - rtl_433_ESP/Fineoffset-WHx080/150 {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":27.1,"humidity":81,"wind_dir_deg":45,"wind_avg_km_h":0,"wind_max_km_h":0,"rain_mm":716.10004,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-64,"duration":467482,"espid":"2115"}

I have been using both the stubby antenna that came with the CC1101 module and a 173mm antenna which I replaced a pig tail antenna with. The 173mm one has the best range, but gets the most double signals. The stubby antenna has very limited range, and drops around 1/2 of massages, but also has less of the double messages.

1technophile, 2022-01-18:
> Hi, I tried the openmqttgateway a few months back, and it bricked my ESP32, so wasn't game to try again.

Feel free to report the conditions of your test, it can help the community to progress or the community can help you identify what was going wrong.

davoau, 2022-01-18:
> Hi 1technophile. I will revisit the ESP32 with OMG on it, and see if it can be made to work - it will no longer connect via USB com-port, so can not install anything from arduino IDE or platformIO. I used the web-install method, this is when I first started playing around with the ESP so had no idea what I was doing. Anyways thanks for reaching out.

davoau, 2022-01-18:
Hi, the master I am using is either dated 21/12 or 24/12. I just noticed an update on here and downloaded it.

I dont have the board in an enclosure so I think the BME280 isnt effected by heat. I will probably put board in a stevensons screen - made of wood so I am thinking it will probably be ok. I have read that temperature and humidity readings should be taken 1.25M above ground level, and away from any heat radiation source - concrete walls, paving etc..  my weather station is 5M above ground so probably isn't 100% correct anyways. The ESP32 will also be put into deep sleep between sensor/weather station readings so that will keep temp down If I do put in a case.

As for the board itself, it works ok, however I have got a new version on the way as I am not happy with the voltage regulator i used and so will use another one, which of course has different pin layout. I have also decided to not have the battery built into board so it can sit behind the board in a smaller case if I decide to put it in one in the future. I have made the new board with ADC pins exposed so can be used for other projects as well.

I can sent you the gerber/schematics file of the PCB if you are interested (or anyone else for that matter) once I have tested and am happy it works as it should. They cost US$5.90 for 5 boards including freight from JLCPCB in HK, and I have within 3 weeks of ordering. Maybe can be had for cheaper in USA or EU if they do small run PCB manufacturing there, or other HK/China manufacturers.

NorthernMan54, 2022-01-19:
So I attempted to recreate your issue in the branch 'issue-#23', using the example code.  And I could not recreate the issue.  Could you take that branch and attempt to recreate on a test device?

1 - To use the example, you need to clone to repo
2 - Open a platformio workspace in the directory rtl_433_ESP/example/sample
3 - As I was editing the library locally I told platformio to use the library from the clone'd repo ie
```
cd ~/.platformio/lib/
ln -s ~/Code/rtl_433_ESP rtl_433_ESP
```

I have included in the sample a couple of compiler directives to provide more details about the received signal.

This is what I receive

In my test environment I have 2 LaCrosse TX141 devices that are pretty chatty

The first a TX141-Bv2 sends the same data with a tiny gap, so the receiver thinks its two signals and processes each separately.

The second a TX141W ( a temperature/humdity/wind sensor ) sends a single signal, with 2 parts, the first part is temperature/humidity data and the second part is the wind data.

```
rtl_433_ESP(6): Signal length: 229636, Gap length: 21631378, Signal RSSI: -46, train: 1, messageCount: 49, pulses: 146
rtl_433_ESP(7): demod(6) - LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
rtl_433_ESP(6): data_output {"model":"LaCrosse-TX141Bv3","id":10,"channel":1,"battery_ok":1,"temperature_C":20.5,"test":"No","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-46,"duration":229795}
N: Received message : {"model":"LaCrosse-TX141Bv3","id":10,"channel":1,"battery_ok":1,"temperature_C":20.5,"test":"No","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-46,"duration":229795}
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
bitbuffer:: Number of rows: 4 
[00] {33} 0a 92 c1 b4 00 : 00001010 10010010 11000001 10110100 0
[01] {33} 0a 92 c1 b4 00 : 00001010 10010010 11000001 10110100 0
[02] {33} 0a 92 c1 b4 00 : 00001010 10010010 11000001 10110100 0
[03] {32} 0a 92 c1 b4    : 00001010 10010010 11000001 10110100 
rtl_433_ESP(6): # of messages decoded 1
 
rtl_433_ESP(6): Signal length: 228123, Gap length: 26585, Signal RSSI: -36, train: 0, messageCount: 50, pulses: 146
rtl_433_ESP(7): demod(6) - LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
rtl_433_ESP(6): data_output {"model":"LaCrosse-TX141Bv3","id":10,"channel":1,"battery_ok":1,"temperature_C":20.5,"test":"No","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-36,"duration":228271}
N: Received message : {"model":"LaCrosse-TX141Bv3","id":10,"channel":1,"battery_ok":1,"temperature_C":20.5,"test":"No","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-36,"duration":228271}
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
bitbuffer:: Number of rows: 4 
[00] {33} 0a 92 c1 b4 00 : 00001010 10010010 11000001 10110100 0
[01] {33} 0a 92 c1 b4 00 : 00001010 10010010 11000001 10110100 0
[02] {33} 0a 92 c1 b4 00 : 00001010 10010010 11000001 10110100 0
[03] {32} 0a 92 c1 b4    : 00001010 10010010 11000001 10110100 
rtl_433_ESP(6): # of messages decoded 1
 
rtl_433_ESP(6): Signal length: 718819, Gap length: 8673405, Signal RSSI: -52, train: 1, messageCount: 51, pulses: 825
rtl_433_ESP(7): demod(6) - LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
rtl_433_ESP(6): data_output {"model":"LaCrosse-TX141W","id":391932,"channel":0,"battery_ok":1,"temperature_C":16.1,"humidity":57,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-52,"duration":718971}
N: Received message : {"model":"LaCrosse-TX141W","id":391932,"channel":0,"battery_ok":1,"temperature_C":16.1,"humidity":57,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-52,"duration":718971}
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
bitbuffer:: Number of rows: 7 
[00] {65} 0d fa fc 01 29 50 39 1a 00 
[01] {65} 0d fa fc 01 29 50 39 1a 00 
[02] {65} 0d fa fc 01 29 50 39 1a 00 
[03] {65} 0d fa fc 01 29 50 39 1a 00 
[04] {65} 0d fa fc 01 29 50 39 1a 00 
[05] {65} 0d fa fc 01 29 50 39 1a 80 
[06] {65} 0d fa fc 01 29 50 39 1a 00 
rtl_433_ESP(6): data_output {"model":"LaCrosse-TX141W","id":391932,"channel":0,"battery_ok":1,"wind_avg_km_h":0,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-52,"duration":718971}
N: Received message : {"model":"LaCrosse-TX141W","id":391932,"channel":0,"battery_ok":1,"wind_avg_km_h":0,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-52,"duration":718971}
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
bitbuffer:: Number of rows: 6 
[00] {65} 0d fa fc 02 00 00 00 41 00 
[01] {65} 0d fa fc 02 00 00 00 41 00 
[02] {65} 0d fa fc 02 00 00 00 41 00 
[03] {65} 0d fa fc 02 00 00 00 41 00 
[04] {28} 0d fa fc 00                : 00001101 11111010 11111100 0000
[05] {35} 40 00 00 08 20             : 01000000 00000000 00000000 00001000 001
rtl_433_ESP(6): # of messages decoded 2
 
[... 17 lines cut ...]
rtl_433_ESP(7): demod(6) - LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
[... cut ...]
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
bitbuffer:: Number of rows: 7 
[00] {65} 0d fa fc 02 00 00 00 41 00 
[01] {65} 0d fa fc 02 00 00 00 41 00 
[02] {30} 0d fa fc 00                : 00001101 11111010 11111100 000000
[03] { 1} 80                         : 1
[04] {33} 00 00 00 41 00             : 00000000 00000000 00000000 01000001 0
[05] {65} 0d fa fc 02 00 00 00 41 00 
[06] {64} 0d fa fc 02 00 00 00 41 
rtl_433_ESP(6): # of messages decoded 2
```

davoau, 2022-01-20:
Hi,  I have cloned the new repo issue 23, I compiled your simple example, and the only changed I have made are as follows:
platformio.ini

`
monitor_speed = 115200
upload_port = COM7
monitor_port = COM7
upload_speed = 115200

#define DEVICES        \
  DECL(fineoffset_wh1080) \
  /* Add new personal decoders here. */
  #define NUMOFDEVICES 1
#endif

#else
  memcpy(&cfg->devices[0], &fineoffset_wh1080, sizeof(r_device));
#endif
`

Here is the output from terminal monitor - I used with both CC1101 with stubby antenna it came with and the CC1101 with the 173mm antenna I soldered to board:

STUBBY ANTENNA:
`14:03:29.060 -> ets Jun  8 2016 00:22:57
14:03:29.060 -> 
14:03:29.060 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:03:29.060 -> configsip: 0, SPIWP:0xee
14:03:29.060 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:03:29.060 -> mode:DIO, clock div:2
14:03:29.060 -> load:0x3fff0018,len:4
14:03:29.060 -> load:0x3fff001c,len:1044
14:03:29.060 -> load:0x40078000,len:10124
14:03:29.060 -> load:0x40080400,len:5828
14:03:29.060 -> entry 0x400806a8
14:03:30.247 -> N:  
14:03:30.247 -> N: ****** setup ******
14:03:30.247 -> rtl_433_ESP(6): Pre initReceiver: 348292
14:03:30.247 -> rtl_433_ESP(6): CC1101 gpio receive pin: 4
14:03:30.247 -> rtl_433_ESP(6): CC1101 receive frequency: 433.920013
14:03:30.247 -> rtl_433_ESP(6): sizeof(*cfg->demod) 16
14:03:30.247 -> rtl_433_ESP(6): sizeof(*cfg->demod) 16
14:03:30.247 -> rtl_433_ESP(6): size of bitbuffer: 6604
14:03:30.247 -> rtl_433_ESP(6): size of pulse_data: 9620
14:03:30.247 -> rtl_433_ESP(6): # of device(s) configured 1
14:03:30.247 -> rtl_433_ESP(6): ssizeof(r_device): 108
14:03:30.247 -> rtl_433_ESP(6): cfg->devices size: 108
14:03:30.247 -> rtl_433_ESP(6): # of device(s) enabled 1
14:03:30.247 -> Registering protocol [1] "Fine Offset Electronics WH1080/WH3080 Weather Station"
14:03:30.280 -> rtl_433_ESP(6): CC1101 minumum rssi: -82
14:03:30.280 -> rtl_433_ESP(6): Post initReceiver: 347256
14:03:30.280 -> N: ****** setup complete ******
14:04:14.388 ->  
14:04:14.388 -> rtl_433_ESP(6): Signal length: 229564, Gap length: 44940679, Signal RSSI: -74, train: 0, messageCount: 0, pulses: 89
14:04:14.388 -> rtl_433_ESP(6): RAW (229787): +506-956+510-937+524-943+521-968+507-942+515-956+498-967+508-942+1510-952+496-955+1478-970+526-926+1486-963+1492-921+548-938+1506-934+509-965+507-954+1497-945+1500-939+1477-967+505-967+1490-938+526-943+521-941+1489-938+529-938+1521-954+1478-941+526-941+1494-952+1501-946+498-939+1531-934+1490-945+1487-935+1535-951+1480-953+495-969+1476-936+1530-916+1508-959+1467-942+1525-936+1500-944+501-966+1488-938+1511-954+1483-947+1492-950+1502-939+1483-968+1483-937+529-942+528-954+1474-962+1480-956+1484-964+1487-959+506-938+1504-953+1472-960+520-962+1486-936+532-940+1493-951+522-936+504-971+1492-939+1504-954+496-960+1482-960+1482-939+1512-954+1468-972+1475-955+499-955+1512-934+1487-957+538-925+1499-946+1497-940+1506-962+1463-950+1485-974+503-921+543-1013+127-201+0-100 
14:04:14.428 -> rtl_433_ESP(6): Pre run_ook_demods: 337568
14:04:14.428 -> rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
14:04:14.476 -> Fine Offset WH1080 data {80} ff 52 c5 a4 82 04 06 12 b2 09 
14:04:14.476 -> rtl_433_ESP(6): # of messages decoded 0
14:04:14.476 -> rtl_433_ESP(6): Signal processing time: 87875
14:04:14.476 -> rtl_433_ESP(6): Post run_ook_demods memory 337568
14:05:02.617 ->  
14:05:02.617 -> rtl_433_ESP(6): Signal length: 468138, Gap length: 47772518, Signal RSSI: -74, train: 1, messageCount: 1, pulses: 175
14:05:02.617 -> rtl_433_ESP(6): RAW (468309): +488-970+499-942+520-966+505-974+513-949+499-962+510-936+531-953+1475-944+522-943+1501-959+511-940+1480-971+1474-968+494-966+1484-932+529-960+517-959+1473-945+1501-940+1488-966+507-943+1478-950+522-957+525-940+1490-946+518-961+1476-953+1454-1039+433-1031+1424-993+1445-966+524-945+1486-966+1486-953+1488-940+1500-949+1499-941+1478-959+1505-956+1484-945+1450-1036+1427-983+1448-984+498-956+1486-942+1510-946+1493-943+1497-931+1524-931+1486-965+505-963+515-939+502-963+1486-944+1492-949+1510-953+1475-950+1497-939+523-941+1482-966+1487-952+515-943+1495-962+503-941+1509-938+507-976+494-965+1477-964+1485-961+484-961+1505-945+1492-949+1472-965+1503-933+1480-962+508-967+1470-952+522-963+506-964+507-952+500-952+1500-942+1479-965+504-971+1486-920+1527-32435+512-953+545-915+525-940+508-945+531-942+518-957+529-921+515-969+1474-966+526-931+1489-943+508-967+1492-934+1498-954+504-963+1483-950+516-952+524-942+1505-935+1487-968+1484-939+528-921+1526-933+501-967+506-937+1514-953+517-927+1502-952+1476-961+528-946+1491-938+1504-944+519-940+1505-938+1485-955+1487-961+1482-951+1501-941+1494-959+1474-967+1482-942+1509-954+1463-971+1486-948+498-959+1499-961+1476-966+1483-941+1479-1006+1432-993+1446-991+473-982+478-992+484-972+1474-941+1519-936+1482-981+1484-916+1535-928+508-965+1472-973+1474-954+501-960+1490-936+528-948+1499-956+502-941+530-946+1487-948+1501-952+520-939+1480-966+1454-1009+1458-977+1460-985+1447-971+521-942+1500-934+513-964+503-939+526-956+520-943+1484-950+1510-949+504-945+1501-939+1501-1036+102-203 
14:05:02.769 -> rtl_433_ESP(6): Pre run_ook_demods: 337568
14:05:02.769 -> rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
14:05:02.769 -> Fine Offset WH1080 data {80} ff a9 62 d2 40 04 0e 09 59 05 
[... cut ...] 80+1474-970+479-980+485-972+498-1083+51-410+807-23918+76-332+0-1139+51-1884+75-770 
14:31:26.453 -> rtl_433_ESP(6): Pre run_ook_demods: 337568
14:31:26.453 -> rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
14:31:26.500 -> Fine Offset WH1080 data {80} ff 52 c5 96 88 06 86 09 59 05 
14:31:26.500 -> rtl_433_ESP(6): # of messages decoded 0
14:31:26.500 -> rtl_433_ESP(6): Signal processing time: 91161
14:31:26.500 -> rtl_433_ESP(6): Post run_ook_demods memory 337568`

davoau, 2022-01-20:
Here is one where there are 2 messages together:

`14:06:38.637 -> rtl_433_ESP(6): Signal length: 483791, Gap length: 47774573, Signal RSSI: -74, train: 1, messageCount: 3, pulses: 176
14:06:38.637 -> rtl_433_ESP(6): RAW (483960): +508-956+504-962+492-1012+425-1038+431-1028+450-1000+476-984+478-986+1469-948+525-960+1483-961+486-960+1504-956+1472-945+517-971+1482-942+505-969+508-949+1498-918+1521-940+1484-966+508-952+1485-967+503-950+489-1010+1427-1014+455-1009+1459-937+1507-950+520-934+527-940+1508-957+486-968+1471-973+1470-964+1477-957+1485-971+1462-965+1510-928+1504-946+1495-910+1527-955+1478-956+1487-936+1508-951+514-955+1477-959+1479-930+1530-941+1490-942+1484-947+1501-963+517-934+1508-942+533-929+1489-943+1511-925+1498-943+1496-939+522-964+1488-931+1526-924+521-955+1482-943+529-939+1492-967+509-944+540-929+1475-955+1511-936+505-943+1533-943+1469-958+1497-918+1529-934+1489-964+509-969+1478-941+1505-955+1472-954+1498-968+450-1033+432-1021+1446-977+499-960+1482-964+1486-31201+301-959+496-966+506-969+511-930+524-946+526-935+528-938+524-951+544-927+1487-935+515-974+1471-942+522-962+1491-917+1532-925+527-953+1476-970+522-935+511-954+1499-946+1496-938+1524-910+531-936+1504-945+516-947+524-944+1507-933+518-941+1495-957+1495-917+526-963+506-969+1492-925+522-968+1482-941+1508-944+1486-940+1509-942+1493-959+1473-942+1526-943+1487-939+1486-941+1531-947+1489-940+1495-934+526-942+1511-951+1482-944+1495-922+1526-944+1482-970+1482-941+508-938+1531-928+521-935+1503-967+1484-941+1486-962+1490-947+539-919+1494-958+1475-942+531-942+1510-956+505-943+1500-940+525-939+511-939+1507-965+1483-938+515-963+1477-941+1532-934+1488-945+1488-924+1532-951+499-952+1499-944+1479-940+1532-930+1489-961+512-948+523-954+1478-943+528-956+1486-942+1508-1041+99-297 
14:06:38.767 -> rtl_433_ESP(6): Pre run_ook_demods: 337568
14:06:38.767 -> rtl_433_ESP(7): demod(6) - Fine Offset Electronics WH1080/WH3080 Weather Station
14:06:38.767 -> Fine Offset WH1080 data {80} ff a9 62 d3 40 02 05 09 59 04 
14:06:38.767 -> rtl_433_ESP(6): data_output {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.3,"humidity":64,"wind_dir_deg":90,"wind_avg_km_h":2.448,"wind_max_km_h":6.12,"rain_mm":717.90002,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-74,"duration":483960}
14:06:38.804 -> N: Received message : {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.3,"humidity":64,"wind_dir_deg":90,"wind_avg_km_h":2.448,"wind_max_km_h":6.12,"rain_mm":717.90002,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-74,"duration":483960}
14:06:38.804 -> pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
14:06:38.870 -> bitbuffer:: Number of rows: 1 
14:06:38.870 -> [00] {87} ff 52 c5 a6 80 04 0a 12 b2 08 68 
14:06:38.870 -> Fine Offset WH1080 data {80} ff a9 62 d3 40 02 05 09 59 04 
14:06:38.870 -> rtl_433_ESP(6): data_output {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.3,"humidity":64,"wind_dir_deg":90,"wind_avg_km_h":2.448,"wind_max_km_h":6.12,"rain_mm":717.90002,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-74,"duration":483960}
14:06:38.870 -> N: Received message : {"model":"Fineoffset-WHx080","subtype":0,"id":150,"battery_ok":1,"temperature_C":32.3,"humidity":64,"wind_dir_deg":90,"wind_avg_km_h":2.448,"wind_max_km_h":6.12,"rain_mm":717.90002,"mic":"CRC","protocol":"Fine Offset Electronics WH1080/WH3080 Weather Station","rssi":-74,"duration":483960}
14:06:38.917 -> pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
14:06:38.917 -> bitbuffer:: Number of rows: 1 
14:06:38.917 -> [00] {89} ff a9 62 d3 40 02 05 09 59 04 34 80 
14:06:38.917 -> rtl_433_ESP(6): # of messages decoded 2
14:06:38.917 -> rtl_433_ESP(6): Signal processing time: 295518
14:06:38.917 -> rtl_433_ESP(6): Post run_ook_demods memory 337568`

NorthernMan54, 2022-01-21:
PS I reformatted the comment so it was more readable, the usage of triple back ticks makes it easy on the eyes to spot things

Looking at the log data, my receiver code thinks a single message was received ( the single line of raw data and signal details ), and it was passed once to the rtl_433 code ( `Pre run_ook_demods:` )

Once in the method `pulse_demod_pwm` it hit a long silence https://github.com/merbanan/rtl_433/blob/32c1f4a65fd238be70dee4af7ef8c58cec7af554/src/pulse_demod.c#L427 and passed what was received to the device decoder.

Looking at the raw signal, there is this `+1486-31201` at about the half way mark, which is triggering this.

Am thinking that this is the expected behaviour, and the second message is valid data received from the sensor.

davoau, 2022-01-22:
Thanks for formatting it, I could not work out how to do it.

I don't get the 2 messages with the SDR rtl_433 receiver, so does this mean that the SDR rtl_433 is ignoring the second message ?
Is there something I can put in the code to stop it processing the second signal ?

I was wondering if me changing it to listen for only my weather station is when it started to happen. I will have it listen for a few more devices and see if it goes back to just decoding the one signal.

NorthernMan54, 2022-01-22:
What issue is occurring as a result of the second signal ?  Do you need to code a duplicate message filter?  Ie if current message equals previous message and gap is less than x minutes, drop the message

davoau, 2022-01-22:
Mostly the issue is the data is all being written to a database on the rpi, so I am now storing twice as much data as before or I will be when I switch over from SDR to this.  I graph it over a week/month/year, and also compare previous months/years.. its a farmers obsession with the weather.

Would just do the duplicated message filter in my code before sending to mqtt (compare the last json data to the current json data) or could I do a signal time check in rtl_433_esp code somewhere, as in if new message within 1/2 sec of last one drop it, which is maybe easier ?

NorthernMan54, 2022-01-22:
As the message callback is in the middle of the rtl_433 code am thinking it would make more sense in your code.  As their may be other users with different use cases.

I’m glad we found the issue though, I had thought it was a code error from the recent update and not a signal issue.

davoau, 2022-01-24:
I have been playing around to see if I could recreate what I had before with just getting the one message, and also with the rtl_433_callback check for multiple messages from the weather station.

In the callback function I put a if new message is less than 200 millisec's since last message return - I saw that the messages were around 97ms apart. It fixed the problem of multiple messages.

Then I changed the number of devices I check to what I had before, as below, and I only get one message again now (I removed the time check in callback). 
```
  memcpy(&cfg->devices[0], &fineoffset_WH2, sizeof(r_device));
  memcpy(&cfg->devices[1], &fineoffset_WH0530, sizeof(r_device));
  memcpy(&cfg->devices[2], &fineoffset_wh1050, sizeof(r_device));
  memcpy(&cfg->devices[3], &fineoffset_wh1080, sizeof(r_device));
  memcpy(&cfg->devices[4], &smoke_gs558, sizeof(r_device));

#define DEVICES           \
  DECL(fineoffset_WH2) \
  DECL(fineoffset_WH0530) \
  DECL(fineoffset_wh1050) \
  DECL(fineoffset_wh1080) \
  DECL(smoke_gs558) \
  /* Add new personal decoders here. */
  #define NUMOFDEVICES 5
#endif
```

Do you know why this is - I am just curious ?

NorthernMan54, 2022-01-24:
As the duplicate message was part of the signal decode and triggered by a short gap I'm not sure how this change would have affected that.

Also I was thinking further your comment around not sending a message if the previous message was sent within 200 milliseconds, and this would break the LaCrosse TX141W Wind speed sensor.  It sends a 2 part message which is then transmitted as 2 separate messages, one with temperature and one with wind speed.

davoau, 2022-01-26:
I am wondering if it because it takes time to process the the signal, as in check which device it is from, and then the rest is lost.
I will continue testing, as maybe (very likely) its something in my code - I have only found out about ESP's a few months ago and that made me to tpuch C++ for the first time.

I have now got the ESP32 with rtl_433_ESP working as well as the SDR_RTL_433 setup (not as reliable over distance of +15M from weather station though), just waiting for my new PCB's to come so I can connect to solar and put out in yard.

When/if I am happy with the new PCB's, where do you I suggest I post about it to make available gerber/schematics files if anyone else is interested ?

NorthernMan54, 2022-01-27:
The signal receiver logic and the decoder logic have rudimentary separation from each other.  The signal receiver is interrupt driven, and the decoder is part of the Arduino loop.  The only part that is shared is the start and end of signal detection, which is also part of the Arduino loop.  And the signal receiver uses multiple 'trains' to store the interim messages, to avoid overwriting a signal being processed with one being received.

For this scenario, the decoder time is not the issue as the signal receiver is receiving a single message, that is passed to the decoder as single decode stream.  If this was problematic, it would have shown up when I dig into the received signals.

For the gerber/schematics files, you could add them to this repo thru a pull request and include any required notes in a file.  Or I wonder if the OpenMQTTGateway team would be interested ?

PS I consider myself a junior / intermediate ESP coder, and haven't been doing it much longer than yourself.

cgmckeever, 2022-06-06:
@davoau stumbled on this learning more about the ESP project and saw that TASTEY looking board .. Is that something you are still working on?

Did you ever land on the best antenna ? Im struggling with range issues as well .. if you would like to chat out-of-band on this, please feel free to ping me at cgmckeever [at] gmail ... 

```
I have been using both the stubby antenna that came with the CC1101 module and a 173mm antenna which I replaced a pig tail antenna with on another module. The 173mm one has the best range, but gets the most double signals. The stubby antenna has very limited range, and drops around 1/2 of massages, but also has less of the double messages. I have ordered a couple more CC1101 modules with different antenna from aliexpress to see if that makes a difference.
```

closed 2023-04-03 by NorthernMan54