NorthernMan54/rtl_433_ESP #28: How to enable protocol

hvorragend, 2022-02-20:
Hi,
I am using OpenMQTTGateway and I want to enable to rtl_433 Intertechno protocol. 
But I don't know how to do that. 

Could you please help me?

Regards

NorthernMan54, 2022-02-21:
It is already enabled in the default build

https://github.com/NorthernMan54/rtl_433_ESP/blob/77707f473414e05c0af26d1c61d1a2b0fa7735a7/src/rtl_433/devices/intertechno.c#L11

hvorragend, 2022-02-21:
Thank you very much. 
Is it possible to use the command line interface of rtl_433 in OMG?

Do I have a chance to sniff the raw messages?
Or is it enough if I use "DPUBLISH_UNPARSED=true"?

I use a doorbell that I can receive with various other 433 MHz variants (SIGNALduino as MapleCUN.  ased on CC1101).  No FSK.  It only doesn't work with rtl_433.

Several protocols are possible with this doorbell: EV1527, Intertechno, FreeTec_PE-6946

I just see:
`rtl_433_ESP(6): RAW (3283164): +231-600+222-617+630-206+639-183+239-614+209-615+216-612+641-196+221-602+228-614+636-208+616-208+213-610+221-612+651-205+208-611+637-209+616-208+213-611+642-198+220-600+233-614+206-638+611-210+218-6438+221-611+226-630+609-207+643-209+215-618+210-620+210-610+631-205+213-631+206-628+614-200+650-206+208-613+208-613+637-210+216-612+638-197+640-200+202-633+634-184+238-591+235-616+216-612+642-197+195-6458+235-613+213-613+642-204+627-198+215-615+203-636+205-615+635-210+193-637+201-633+617-202+650-206+209-613+233-589+638-210+214-619+638-184+639-204+213-610+644-203+204-630+207-613+235-616+614-209+213-6452+236-607+208-606+645-196+637-202+203-634+206-614+233-612+612-208+218-634+208-627+615-198+648-206+208-611+210-637+613-209+213-619+636-210+611-200+214-634+622-202+206-635+209-611+235-616+615-211+211-6456+211-609+234-607+622-232+625-195+195-623+232-612+207-616+639-209+214-611+220-609+644-202+630-206+209-615+235-590+639-209+213-618+636-193+635-199+214-611+652-205+208-612+207-617+234-616+615-209+215-6442+222-610+225-629+609-209+642-209+214-619+210-615+210-615+641-197+214-611+226-628+632-182+643-210+214-594+236-611+629-208+213-605+640-198+641-200+204-634+636-208+196-611+216-637+195-637+629-205+209-6441+239-609+194-634+617-200+651-206+209-615+206-612+236-617+613-201+213-634+223-602+632-207+640-209+213-616+211-613+641-209+213-612+636-181+646-207+215-607+624-204+213-627+198-640+200-632+637-207+213-6424+239-608+210-608+640-205+626-197+217-621+230-610+208-615+638-213+211-620+209-614+640-200+633-198+221-601+231-611+636-210+194-639+620-204+628-196+217-620+632-209+209-615+236-616+208-614+640-202+212-6439+230-611+234-590+636-210+618-212+211-619+210-619+236-611+614-199+236-618+200-631+634-210+615-209+192-633+219-610+624-206+233-610+636-210+614-209+214-615+640-211+213-594+238-613+208-619+642-181+215-6457+218-608+224-617+631-206+638-213+190-613+233-609+221-610+627-202+234-612+206-613+637-210+615-211+212-619+237-593+642-208+212-612+646-195+617-203+235-608+635-211+212-617+214-615+208-615+640-201+213-6442+229-611+235-589+638-210+614-208+213-634+196-638+201-631+634-184+218-613+218-636+616-200+649-205+209-611+208-614+637-215+210-612+639-197+638-200+203-632+635-183+239-613+212-614+209-610+645-198+218-6445+236-591+235-616+611-210+640-203+213-610+222-617+228-609+635-214+189-613+218-634+618-200+650-205+208-613+209-611+638-209+215-611+643-195+637-201+204-632+635-183+240-590+237-614+216-610+643-198+193-6461+235-592+235-616+616-201+632-200+218-619+230-610+209-637+613-208+217-611+218-612+642-202+631-207+209-614+235-615+612-209+213-618+635-211+610-202+213-635+626-199+231-609+207-615+234-616+614-208+215-6444+219-636+203-628+609-208+640-210+213-620+208-614+210-619+635-210+213-621+212-618+641-181+648-206+188-634+197-639+630-207+209-613+635-210+617-206+212-635+616-200+222-628+206-611+235-613+616-207+216-6452+217-608+232-600+644-202+632-204+209-612+234-615+210-616+614-209+216-637+206-628+619-197+624-228+208-613+207-636+611-210+217-613+644-197+636-201+202-631+608-236+211-616+209-616+192-636+640-198+195-6454+235-617+211-616+640-210+619-211+215-619+190-635+196-637+621-202+234-609+206-614+639-208+618-209+213-619+210-620+637-211+213-619+615-208+637-203+212-629+618-197+223-627+208-609+234-613+610-210+214-6445+232-605+220-609+645-202+632-206+209-615+234-613+210-617+639-199+213-611+224-600+635-205+640-210+195-634+193-633+618-225+202-629+608-209+642-207+214-617+611-211+213-620+236-593+237-612+617-196+216-6472+207-614+234-615+614-212+617-210+212-614+238-618+210-618+616-208+214-611+232-607+640-207+627-196+217-617+202-636+636-207+214-614+613-207+639-206+187-634+645-199+202-630+207-612+235-613+614-210+212-6456+236-596+234-596+642-202+634-198+215-615+203-634+209-611+641-206+216-612+209-619+639-211+621-207+213-611+210-619+641-209+211-607+646-198+638-199+202-630+635-211+211-615+210-617+212-615+636-210+190-6470+195-637+202-633+609-236+616-209+189-636+195-635+224-600+633-207+214-611+236-616+615-202+633-198+222-618+232-607+636-209+213-614+614-208+640-204+214-610+643-201+202-633+205-613+234-617+610-209+214-6456+212-620+210-621+640-204+610-227+186-638+202-634+208-613+637-210+214-614+210-618+636-211+618-209+214-610+234-595+642-207+214-608+646-197+636-199+203-634+633-182+219-612+218-635+197-637+627-208+207-6450+233-587+232-609+641-196+620-207+230-610+208-613+235-617+614-210+213-619+211-614+653-198+634-197+219-600+231-610+638-209+191-636+615-198+636-223+201-630+636-183+217-612+219-633+197-640+624-206+210-6447+238-593+211-620+638-208+636-177+214-634+223-600+228-612+635-211+191-636+219-608+644-200+628-205+210-612+234-588+638-210+215-619+640-184+639-208+214-605+647-197+214-618+227-608+208-614+639-210+192-6465+220-608+225-602+634-208+639-213+190-637+208-605+222-631+622-201+233-607+208-611+637-210+615-211+209-634+198-637+627-206+210-614+638-209+615-210+213-609+642-197+222-600+232-612+233-587+641-37314 
N: Send on /RTL_433toMQTT/unknown/null msg {"model":"unknown","protocol":"signal parsing failed","duration":3283164,"rssi":-21,"pulses":599,"train":0,"messageCount":2,"_enabledReceiver":1,"receiveMode":0,"currentRssi":-92,"minimumRssi":-82}
 `

NorthernMan54, 2022-02-22:
As you found published unparsed will output the raw ook signal

the other option to try is rtl_verbose, this will turn on verbose/debug for a particular decoder, so you may be able to why the decode is failing.  The level of output is dependent on what the rtl_433 has for that device signal decoder.

in regards to command access, no but I did try to include some useful options as a defines.

with the cc1101 device I have not been impressed with the signal reception quality and range.  And find it is significantly shorter than what you see with a rtl_sdr.  Maybe I have a cc1101 setting wrong, I don’t know.

NorthernMan54, 2022-02-22:
Looking at that log, I don’t see anything for your decoder, so it appears the device decoder for your device is not enabled by default in the rtl_433 codebase

https://github.com/NorthernMan54/rtl_433_ESP/blob/77707f473414e05c0af26d1c61d1a2b0fa7735a7/src/rtl_433/devices/intertechno.c#L88

set that to 0 and you should see some debug output and fingers crossed a decoded message

hvorragend, 2022-02-22:
I've tried several protocols. No chance. 
So I have to keep running my SIGNALduino CUL in FHEM. 
Thanks for your help.

closed 2022-02-22 by hvorragend

NorthernMan54, 2022-02-22:
Tks for the update