NorthernMan54/rtl_433_ESP #32: Add that project ti Platform IO registry

joroMaser, 2022-04-24:
Please add that project to Platform IO registry to we can easily install to esp32

NorthernMan54, 2022-04-24:
Personally I use the project from within OpenMQTTGateway, and use it with platformio to do my builds etc.

joroMaser, 2022-04-24:
But most of time I just want to use receiver to print RF data without any use of OpenMQTTGateway like
https://github.com/NorthernMan54/rtl_433_ESP/blob/master/example/OOK_Receiver/OOK_Receiver.ino

So I offer to add that project stand-alone to PlatformIO registry

KiCADHugo, 2023-05-28:
OpenMQTTGateway is too large for me, and i don't understand configure it. so how can i use this library with platformio separately?
I have a LilyGo T3_v1.6.1 433 MHz.

NorthernMan54, 2023-05-28:
The folder example/OOK_Receiver is a standalone example

closed 2023-05-28 by NorthernMan54