NorthernMan54/rtl_433_ESP #33: Add support to EV1527

joroMaser, 2022-04-24:
Please add support to EV1527 like [rc-switch](https://github.com/sui77/rc-switch)

NorthernMan54, 2022-04-24:
Is this device decoder included within the rtl_433 device decoders?  I couldn't see it.

joroMaser, 2022-04-24:
@NorthernMan54 No.
That why I ask you to add that chip to your project  ,so I can read the data from that RF chip with esp32 and rf receiver 
Like I can read here
https://github.com/sui77/rc-switch/blob/master/examples/ReceiveDemo_Simple/ReceiveDemo_Simple.ino

NorthernMan54, 2022-04-24:
Sorry, the scope of this project is a port of the rtl_433 device decoders to esp32.

joroMaser, 2022-04-24:
@NorthernMan54 So should I ask here
https://github.com/merbanan/rtl_433
?

joroMaser, 2022-04-25:
@NorthernMan54 
Please take a look [here](https://github.com/merbanan/rtl_433/issues/2055)

this device include in flex decoder.

NorthernMan54, 2022-04-25:
The flex decoder is not included due to the memory usage of that decoder.  It causes a stack overflow when included in the build.

closed 2023-02-22 by NorthernMan54