NorthernMan54/rtl_433_ESP #35: Support for Honeywell Sensors

sandstormkeshav, 2022-04-30:
Hello! Thank you for this project! I was wondering if it is possible to implement the Honeywell decoder in this project? The source in rtf_433 is here https://github.com/merbanan/rtl_433/blob/master/src/devices/honeywell.c and it seems to use the OOK_PULSE_MANCHESTER_ZEROBIT modulation.

NorthernMan54, 2022-05-01:
It would need some effort and access to a Honeywell device to test with.  Also as the cc1101 is a bit deaf and the receiver is not that sensitive, will it work with your use case?

sandstormkeshav, 2022-05-01:
Thank you! My sensors are a few meters away from the cc1101 so I hope the signal will be strong enough. I noticed some references to the ook_pulse_manchester in your flex.c file but I'm not sure if there are more components needed.

NorthernMan54, 2022-05-01:
Likely, I had removed a lot of the rtl433 codebase to get it to work on a esp32

sandstormkeshav, 2022-05-01:
Ah okay, do you have any tips on anything I can try to help?

NorthernMan54, 2022-05-01:
Give me a few days, I need to do refresh and will try to include these as well

sandstormkeshav, 2022-05-01:
Okay thank you very much!

sandstormkeshav, 2022-05-24:
I have tried adding the Honeywell.c file into the devices but I had trouble building the OMG firmware. Will wait for your refresh @NorthernMan54

NorthernMan54, 2022-05-24:
Okay, I'm just awaiting the next rtl-433 refresh.  Their last release was 21.12 ( Dec 14, 2021 ), and will update when the next release is published.

sandstormkeshav, 2022-05-28:
Got it! Thank you!

sandstormkeshav, 2022-11-27:
Hi! It looks like 22.11 has been released a few days ago, when you have the time can we revisit trying to add the Honeywell decoder? Thank You!

NorthernMan54, 2022-11-27:
I actually started it last week before heading out for vacation, https://github.com/NorthernMan54/rtl_433_ESP/tree/rtl_433_Release-22.11.  I’m still in the middle of updating, and it is non-functional at the moment.

sandstormkeshav, 2022-11-29:
wow, thank you I really appreciate it. Please let me know if there's anything you'd like me to test as the development progresses.

sandstormkeshav, 2023-02-22:
Small update but I was able to add the honeywell.c files into the project now that OOK_PULSE_MANCHESTER has been added, however my decode success rate is quite low, even right next to the sensor. Do you have any tips or parameters I can adjust that may help?

NorthernMan54, 2023-02-22:
Try pull request #57

sandstormkeshav, 2023-02-26:
Thank you for the quick reply! I tweaked some of those settings that were applicable to the CC1101 but still am experiencing decodes only once in a while.

NorthernMan54, 2023-02-26:
If you refresh to the current release, I have merged that pull request and put in the Honeywell.c decoder as well.  Hopefully this resolves it for you

sandstormkeshav, 2023-02-27:
Thank you! I just rebuilt using the current release, however I am still getting this:
```
rtl_433_ESP(7): Average RSSI Signal -85 dbm, adjusted RSSI Threshold -76, samples 50000
rtl_433_ESP(6): Signal length: 383044, Gap length: 38567690, Signal RSSI: -46, train: 0, messageCount: 36, pulses: 194
rtl_433_ESP(6): Unparsed Signal length: 383044, Signal RSSI: -46, pulses: 195
rtl_433_ESP(6): RAW (383044): +0-124(-46)+135-124(-46)+160-110(-46)+140-135(-46)+134-110(-46)+166-110(-46)+137-137(-46)+136-111(-46)+163-110(-46)+137-138(-46)+138-110(-46)+164-110(-45)+138-110(-45)+163-109(-45)+259-263(-45)+269-132(-45)+136-139(-45)+137-242(-43)+161-109(-43)+138-136(-43)+255-263(-44)+133-133(-44)+146-138(-44)+135-110(-44)+137-136(-44)+275-106(-44)+137-257(-44)+264-267(-44)+132-135(-44)+258-150(-44)+129-248(-44)+279-260(-43)+268-265(-43)+252-147(-43)+128-138(-43)+135-131(-43)+138-109(-43)+137-136(-44)+136-248(-44)+270-132(-44)+137-135(-43)+122-269(-43)+267-253(-43)+142-120(-43)+139-121(-43)+156-138(-43)+266-107(-44)+136-275(-44)+133-137(-44)+138-107(-44)+275-82569(-41)+138-109(-41)+136-135(-41)+138-137(-41)+134-108(-42)+138-138(-42)+137-131(-42)+137-109(-42)+138-135(-42)+124-132(-42)+138-139(-42)+135-134(-42)+274-240(-42)+256-140(-42)+147-114(-40)+148-248(-40)+147-128(-40)+137-135(-41)+251-263(-41)+160-106(-41)+147-116(-40)+139-118(-40)+139-120(-40)+270-147(-40)+128-274(-40)+252-262(-40)+132-134(-40)+265-118(-40)+142-254(-41)+264-278(-41)+266-252(-41)+265-133(-41)+134-125(-41)+141-139(-41)+119-139(-41)+146-129(-41)+138-240(-41)+270-134(-41)+134-135(-41)+138-259(-41)+261-267(-41)+133-139(-41)+135-107(-41)+164-110(-41)+275-136(-41)+134-232(-40)+143-122(-40)+162-108(-40)+82148-138(-40)+138-134(-40)+104-162(-40)+110-137(-40)+135-134(-41)+137-137(-41)+107-162(-41)+110-138(-41)+134-121(-41)+132-138(-41)+138-135(-41)+132-136(-40)+111-135(-40)+272-270(-40)+265-132(-41)+138-134(-41)+102-272(-41)+133-134(-41)+135-138(-41)+260-262(-41)+133-133(-41)+133-137(-41)+137-133(-41)+135-111(-42)+275-133(-42)+136-233(-42)+269-267(-41)+158-109(-41)+258-151(-41)+130-274(-41)+253-262(-41)+267-264(-41)+252-148(-41)+129-138(-41)+135-130(-41)+137-137(-41)+109-135(-41)+137-275(-42)+240-134(-42)+136-136(-42)+122-270(-41)+266-253(-41)+142-121(-41)+139-147(-41)+131-139(-41)+239-160(-41)+108-274(-41)+133-137(-41)+137-106(-42)+82721-111(-43)+137-136(-43)+137-111(-43)+166-107(-43)+137-138(-44)+138-109(-44)+136-137(-44)+139-110(-44)+162-110(-44)+138-138(-44)+137-109(-44)+272-273(-42)+268-133(-42)+139-108(-42)+126-271(-42)+133-134(-42)+136-136(-42)+259-263(-43)+133-134(-43)+134-135(-43)+138-107(-43)+163-110(-43)+276-133(-43)+137-231(-42)+293-244(-42)+156-109(-42)+258-151(-42)+130-274(-42)+252-261(-42)+268-265(-42)+252-148(-42)+129-138(-42)+135-130(-42)+138-109(-43)+137-137(-43)+136-249(-43)+267-134(-42)+137-135(-42)+122-270(-42)+267-252(-43)+141-120(-43)+141-119(-43)+157-138(-43)+267-107(-43)+136-274(-43)+134-137(-42)+138-108(-42)+275-0(-42)
N: uniqueid: unknown
T: isAdupl?
T: getDeviceById unknown-rssi
T: update unknown-rssi
N: Send on /RTL_433toMQTT/unknown msg {"model":"unknown","protocol":"signal parsing failed","duration":383044,"rssi":-46,"pulses":195}
T: jsonPubl - ON
T: [ OMG->MQTT ] topic: home/OpenMQTTGateway_multi_receiver/RTL_433toMQTT/unknown msg: {"model":"unknown","protocol":"signal parsing failed","duration":383044,"rssi":-46,"pulses":195} 
T: Min ind: 10
T: store code : 0 / 265608
T: Col: val/timestamp
T: mem code : 0 / 47215
T: mem code : 17836 / 47491
T: mem code : 0 / 52856
T: mem code : 538872 / 53222
T: mem code : 0 / 180171
T: mem code : 0 / 189325
T: mem code : 981349 / 189555
T: mem code : 0 / 192840
T: mem code : 0 / 215120
T: mem code : 0 / 224406
T: mem code : 0 / 265608
T: mem code : 0 / 0
rtl_433_ESP(6): # of messages decoded 0
rtl_433_ESP(6): Signal length: 382996, Gap length: 947261, Signal RSSI: -35, train: 1, messageCount: 37, pulses: 195
rtl_433_ESP(6): Unparsed Signal length: 382996, Signal RSSI: -35, pulses: 196
rtl_433_ESP(6): RAW (382996): +64-164(-35)+97-164(-35)+119-154(-35)+108-161(-36)+102-189(-36)+83-159(-36)+106-159(-36)+111-160(-36)+74-184(-36)+82-189(-36)+83-180(-36)+84-163(-36)+105-160(-36)+232-297(-36)+238-156(-37)+109-164(-37)+106-296(-37)+106-156(-37)+98-160(-37)+232-281(-37)+133-132(-37)+131-137(-37)+134-123(-37)+130-139(-38)+246-158(-38)+109-285(-38)+258-264(-38)+130-137(-38)+235-153(-38)+132-273(-38)+255-263(-38)+266-262(-39)+254-148(-39)+131-138(-39)+135-132(-39)+137-138(-39)+106-162(-39)+109-274(-39)+266-134(-39)+137-108(-40)+127-271(-40)+265-253(-40)+143-122(-40)+139-147(-40)+131-138(-40)+241-160(-39)+107-273(-39)+133-138(-39)+136-132(-40)+82684-153(-40)+108-165(-40)+107-163(-40)+105-164(-40)+108-121(-41)+132-166(-41)+109-135(-41)+104-162(-42)+111-159(-42)+107-162(-42)+111-136(-42)+243-287(-41)+264-133(-41)+133-138(-42)+111-265(-42)+132-161(-42)+109-136(-42)+255-265(-43)+133-132(-43)+123-160(-43)+137-109(-42)+136-135(-42)+279-129(-42)+108-260(-43)+267-266(-43)+131-135(-43)+258-150(-43)+129-274(-43)+253-262(-43)+267-263(-43)+252-149(-43)+129-138(-43)+136-129(-43)+137-137(-43)+108-135(-43)+137-274(-43)+239-159(-43)+110-136(-43)+125-271(-43)+265-253(-44)+143-122(-44)+141-146(-44)+130-138(-43)+240-160(-43)+108-272(-43)+133-137(-43)+137-132(-43)+246-82317(-40)+138-120(-40)+153-139(-40)+135-101(-40)+162-110(-41)+136-136(-41)+136-137(-41)+137-108(-41)+135-138(-41)+138-137(-41)+131-136(-41)+137-111(-41)+135-124(-41)+270-269(-41)+250-142(-40)+121-139(-40)+147-246(-40)+147-127(-40)+138-136(-40)+250-263( [... cut ...]
rtl_433_ESP(6): Signal length: 76000, Gap length: 103374, Signal RSSI: -46, train: 1, messageCount: 41, pulses: 50
rtl_433_ESP(6): data_output {"model":"Honeywell-Security","id":981349,"channel":8,"event":0,"state":"closed","contact_open":0,"reed_open":0,"alarm":0,"tamper":0,"battery_ok":1,"heartbeat":0,"mic":"CRC","protocol":"Honeywell Door/Window Sensor, 2Gig DW10/DW11, RE208 repeater","rssi":-46,"duration":76000}
N: uniqueid: Honeywell-Security-8-981349
T: isAdupl?
T: no pub. dupl
rtl_433_ESP(6): # of messages decoded 1
```
Please let me know if you have anything else in mind I could try.

sandstormkeshav, 2023-03-04:
I wonder if it's possible the sampling rate is too low for this sensor, it's odd that out of the 6 packets the Honeywell sensor sends, only about 1 gets decoded.

rct, 2023-04-18:
What frequency are your Honeywell sensors using?

Are you using a radio module for the correct frequency?

At least in the US/NA, the Honeywell modules covered in src/devices/honeywell.c trasmit on 345Mhz.  I didn't think the cc1011 modules offer that option, but would be interested in learning more.

Also note, there is currently some room for improvement in rtl_433 for improving the reliability of Honeywell decoding.

https://github.com/merbanan/rtl_433/issues/2015

sandstormkeshav, 2023-04-18:
> What frequency are your Honeywell sensors using?
> 
> Are you using a radio module for the correct frequency?
> 
> At least in the US/NA, the Honeywell modules covered in src/devices/honeywell.c trasmit on 345Mhz. I didn't think the cc1011 modules offer that option, but would be interested in learning more.
> 
> Also note, there is currently some room for improvement in rtl_433 for improving the reliability of Honeywell decoding. The flex decoder is currently getting more decodes for me than the builtin decoder.
> 
> [merbanan/rtl_433#2015](https://github.com/merbanan/rtl_433/issues/2015)

Hi! They are using 345MHz, the cc1101 should be able to receive and transmit on this frequency and I have got some successful decodes just not as many as rtl433 on an sdr. I will try to add this flex decoder in and report back. Do you feel as if you are still missing some decodes with the new decoder?

rct, 2023-04-19:
> the cc1101 should be able to receive and transmit on this frequency

Yes, but there is usually a bit more discrete supporting circuitry that is tuned for the desired frequency. 

> Do you feel as if you are still missing some decodes with the new decoder?

Yes.  It is definitely still a bit of a work in progress. There are some challenges to overcome there.  It would be interesting to see how well a purpose built receiver does.

NorthernMan54, 2023-04-19:
The latest release included tuning for problematic reception.  You may want to upgrade and try again.

closed 2024-11-30 by NorthernMan54