NorthernMan54/rtl_433_ESP #36: Repeating Error Message: E (251696) gpio: gpio_set_level(226): GPIO output gpio_num error

cgmckeever, 2022-06-04:
```
E (416810) gpio: gpio_set_level(226): GPIO output gpio_num error
E (416814) gpio: gpio_set_level(226): GPIO output gpio_num error
E (416826) gpio: gpio_set_level(226): GPIO output gpio_num error
E (416847) gpio: gpio_set_level(226): GPIO output gpio_num error
E (417909) gpio: gpio_set_level(226): GPIO output gpio_num error
E (417919) gpio: gpio_set_level(226): GPIO output gpio_num error
E (417942) gpio: gpio_set_level(226): GPIO output gpio_num error
E (417943) gpio: gpio_set_level(226): GPIO output gpio_num error
E (417945) gpio: gpio_set_level(226): GPIO output gpio_num error
E (417950) gpio: gpio_set_level(226): GPIO output gpio_num error
E (419861) gpio: gpio_set_level(226): GPIO output gpio_num error
I: Received message : {"model":"Markisol","id":0,"control":"Limit (0)","channel":1,"zone":1,"mic":"CHECKSUM","protocol":"Markisol, E-Motion, BOFU, Rollerhouse, BF-30x, BF-415 curtain remote","rssi":-70,"duration":1911360}
 
```

I have the library working, and its receiving/decoding various devices. However, I continually see this error message in the serial output. Having trouble tracking down what/why is calling this. Any ideas?

NorthernMan54, 2022-06-04:
It sounds like one of the GPIO definition's has an invalid value or not defined.  Could be one from my library or the dependant CC1101 library.

cgmckeever, 2022-06-04:
@NorthernMan54 THANKS .. yeah, that is where Im digging down into now ... have everything set (I think) very vanilla with your defaults. Must be something out of alignment. 

Will update this if I stumble across anything .. appreciate it

cgmckeever, 2022-06-04:
@NorthernMan54 I tracked this down. I think its a combo of a few things mainly, if I remove 

```
pinMode(ONBOARD_LED, OUTPUT);
digitalWrite(ONBOARD_LED, LOW);
```

from the 4 places it appears in `rtl_433_ESP.cpp` the noisey debug message goes away.

I _think_ its because my chip `Chip is ESP32-D0WDQ6 (revision 1)` does not have an `ONBOARD_LED` and that call does not fail gracefully (even set to `-1`) and having debug high just catches it.

Could create a `toggle_led` method that just bails out if `ONBOARD_LED < 1` or something ..

NorthernMan54, 2022-06-05:
Or just set it to a valid, but unused GPIO pin

cgmckeever, 2022-06-05:
Thanks @NorthernMan54 .. setting to a unused/valid pin that the ESP32 doesnt need at boot worked 

https://randomnerdtutorials.com/esp32-digital-inputs-outputs-arduino/
```
All GPIOs can be used as outputs except GPIOs 6 to 11 (connected to the integrated SPI flash) and GPIOs 34, 35, 36 and 39 (input only GPIOs);
```

closed 2022-06-05 by cgmckeever