NorthernMan54/rtl_433_ESP #37: Additional Receivers

cgmckeever, 2022-06-06:
👋 @NorthernMan54  ... Got this running using the basic configs. Works like magic for sniffing some of my devices. THANK YOU

TLDR; Whats your "BUY YOU A BEER ADDRESS" -- because I'd like to prime you for an ask ;) 

Having the same receiver distance issue as noted in [other issues](https://github.com/NorthernMan54/rtl_433_ESP/issues/10#issuecomment-974693605) and also stumbled across some posts about possible [future ](https://github.com/NorthernMan54/rtl_433_ESP/issues/11#issuecomment-1003369889)receiver [work](https://github.com/NorthernMan54/rtl_433_ESP/issues/6#issuecomment-816150027)  .. @skiphansen (ever resolve your distance issue)

Wondering if there has been any further thought into different receivers .. and also curious if this library is able to use very basic receivers such as shown in the attached images .. OR .. if the chipset (or base library) needs to be smarter as alluded to in your other [thread](https://github.com/NorthernMan54/rtl_433_ESP/issues/11#issuecomment-1003369889)? 

# Standard Generic Arduino Module
[image]

# Open-Smart 
[image]

# AdaFruit (discussion that its receive it very good with distance)
[image]

NorthernMan54, 2022-06-06:
When I made this port almost 2 years ago, I was doing a quick and dirty effort to see if it was even possible so I picked a module I already had access to ( a CC1101 ) and went.  For a radio receiver it needs a module that provides the raw signal data, and supports ASK/OOK modulation.  And the code is also using signal RSSI to detect start and end of message.

These other modules that support ASK/OOK modulation, supply raw signal data and supply RSSI would be a simple rewiring of the code to a different receiver library ( not extremely complex ).

When it comes to other 433 Mhz modules, this is a good resource - https://www.rflink.nl/wires.php please note the comments on the cheap receiver.

Where things get tricky around support is that a lot of the chipset based receivers also do signal decoding onboard, and pass a decoded packet rather than a raw signal.

While writing this, did a quick GitHub survey and saw this library https://github.com/jgromes/RadioLib and this https://github.com/rfquack/RFQuack which I have not explored yet. On a quick glance at RFQuack it appears to expose the raw signal data.

cgmckeever, 2022-06-06:
you didnt add how I can get you a beer or coffee -- or send you some alixpress RF modules ;)

skiphansen, 2022-06-06:
@NorthernMan54 Thanks for the updated info!  

No I didn't resolve the range issue.  I did some reading about the CC1101 and thought of some changes I wanted to try but I got distracted by other projects (look a squirrel!) and never got back to it.

Great project, I'll get back to it eventually.

NorthernMan54, 2022-06-06:
Isn't the nRF24 2.4 Ghz and not 433 Mhz ?

cgmckeever, 2022-06-06:
well ... probably -- Hadnt gotten that far yet. I was just excited about trying something else out :( 
Possibly then try to source the `RFM69HW` then

closed 2022-06-09 by cgmckeever

dreamcat4, 2023-03-01:
can we... reopen this issue maybe? because it seems like this is generally a good discussion place for finding / experimenting with better receivers. many thanks

melyux, 2024-04-09:
There's a guy on Hackaday forums named Zulea (@Zulea ?) who says he modified this library to run with a classic superhererodyne receiver SRX882S instead of a CC1101. I wish he would respond to the replies and tell us what exactly he did, so we could test it out.

> I made a lot of tests with CC1101 and SX1278 modules using this library. First test was with OpenMqttGateway then later I integrated this in Tasmota firmware for personal use to receive different 433 MHz RF meteo sensors in more locations. Also extended the receive/decode to Oregon Scientific v2, v3 sensors (OOK_PULSE_MANCHESTER_ZEROBIT) and on RF gate opener Nice Flor-S. Both modules I used with cable/external high gain (tuned to 433 MHz) omnidirectional antennas. The radio sensitivity of this modules (at least in ASK/OOK mode and with settings from this library) is VERY bad (compared with old analog SH receivers). So I ended using the old SRX882S v2.0 (and modified this library to receive from data pin of this module). The con thing is I no more have RSSI information, but compared to higher sensitivity of SRX882S, I choose this solution.
Many thanks to NorthernMan54 for this great library, but unfortunately I was not able to use any of this modules (CC1101 or SX1278) to get good signal from my spreaded RF sensors. I supose this is because both modules was not designed for this kind of job (ASK/OOK “blind” receiver).

@NorthernMan54 I assume these classic superheterodyne modules wouldn't work with FSK signals?

NorthernMan54, 2024-04-09:
In my library I use the RSSI value supplied by the receiver to detect start and end of signal. If the receiver has the ability to supply signal detect or carrier detect on a pin then it is a simple change. 

FSK is different and would need to have the receiver module do FSK signal demodulation.

melyux, 2024-12-11:
@zulea1 Finally found your GitHub: think you could tell us how you modified this library to use an analog receiver like the SRX882S?