NorthernMan54/rtl_433_ESP #38: Switch between RTL Rx and RcSwitch TX -- Rx stops working

cgmckeever, 2022-06-09:
I'm trying to 
- use Rx from the `RTL_433_ESP` library to listen for sensors
- disable it with a Tx request
- switch to Tx with the RC Switch library; send a pulse and switch back to listening 

The good news (that may be surprising) is that it works ... prototype is [here](https://github.com/cgmckeever/arduino-sketch/pull/11/files#diff-dfedaf8f8a09b0cd28744e3f49011025131deaf6929ce03f608aa98a3228736cR161)

The bad news is, after a few toggles Rx just stops. I can't figure out if it similar to the [deaf-state](https://github.com/NorthernMan54/rtl_433_ESP/issues/16) or even the [freq change Rx stop](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/65) OR if Im just doing something either impossible or missing something silly.

The swap works, sometimes. I usually get 1 swap before I can't get any more Rx. But there are some times I can get a few. But it always goes deaf. Any ideas .. appreciate it 

The main bits for enabling/disabling are 

```
// RTL
//
void rtlSetup() {
    Log.notice(F(" " CR));
    Log.notice(F("****** RTL setup begin ******" CR));
    Log.notice(F("Frequency: %F" CR), config.frequency);
    rf.initReceiver(config.receivePin, config.frequency);
    enableRx();
    rf.setCallback(rtl433Callback, messageBuffer, messageBufferLen);
    Log.notice(F("****** RTL setup complete ******" CR));
}

void enableRx() {
    ELECHOUSE_cc1101.SpiStrobe(CC1101_SIDLE);
    ELECHOUSE_cc1101.SetRx(config.frequency);
    //ELECHOUSE_cc1101.setMHZ(config.frequency);
    rf.enableReceiver(config.receivePin);
}

void switchTransmit(float freq, int pulse, int decimal, int bits) {
    Log.notice(F("Sending:" CR));
    Log.notice(F("  freq: %F" CR), freq);
    Log.notice(F("  pulse: %d" CR), pulse);
    Log.notice(F("  decimal: %d" CR), decimal);
    Log.notice(F("  bits %d" CR), bits);

    rf.disableReceiver();
    ELECHOUSE_cc1101.SpiStrobe(CC1101_SIDLE);
    //ELECHOUSE_cc1101.setMHZ(freq);
    ELECHOUSE_cc1101.SetTx(freq);

    mySwitch.enableTransmit(config.transmitPin);
    mySwitch.setPulseLength(pulse);
    mySwitch.send(decimal, bits);
    mySwitch.disableTransmit();

    enableRx();
}

void rtl433Callback(char* message) {
    Log.notice(F("Received message: %s" CR), message);
    messagePost("sensor", message);
}

void messagePost(String path, char* message) {
    HTTPClient http;
    WiFiClient client;

    String url = String(config.serverURL);
    url.trim();

    http.begin(client, url + path);
    http.addHeader("Content-Type", "application/json");
    int httpResponseCode = http.POST(message);
}
```

```
void setup() {
    Serial.begin(115200);
    Log.begin(logLevel, &Serial);

    ELECHOUSE_cc1101.Init();
    configSetup();
    rtlSetup();
}

void loop() {
    configManager.loop();
    rf.loop();
}
```

NorthernMan54, 2022-06-09:
If you look at the OpenMQTTGateway implementation I have implemented that particular use case, as I use it all the time.

Listen with RTL_433 and Transmit with RC_SWITCH

cgmckeever, 2022-06-09:
Thanks! Least Im not crazy .. 
Confirming this project https://github.com/1technophile/OpenMQTTGateway/blob/development/main/main.ino .. any hint to which hood to lift up? Appreciate it

NorthernMan54, 2022-06-09:
Unfortunately multiple.  I had to update the driver for each protocol decoder to support switching etc

This was the PR with the enhancement

NorthernMan54, 2022-06-09:
Ooops

https://github.com/1technophile/OpenMQTTGateway/pull/936

cgmckeever, 2022-06-09:
Similar patterns .. no 💡 
I did notice `rf.initReceiver(-1);` .. which I cant do. Probably an earlier library version. 

two new discoveries. If I just `rf.disableReceiver();` and dont even do Tx, I cant cant seem to get back to Rx

ie

- Receiving
- `rf.disableReceiver();`
- Try to re-enable
```
ELECHOUSE_cc1101.SetRx(config.frequency);
ELECHOUSE_cc1101.setMHZ(config.frequency);

rf.setCallback(rtl433Callback, messageBuffer, messageBufferLen);
rf.enableReceiver(config.receivePin);
```

I more often than not, get no Rx. With that, attempting to `rf.initReceiver(config.receivePin, config.frequency);`  throws

```
WARNING: low memory? malloc() failed in blueline_create() from /Users/cgmckeever/Documents/Arduino/libraries/rtl_433_esp/blueline.c:416
21:05:05.746 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

cgmckeever, 2022-06-09:
It seems (or at least works) magic combo was

```
void disableRX() {
    rf.enableReceiver(-1);
    rf.disableReceiver();
}
```

Thanks for leading me to that other code!!

closed 2022-06-09 by cgmckeever