NorthernMan54/rtl_433_ESP #41: Issues parsing LaCrosse TX141-Bv2 protocol

oleksiikutuzov, 2022-09-07:
First of all, thanks a lot for your work on this library! That's exactly what I was looking for for a long time. 

I have seen from the other issues that this sensor (or the one from TFA, in my case) is quite popular and seems to work fine for others. I have built the OOK receiver example with the latest platform package (espressif32@5.1.1) and the latest library version (0.1.0) with the CC1101 receiver. 

As far as the sensor is located directly near the receiver, it is parsed correctly:
```
rtl_433_ESP(6): Signal length: 602996, Gap length: 49246189, Signal RSSI: -14, train: 0, messageCount: 32, pulses: 358
rtl_433_ESP(6): RAW (602996): +813-851(-16)+818-881(-17)+796-882(-18)+218-510(-19)+219-501(-19)+460-292(-19)+197-525(-20)+217-525(-20)+218-510(-20)+200-521(-21)+459-278(-21)+200-521(-21)+221-517(-21)+202-546(-21)+196-519(-23)+203-546(-23)+195-536(-24)+457-260(-24)+466-280(-25)+222-498(-25)+215-537(-25)+442-281(-25)+206-528(-25)+200-521(-25)+222-515(-25)+200-521(-25)+222-517(-25)+202-546(-25)+224-496(-25)+461-277(-25)+216-517(-25)+464-253(-25)+237-512(-25)+219-500(-25)+205-530(-25)+224-523(-25)+219-514(-25)+440-282(-25)+206-536(-25)+219-500(-25)+459-293(-25)+197-525(-25)+458-279(-25)+224-547(-25)+800-874(-26)+805-875(-25)+815-851(-25)+822-881(-25)+218-514(-25)+218-501(-25)+459-292(-25)+198-525(-25)+217-525(-25)+191-538(-25)+192-528(-25)+459-290(-25)+197-524(-25)+217-524(-25)+218-485(-25)+248-501(-25)+204-530(-25)+223-498(-25)+457-285(-25)+458-280(-25)+221-498(-25)+218-533(-25)+441-281(-25)+206-534(-25)+218-501(-25)+204-531(-25)+220-527(-25)+204-525(-25)+197-525(-25)+216-524(-25)+459-280(-25)+191-528(-25)+459-291(-25)+197-524(-25)+217-524(-25)+218-513(-25)+220-500(-25)+204-528(-25)+444-275(-25)+230-525(-25)+219-500(-25)+460-292(-25)+196-525(-25)+457-281(-25)+224-546(-25)+798-874(-25)+806-874(-25)+816-852(-26)+821-857(-25)+242-513(-26)+219-500(-25)+460-293(-25)+196-524(-25)+216-525(-25)+219-514(-25)+221-500(-25)+458-266(-25)+224-525(-25)+216-498(-25)+216-538(-25)+197-521(-25)+222-515(-25)+200-548(-25)+433-277(-25)+476-253(-25)+219-518(-25)+211-521(-25)+459-280(-25)+198-546(-25)+196-518(-25)+202-547(-25)+196-523(-25)+217-529(-25)+212-526(-25)+192-528(-25)+457-264(-25)+226-504(-25)+459-294(-25)+197-524(-25)+216-525(-25)+220-510(-25)+200-521(-25)+222-517(-25)+452-279(-25)+217-515(-25)+200-547(-25)+433-280(-25)+224-524(-25)+446-275(-25)+221-551(-25)+809-874(-25)+815-879(-25)+797-881(-25)+797-879(-25)+221-511(-25)+219-501(-25)+460-293(-25)+198-524(-25)+219-522(-25)+191-538(-25)+191-528(-25)+460-292(-25)+195-526(-25)+212-527(-25)+199-522(-25)+220-514(-25)+200-521(-25)+223-519(-25)+465-253(-25)+476-280(-25)+221-500(-25)+205-531(-25)+447-288(-25)+200-540(-25)+199-523(-25)+220-517(-25)+220-499(-25)+205-529(-25)+221-526(-25)+204-525(-25)+451-282(-25)+205-531(-25)+440-280(-25)+208-536(-25)+219-500(-25)+205-528(-25)+221-525(-25)+204-523(-25)+452-281(-25)+206-530(-25)+219-503(-25)+458-292(-25)+197-525(-25)+458-280(-25)+437-147438(-87)+868-862(-19)+798-885(-21)+794-881(-21)+796-882(-22)+219-513(-23)+218-505(-24)+468-274(-24)+221-524(-24)+204-522(-25)+197-523(-25)+214-535(-25)+441-281(-25)+206-531(-25)+219-502(-25)+204-529(-25)+221-525(-25)+205-525(-25)+196-523(-25)+455-305(-25)+444-258(-25)+226-525(-25)+219-502(-25)+457-290(-25)+197-525(-25)+217-525(-25)+218-513(-25)+214-523(-25)+191-535(-25)+200-522(-25)+221-515(-25)+452-278(-25)+216-514(-25)+452-278(-25)+218-518(-25)+219-527(-25)+210-525(-25)+192-528(-25)+204-527(-25)+468-275(-25)+211-498(-25)+224-524(-25)+457-255(-25)+248-495(-25)+461-278(-25)+217-568(-25)+797-874(-25)+805-875(-26)+814-852(-25)+820-879(-25)+219-513(-25)+218-506(-25)+466-273(-25)+223-497(-25)+215-523(-25)+220-518(-25)+219-501(-25)+460-293(-25)+196-525(-25)+216-524(-25)+219-514(-25)+220-500(-25)+205-529(-25)+196-548(-25)+457-279(-25)+443-281(-25)+205-530(-25)+219-501(-25)+460-293(-25)+197-525(-25)+217-500(-25)+242-512(-25)+191-528(-25)+205-528(-25)+222-524(-25)+204-523(-25)+452-282(-25)+204-528(-25)+440-281(-25)+210-536(-25)+218-501(-25)+204-531(-25)+222-524(-25)+204-523(-25)+452-257(-25)+229-529(-25)+219-500(-25)+459-293(-25)+197-524(-25)+457-280(-25)+224-546(-25)+799-874(-25)+801-878(-25)+814-852(-26)+820-882(-26)+218-514(-26)+216-504(-25)+471-273(-25)+222-524(-25)+180-521(-25)+221-524(-25)+216-531(-25)+439-289(-25)+199-534(-25)+199-522(-25)+222-517(-25)+218-500(-25)+205-530(-25)+223-523(-25)+458-279(-25)+442-262(-25)+222-534(-25)+202-519(-25)+460-280(-25)+220-505(-25)+208-542(-25)+198-522(-25)+220-516(-25)+219-507(-25)+206-518(-25)+223-524(-25)+458-280(-25)+192-528(-25)+459-290(-25)+198-524(-25)+217-525(-25)+192-541(-25)+192-527(-25)+205-527(-25)+452-280(-25)+205-532(-25)+221-499(-25)+461-291(-25)+197-524(-25)+458-280(-25)+464-306(-25)+803-874(-25)+808-870(-25)+817-853(-25)+822-855(-25)+243-513(-26)+220-500(-25)+461-292(-25)+197-524(-25)+216-525(-25)+190-537(-25)+219-500(-25)+460-293(-25)+197-523(-25)+216-524(-25)+220-513(-25)+221-500(-25)+204-528(-25)+223-498(-25)+457-286(-25)+464-280(-25)+222-498(-25)+214-507(-25)+472-282(-25)+205-530(-25)+221-499(-25)+204-527(-25)+198-548(-25)+219-512(-25)+219-500(-25)+205-528(-25)+451-282(-25)+207-536(-25)+467-258(-25)+199-543(-25)+199-521(-25)+221-518(-25)+218-504(-25)+206-545(-25)+440-286(-25)+200-538(-25)+199-522(-25)+461-280(-25)+220-501(-25)+458-293(-25) 
rtl_433_ESP(6): data_output {"model":"LaCrosse-TX141THBv2","id":33,"channel":0,"battery_ok":1,"temperature_C":30.0,"humidity":40,"test":"No","mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-14,"duration":602996}
N: Received message : {"model":"LaCrosse-TX141THBv2","id":33,"channel":0,"battery_ok":1,"temperature_C":30.0,"humidity":40,"test":"No","mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor","rssi":-14,"duration":602996}
```

If I move it further away, starting from 50 cm from the receiver, it cannot parse the data anymore:
```
rtl_433_ESP(6): Signal length: 575957, Gap length: 49273240, Signal RSSI: -26, train: 1, messageCount: 57, pulses: 355
rtl_433_ESP(6): RAW (575957): +543-992(-28)+686-965(-29)+135-613(-29)+132-607(-29)+370-349(-29)+134-596(-30)+155-569(-30)+163-570(-29)+155-584(-30)+381-330(-31)+154-596(-31)+153-580(-30)+163-562(-31)+160-585(-32)+158-554(-32)+188-549(-31)+411-315(-32)+425-327(-32)+158-555(-33)+194-548(-32)+188-549(-33)+404-328(-33)+411-289(-34)+452-302(-33)+190-525(-33)+433-314(-34)+196-543(-34)+161-556(-33)+430-303(-33)+190-554(-34)+434-291(-34)+195-523(-35)+457-282(-35)+222-520(-35)+189-555(-34)+190-533(-35)+190-528(-35)+458-290(-34)+430-303(-34)+437-282(-34)+195-550(-34)+189-529(-34)+214-556(-35)+798-880(-34)+798-897(-34)+786-903(-34)+768-879(-34)+214-535(-35)+218-526(-34)+439-289(-34)+195-523(-34)+216-526(-35)+219-510(-34)+201-548(-35)+433-304(-35)+174-547(-34)+195-517(-34)+202-546(-34)+195-538(-34)+202-526(-34)+195-525(-34)+457-283(-34)+457-278(-34)+221-525(-34)+189-536(-35)+190-528(-35)+461-291(-35)+426-303(-34)+441-281(-34)+222-518(-34)+457-279(-34)+193-545(-34)+189-530(-34)+450-275(-34)+220-525(-34)+432-311(-34)+198-523(-34)+434-303(-34)+194-546(-34)+189-529(-35)+212-528(-35)+190-526(-35)+460-291(-34)+427-303(-34)+439-283(-34)+223-522(-34)+189-528(-34)+214-555(-35)+824-879(-34)+774-901(-34)+784-876(-35)+817-878(-34)+190-538(-34)+218-526(-34)+434-292(-34)+195-525(-34)+217-522(-34)+220-514(-34)+218-526(-34)+435-292(-35)+195-525(-34)+215-524(-35)+191-539(-35)+218-526(-34)+179-531(-34)+220-525(-34)+433-286(-34)+457-304(-34)+194-523(-34)+219-513(-34)+217-532(-35)+418-298(-34)+449-290(-34)+429-297(-34)+221-523(-34)+432-284(-34)+223-522(-34)+189-529(-34)+452-302(-34)+193-524(-34)+433-286(-34)+224-524(-34)+432-305(-34)+195-544(-34)+189-530(-34)+212-529(-35)+190-527(-34)+460-290(-34)+425-303(-34)+439-283(-34)+225-521(-34)+189-528(-34)+214-557(-34)+823-879(-34)+775-901(-34)+784-876(-34)+817-879(-34)+193-533(-34)+217-526(-35)+434-292(-35)+195-524(-34)+215-523(-35)+220-514(-35)+218-526(-34)+436-293(-34)+194-551(-34)+188-549(-34)+195-516(-34)+217-505(-34)+209-542(-35)+199-523(-35)+457-279(-34)+440-276(-34)+209-536(-34)+216-533(-34)+182-546(-34)+445-283(-34)+440-303(-34)+417-286(-34)+201-546(-34)+452-277(-34)+216-515(-34)+217-531(-35)+444-273(-34)+221-523(-34)+432-286(-34)+223-522(-34)+432-279(-34)+218-545(-34)+189-530(-34)+211-527(-34)+190-527(-34)+460-289(-34)+428-303(-34)+438-283(-34)+223-522(-34)+188-529(-34)+147908-2989(-28)+699-979(-27)+697-965(-29)+129-610(-28)+137-592(-28)+383-335(-29)+137-592(-30)+160-577(-29)+ [... cut ...] 189-529(-48)+214-528(-48)+443-282(-48)+441-306(-48)+190-526(-48)+459-288(-48)+196-525(-48)+458-279(-48) 
rtl_433_ESP(6): # of messages decoded 0
```

Pulses seem to be more or less okayish (I don't have a lot of experience with radio stuff) except for the value in brackets, which is getting larger with distance. Are there any parameters that can be adjusted?

I have already ordered another Infactory sensor to test with a different protocol, but might there be that I am just doing anything wrong?

NorthernMan54, 2022-09-07:
With the code base I'm still doing regression testing, and recently identified that the CC1101 defaults in the RadioLib library needed additional tuning, and the most recent version on GitHub include this ( promoted last week ).  If you playing around, perhaps version 0.0.4 may work better/differently ( It used a different library for the CC1101 Transceiver ).

One other thing I found was that the CC1101 performed better when the antenna was vertical ( straight up ) rather than horizontal.

This is from the regression testing setup, and I have several lacrosse devices, ranging between 10 feet to 50 feet away from the test bed.

OMG Heltec - is with this [board](https://heltec.org/project/wifi-lora-32/), running v0.1.1 under OpenMqttGateway
OMG North - is with a CC1101, running v0.0.4 under OpenMqttGateway ( Previous version )
OMG CC110 - is with a CC1101, running v0.1.1 under OpenMqttGateway ( Latest version )

[image]

oleksiikutuzov, 2022-09-07:
Thanks for a quick reply! Yes, indeed, it works great with the different library (on older version). Unfortunately, 0.0.4 version doesn't work on a latest espressif platform package, which is required by the library I want to integrate this library with. So I will be looking forward for the further development with the new library.

NorthernMan54, 2022-09-07:
Can you just pull in my latest commit then?

oleksiikutuzov, 2022-09-07:
Yeah, this actually works great. Could receive even through multiple walls

NorthernMan54, 2022-09-08:
Are you okay, and we can close this?

closed 2022-09-08 by oleksiikutuzov