NorthernMan54/rtl_433_ESP #45: Add decoders for AC114-xxB and BeSmart_Sx

TheChatty, 2022-11-20:
SignalDuino already has support for [AC114-xxB](https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/lib/SD_ProtocolData.pm#L1656) and [BeSmart_Sx](https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/lib/SD_ProtocolData.pm#L2177).

Both appear as follows with trl_433_ESP v0.1.4:
```
home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/unknown/null {"model":"unknown","protocol":"signal parsing failed","duration":170000,"rssi":-27,"pulses":98}
home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/unknown/null {"model":"unknown","protocol":"signal parsing failed","duration":135028,"rssi":-52,"pulses":102}
home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/unknown/null {"model":"unknown","protocol":"signal parsing failed","duration":154996,"rssi":-41,"pulses":98}
home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/unknown/null {"model":"unknown","protocol":"signal parsing failed","duration":119044,"rssi":-52,"pulses":35}

home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/unknown/null {"model":"unknown","protocol":"signal parsing failed","duration":510042,"rssi":-24,"pulses":394}
home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/unknown/null {"model":"unknown","protocol":"signal parsing failed","duration":709024,"rssi":-24,"pulses":668}
home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT/unknown/null {"model":"unknown","protocol":"signal parsing failed","duration":468848,"rssi":-23,"pulses":394}
```

How hard is it to add those protocols? Or do they need to be added to rtl_433 beforehand?

TheChatty, 2022-11-20:
I don't know if it helps but here's a raw signal from AC114-xxB:
```
rtl_433_ESP(6): Signal length: 417996, Gap length: 21995386, Signal RSSI: -35, train: 0, messageCount: 6, pulses: 394
rtl_433_ESP(6): RAW (417996): +589-197(-38)+189-610(-36)+594-197(-36)+193-591(-36)+199-596(-36)+198-595(-37)+598-196(-36)+588-198(-36)+196-595(-36)+600-196(-36)+194-596(-36)+200-588(-36)+199-595(-36)+574-220(-36)+590-197(-36)+589-198(-36)+595-198(-36)+197-595(-36)+200-595(-36)+596-198(-37)+192-590(-37)+575-221(-37)+591-197(-37)+193-596(-37)+598-198(-37)+195-594(-36)+199-597(-37)+594-197(-37)+193-595(-37)+571-221(-36)+591-197(-36)+193-595(-37)+199-595(-37)+199-595(-37)+199-595(-36)+198-591(-37)+198-591(-37)+198-594(-37)+199-595(-37)+595-196(-36)+195-591(-37)+200-596(-37)+198-595(-37)+197-593(-37)+198-591(-37)+199-594(-37)+198-594(-38)+197-592(-38)+199-593(-38)+199-594(-37)+573-221(-37)+191-593(-37)+199-594(-37)+198-594(-37)+573-221(-37)+589-197(-37)+595-198(-37)+196-595(-37)+198-596(-37)+573-221(-37)+192-590(-37)+602-196(-37)+567-222(-37)+591-197(-37)+590-247(-37)+5166-601(-37)+599-197(-37)+191-594(-37)+598-197(-37)+195-592(-37)+199-595(-37)+197-593(-37)+597-197(-38)+590-196(-37)+194-598(-37)+595-198(-37)+194-593(-37)+198-596(-37)+199-595(-37)+594-196(-37)+592-196(-37)+594-198(-37)+592-197(-37)+195-597(-37)+198-594(-37)+574-219(-37)+193-597(-37)+595-198(-37)+592-198(-37)+194-593(-38)+600-196(-38)+192-594(-38)+199-593(-37)+597-198(-37)+195-593(-37)+595-197(-37)+590-198(-37)+195-594(-37)+200-598(-37)+196-594(-37)+198-594(-37)+196-594(-37)+198-595(-37)+197-592(-37)+198-591(-37)+597-197(-37)+196-596(-37)+200-595(-37)+197-589(-37)+199-594(-37)+198-596(-37)+198-594(-37)+200-586(-37)+198-593(-37)+200-595(-37)+199-596(-38)+593-197(-37)+192-591(-37)+199-597(-37)+199-594(-37)+594-196(-38)+590-198(-37)+594-197(-37)+195-597(-37)+197-598(-37)+593-197(-37)+196-590(-37)+598-196(-37)+590-198(-37)+593-200(-37)+592-244(-37)+5177-597(-37)+570-222(-37)+191-588(-37)+598-197(-37)+191-593(-37)+199-593(-37)+199-593(-37)+574-221(-37)+588-197(-37)+196-595(-37)+600-198(-37)+195-593(-37)+197-595(-37)+197-590(-36)+599-198(-36)+592-196(-37)+589-197(-37)+594-198(-37)+197-596(-37)+199-594(-37)+599-171(-37)+216-589(-37)+599-199(-37)+592-196(-37)+196-594(-37)+597-197(-37)+195-594(-37)+199-597(-37)+570-221(-37)+192-590(-37)+598-198(-37)+594-196(-37)+194-595(-37)+198-598(-37)+197-593(-37)+198-594(-37)+199-595(-37)+197-592(-37)+199-589(-36)+198-594(-37)+601-194(-37)+191-596(-37)+198-593(-37)+201-592(-37)+198-595(-37)+197-593(-37)+197-592(-37)+197-594(-37)+199-594(-37)+197-595(-37)+198-592(-37)+572-221(-37)+193-594(-37)+198-595(-37)+198-591(-38)+572-221(-38)+591-197(-37)+592-198(-37)+196-594(-37)+199-597(-37)+598-197(-37)+190-593(-37)+597-197(-37)+594-198(-36)+595-170(-36)+612-248(-37)+5173-596(-37)+571-221(-37)+192-594(-37)+597-197(-37)+193-594(-37)+197-597(-37)+199-594(-37)+594-196(-37)+590-197(-37)+196-596(-37)+598-196(-37)+192-593(-37)+198-593(-37)+199-594(-37)+599-173(-37)+610-197(-36)+593-197(-36)+598-194(-36)+195-597(-36)+198-596(-36)+595-197(-36)+193-594(-36)+599-172(-36)+611-198(-36)+194-593(-36)+601-197(-36)+194-595(-36)+198-594(-36)+596-199(-37)+192-592(-36)+600-197(-37)+587-198(-37)+195-593(-37)+199-598(-38)+199-595(-36)+196-593(-36)+198-591(-36)+198-594(-36)+198-596(-37)+197-593(-36)+595-197(-36)+193-593(-36)+200-597(-36)+198-592(-36)+199-589(-36)+199-593(-36)+199-594(-36)+196-593(-36)+197-591(-36)+200-596(-36)+198-594(-36)+598-172(-36)+215-590(-36)+199-594(-36)+199-595(-36)+599-172(-36)+611-197(-37)+595-199(-37)+197-593(-36)+199-597(-36)+594-196(-36)+193-590(-36)+599-195(-36)+592-197(-36)+591-199(-36)+594-15782(-43)+5155-669(-30)+519-270(-30)+119-686(-31)+520-247(-31)+144-639(-31)+148-640(-31)+149-665(-32)+521-247(-32)+566-222(-33)+169-640(-32)+547-223(-33)+171-643(-34)+150-620(-34)+173-614(-33)+571-222(-34)+567-222(-35)+566-222(-35)+591-198(-35)+195-594(-35)+200-594(-36)+574-220(-36)+192-589(-36)+598-199(-36)+592-196(-36)+196-594(-36)+597-198(-36)+194-593(-36)+199-596(-37)+597-196(-36)+192-592(-36)+597-198(-36)+592-197(-36)+194-594(-36)+199-598(-36)+198-591(-36)+199-594(-36)+198-594(-36)+197-593(-37)+200-588(-36)+199-593(-36)+599-194(-36)+192-596(-36)+198-593(-36)+199-594(-36)+199-595(-36)+197-592(-36)+197-592(-36)+198-593(-36)+201-592(-36)+197-595(-36)+197-591(-36)+597-198(-36)+195-593(-36)+197-594(-36)+198-591(-36)+597-198(-36)+593-196(-36)+589-198(-36)+196-594(-36)+199-595(-36)+599-173(-36)+215-593(-36)+596-198(-36)+592-198(-36)+594-197(-36)+591-222(-36)+5193-596(-36)+572-222(-36)+193-593(-36)+597-195(-36)+193-593(-35)+198-595(-35)+200-593(-36)+594-196(-36)+591-198(-36)+196-595(-36)+599-195(-35)+192-594(-35)+198-594(-35)+198-595(-35)+600-172(-36)+609-198(-35)+592-198(-35)+596-197(-36)+194-598(-36)+197-596(-36)+595-197(-36)+193-593(-36)+601-171(-36)+611-199(-35)+193-592(-35)+600-196(-35)+195-597(-35)+198-595(-36)+596-198(-36)+192-591(-35)+599-196(-35)+587-198(-35)+195-593(-35)+200-598(-36)+199-595(-35)+197-593(-35)+197-591(-35)+199-593(-35)+199-594(-36)+197-592(-35)+596-197(-35)+194-594(-35)+199-600(-35)+197-594(-35)+199-586(-36)+198-593(-36)+199-594(-36)+199-597(-35)+195-588(-35)+201-593(-35)+199-594(-35)+599-171(-35)+215-590(-35)+200-595(-35)+198-595(-35)+598-172(-35)+611-197(-35)+595-198(-35)+197-595(-35)+199-595(-35)+594-196(-35)+193-594(-35)+597-196(-35)+591-196(-35)+591-198(-35)
```

NorthernMan54, 2022-11-21:
For device decoders I'm picking them up from rtl_433, so they need to be added there first.

closed 2023-02-22 by NorthernMan54

TheChatty, 2023-03-02:
@zuckschwerdt showed me only the [long sync needs to be handled](https://github.com/merbanan/rtl_433/issues/2242#issuecomment-1451762907). How to config this in OpenMQTTGateway?

NorthernMan54, 2023-03-03:
Unfortunately flex decoders are not supported

TheChatty, 2023-03-04:
To my understanding normal decoders are sufficient. I was hoping that "sync time" is a default in the source code which could be changed?

NorthernMan54, 2023-03-04:
Sync time is a setting within the flex decoder, and is not supported

TheChatty, 2023-03-26:
According to @zuckschwerdt "doorbell style" decoders won't be added (need to be [decoded via conf's](https://github.com/merbanan/rtl_433/pull/2434#issuecomment-1475161854)) but he still helped me to use [generic_motion](https://github.com/merbanan/rtl_433/blob/master/src/devices/generic_motion.c) as template for the above signals.

This [works with RTL2832U](https://github.com/merbanan/rtl_433/issues/2242#issuecomment-1484163666) but far from stable with OMG+rtl_433_esp. Maybe the sampling rate is to low?