NorthernMan54/rtl_433_ESP #46: SX127x range

garudaonekh, 2022-11-22:
Hi, 
what is the range of SX127x?

 I use CC1101 from Ebytes and a few low cost sensors(weather station, rain bucket, soil moisture) but the range is around 20 meters on open field and less than 10 meters behind concrete wall. It is too short.

NorthernMan54, 2022-11-22:
With my heltec board it felt like it was receiving slightly better but with radio and RF Reception but there are a lot of variables so I would not call my results too scientific.   I did not notice a lot of unknown devices showing up with the Heltec board, so the signal reception was not significantly different.  Also  @DigiH and @1technophile have tried it as well, hopefully they can chime in as well with their impressions.

garudaonekh, 2022-11-23:
I don't want to make any change to the sender sides(sensors) and want to focus on improving the receiver side like antenna and use lora chip but not much is improved. Maybe I follow the wrong path or because I don't have enough knowledge on these antenna things.

DigiH, 2022-11-23:
The reception with my CC1101 using OpenMQTTGateway improved with the recent changes to rtl_433_ESP and the switch to RadioLib.

For the SX127x however I went for a separate SX1278 module, instead of the all-in-one Heltec board. With that module my reception is not that great, but I probably still need to try and test some more with different setting and I also think that the included antenna which came with the board might be a bit dodgy.

garudaonekh, 2022-11-24:
Have you guys tried LLCC68?  It is much cheaper than SX127x and also longer distance? Based on China market, It is 30% to 50% cheaper than SX127x series.
However, it is based on SX126x series which is not supported in this library. It is working on RadioLib

NorthernMan54, 2022-11-24:
The Sx126x series can’t be supported by rtl_433_esp as it does not support a OOK mode.So that one is out

garudaonekh, 2023-02-07:
thanks for the info

closed 2023-02-07 by garudaonekh