NorthernMan54/rtl_433_ESP #48: Watchman sonic oil tank monitor

daemonp, 2022-12-22:
First, thank you for this library!

I found a link to it from the home assistant forums as I was looking to update my rtl_433 setup which is monitoring the volume of heating oil in our oil tank. With some of the recent changes with home assistant mqtt it looks like I will have to rework a fair amount of the way I monitor oil usage anyways and would be much happier if this was running on a ESP rather than a full blown pi just to fetch these readings.

Looking through the script that updates from rtl_433 source it removes the `oil_watchman.c` file. Please let me know if there are any specific guidelines or requirements for submitting a pull request for a new feature. I am happy to contribute to this project and look forward to hearing your thoughts on this proposed addition. Thank you for considering my request.

NorthernMan54, 2022-12-22:
It uses FSK modulation, which is not supported with the port to ESP.  It would be technically feasible to enable this decoder but it would require physical access to the signal so that the the receiver can be configured for the details of the FSK signal.

https://github.com/merbanan/rtl_433/blob/616b5d5eba341e589e733e12a87193383c54a7c5/src/devices/oil_watchman.c#L115

daemonp, 2022-12-23:
Thanks, looks like a much larger port to bring `pulse_*` over from rtl_433 to get all the key shifting magic working.

closed 2022-12-23 by daemonp