NorthernMan54/rtl_433_ESP #50: OOKReceiver seems to not work either in Arduino or PlatformIo environment

fenyvesi, 2023-01-17:
May be my fault...
Arduino Win10, v1.8.12:
1. The rtl_433_ESP downloaded from github, the content is placed into ../Arduino/libraries.
2.  After starting Arduino the error msg is:
Invalid library in C:\Users\x\Documents\Arduino\libraries\rtl_433_ESP: no header file (.h) : C:\Users\x\Documents\Arduino\libraries\rtl_433_ESP

PlatformIO (Win10):

1. platformio.ini is adjusted (CC1101, com port, speed)
2. Build, 4 problems:

#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Users\x\Documents\PlatformIO\Projects\rtl_433_ESP\example\OOK_Receiver\OOK_Receiver.ino). ( #include <rtl_433_ESP.h>)
cannot open source file "stdint.h" (dependency of "rtl_433_ESP.h")
cannot open source file "string.h" (dependency of "ArduinoJson.h")
cannot open source file "inttypes.h" (dependency of "ArduinoLog.h")

On the contrary it says that it was successfull and I can upload it to the board. I inserted a Serial.println() statement after opening serial, but I can't see anything, the led blinks.

Thank you in advance.

fenyvesi, 2023-01-17:
I can see the serial output. Sorry.
Is there a chance to work with Arduino?

NorthernMan54, 2023-01-17:
The library is arduino based, it just uses platformio as the ide and build environment.  Take some time to learn the platformio / VSC ide,  it is vastly superior to the arduino ide, and greatly simplifies development.

am thinking the issue you saw with platformio may be related to windows,  as I do my coding on Mac.

closed 2023-02-22 by NorthernMan54

zoomx, 2023-12-19:
The message "Invalid library" in Arduino IDE appers because there is no library.properties file
I create one and the message disappeared

`name=rtl_433_ESP
version=0.3.2
author=Northern Man NorthernMan54
maintainer=Northern Man NorthernMan54
sentence=rtl_433 port to ESP32
paragraph=This is an attempt at creating an Arduino library for use on ESP32 boards with a CC1101 transceiver or SX127X Transceivers with the device decoders from the rtl_433 package. And be available for use with openMQTTGateway as an available module.
category=Communication
url=https://github.com/NorthernMan54/rtl_433_ESP
architectures=esp32
includes=rtl_433_ESP.h
`