NorthernMan54/rtl_433_ESP #51: OOK_PULSE_MANCHESTER_ZEROBIT, ESP32 DOIT DevKIT V1 and Ai Thinker Ra01 (SX1278) problem

diepeterpan, 2023-01-21:
Hi, I have created a fork with the following changes

1) Enabled OOK_PULSE_MANCHESTER_ZEROBIT
2) New build def ESP32 DOIT DevKIT V1 and Ai Thinker Ra01 (SX1278)
3) Added sensor "Oregon Scientific Weather Sensor" added which covers the OWL electricity meter

Now sometimes it works, decodes 3 lines and then crash with kernel error.
Sometimes there's no valid decode that occurs and it still crashes with kernel error.

See - https://github.com/diepeterpan/rtl_433_ESP/tree/master/example/OOK_Receiver

Besides playing around with compile options and RSSI any advice or assistance you can give me to resolve the problem?

NorthernMan54, 2023-01-21:
Can you share the kernel error ?It is likely a resource issue with the device decoder or the Manchester decoder.  Which may not be resolvable.

diepeterpan, 2023-01-21:
I had it unplugged for a while now, plugged it again, started VSCode monitor and after +-5min to 10min the following.

rtl_433_ESP(7): Average RSSI Signal -104 dbm, adjusted RSSI Threshold -92, samples 50000
N: Received message : {"model":"Oregon-CM160","id":7,"power_W":499,"protocol":"Oregon Scientific Weather Sensor","rssi":-62,"duration":130000}
N: Received message : {"model":"Oregon-CM180i","id":10096,"battery_ok":1,"power1_W":32,"power2_W":61051,"power3_W":177,"energy_kWh":1222995.5,"sequence":0,"protocol":"Oregon Scientific Weather Sensor","rssi":-60,"duration":130000}
N: Received message : {"model":"Oregon-CM160","id":7,"power_W":499,"protocol":"Oregon Scientific Weather Sensor","rssi":-59,"duration":129000}
rtl_433_ESP(7): Average RSSI Signal -104 dbm, adjusted RSSI Threshold -92, samples 50000
rtl_433_ESP(7): Average RSSI Signal -104 dbm, adjusted RSSI Threshold -92, samples 50000
N: Received message : {"model":"Oregon-CM160","id":7,"power_W":499,"protocol":"Oregon Scientific Weather Sensor","rssi":-59,"duration":129000}
rtl_433_ESP(7): Average RSSI Signal -104 dbm, adjusted RSSI Threshold -92, samples 50000
N: Received message : {"model":"Oregon-CM180i","id":10096,"battery_ok":1,"power1_W":16,"power2_W":64850,"power3_W":177,"energy_kWh":1.04051e+07,"sequence":0,"protocol":"Oregon Scientific Weather Sensor","rssi":-60,"duration":130000}
N: Received message : {"model":"Oregon-CM160","id":7,"power_W":531,"protocol":"Oregon Scientific Weather Sensor","rssi":-60,"duration":129000}
rtl_433_ESP(7): Average RSSI Signal -104 dbm, adjusted RSSI Threshold -92, samples 50000
N: Received message : {"model":"Oregon-CM160","id":7,"power_W":515,"protocol":"Oregon Scientific Weather Sensor","rssi":-59,"duration":129000}
rtl_433_ESP(7): Average RSSI Signal -104 dbm, adjusted RSSI Threshold -92, samples 50000
rtl_433_ESP(7): Average RSSI Signal -104 dbm, adjusted RSSI Threshold -92, samples 50000
Guru Meditation Error: Core  1 panic'ed (Unknown reason)
Core 1 register dump:
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

NorthernMan54, 2023-01-21:
I looked at your changes and at a high level glance they appear fine. 

I’m guessing that you are running out of stack and that is triggering this

I would turn on RTL_DEBUG and MEMORY_DEBUG and the same time insert some stack monitoring serial prints in the device decoder.   

When I was doing the initial port I found that lines like this were problematic, and needed to move the allocation to the class level to move it from stack to heap

https://github.com/diepeterpan/rtl_433_ESP/blob/db543ea24374a00fe122cd5397ba6d9a80d74c83/src/rtl_433/devices/oregon_scientific.c#L227 <https://github.com/diepeterpan/rtl_433_ESP/blob/db543ea24374a00fe122cd5397ba6d9a80d74c83/src/rtl_433/devices/oregon_scientific.c#L227>

See this

https://github.com/diepeterpan/rtl_433_ESP/blob/db543ea24374a00fe122cd5397ba6d9a80d74c83/src/rtl_433/pulse_demod.c#L92 <https://github.com/diepeterpan/rtl_433_ESP/blob/db543ea24374a00fe122cd5397ba6d9a80d74c83/src/rtl_433/pulse_demod.c#L92>

If you can get the decoder working with changes, what I will do the next release is include your version in the `contrib` directory similar to the LaCross decoder, my release process will overwrite the rtl_433 version

diepeterpan, 2023-01-21:
Thank you for the assistance, I will have a look into it.

diepeterpan, 2023-01-21:
I made the change to oregon_scientific and had two good runs, will let it run overnight and message you tomorrow. 

> On 21 Jan 2023, at 9:26 PM, Northern Man ***@***.***> wrote:
> 
> https://github.com/diepeterpan/rtl_433_ESP/blob/db543ea24374a00fe122cd5397ba6d9a80d74c83/src/rtl_433/devices/oregon_scientific.c#L227

diepeterpan, 2023-01-22:
Ran for 8 hours without a problem.  Thanks.

NorthernMan54, 2023-01-25:
@diepeterpan - Question for you, is this you on Hackaday ?

_So I ended using the old SRX882S v2.0 (and modified this library to receive from data pin of this module). The con thing is I no more have RSSI information, but compared to higher sensitivity of SRX882S, I choose this solution._

How did you wire the SRX882S ?

closed 2023-01-25 by NorthernMan54

diepeterpan, 2023-01-26:
I used a SX1278, not a SRX882S, but this is the wiring, sorry for the poor layout.

￼

TimelessNL, 2024-02-16:
@NorthernMan54 did you ever got/receive the code for the SRX882S? If the range is indeed better as is claimed in [that post](https://hackaday.com/2023/01/13/arduino-library-brings-rtl_433-to-the-esp32/#comment-6584152) it would be a nice addition I'd say.

NorthernMan54, 2024-02-17:
No never looked at it