NorthernMan54/rtl_433_ESP #53: Heltec Lora V1 compile issue

beckmx, 2023-01-22:
Hello guys, I tested yesterday the branch and all seems to compile succesfully as it is, but I have some old Heltec_wifi_lora v1 boards, so I wanted to try those, I change these lines:
```
[env:esp32_heltec]
monitor_filters = esp32_exception_decoder
board = heltec_wifi_lora_32
```
But it seems that the compiler fails with this:
```
Compiling .pio/build/esp32_heltec/libe6e/RadioLib/modules/SX126x/SX1261.cpp.o
Assembler messages:
Fatal error: can't create .pio/build/esp32_heltec/libe6e/RadioLib/modules/CC1101/CC1101.cpp.o: No such file or directory
*** [.pio/build/esp32_heltec/libe6e/RadioLib/modules/CC1101/CC1101.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_heltec/libe6e/RadioLib/modules/RF69/RF69.cpp.o: No such file or directory
*** [.pio/build/esp32_heltec/libe6e/RadioLib/modules/RF69/RF69.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_heltec/libe6e/RadioLib/modules/SX126x/STM32WLx.cpp.o: No such file or directory
*** [.pio/build/esp32_heltec/libe6e/RadioLib/modules/SX126x/STM32WLx.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_heltec/libe6e/RadioLib/modules/RFM9x/RFM97.cpp.o: No such file or directory
*** [.pio/build/esp32_heltec/libe6e/RadioLib/modules/RFM9x/RFM97.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/esp32_heltec/libe6e/RadioLib/modules/SX1231/SX1231.cpp.o: No such file or directory
*** [.pio/build/esp32_heltec/libe6e/RadioLib/modules/SX1231/SX1231.cpp.o] Error 1
```
the real big difference between the heltec v1 vs v2 is 2 connections to the lora radio, I also tried putting everything as it was and changing only the 2 flags I needed to change but it also raises an error after changing only this:
```
'-DRF_MODULE_DIO1=33'    ; SX1276 pin DIO1
;  '-DRF_MODULE_DIO2=32'    ; SX1276 pin DIO2
```
Has anybody managed to run under Heltec v1?

beckmx, 2023-01-22:
For future people, I am working on OSX, seems that the system now rejects to write directly on some folders, I had clone this project under my User folder which is 
`~/`

Also on OSX i have to click "build" on the Platformio IDE several times, but you cal also make it work with
`pio run`
Which at least for me always worked.

Finally to make work the Heltec wifi Lora v1, you will need to update the platformio.ini file in the esp32_heltec section with this board:
```
[env:esp32_heltec]
monitor_filters = esp32_exception_decoder
board = heltec_wifi_lora_32
```
Also uncoment these lines and update the pin numbers as here:

```
'-DRF_MODULE_DIO1=33'    ; SX1276 pin DIO1
  '-DRF_MODULE_DIO2=32'    ; SX1276 pin DIO2
```

Finally line 77 in the file rtl_433.h add this condition:
`#if defined(WIFI_LoRa_32_V2) || defined(WIFI_LoRa_32)// Heltec ESP32 Lora Board`

That allowed me to see my board working

closed 2023-01-22 by beckmx

NorthernMan54, 2023-01-22:
Tks for sharing

I’m surprised you had issues with OSX, as that is what I use.  Now I’m back level on releases so there may be some thing in the more recent releases.

in regards to rtl_433.h, if you include all the wiring compiler directives you should have been able to avoid editing the file.

beckmx, 2023-01-22:
Yes I tried, but even though, the problem with osX not able to write to the file system was there, kinda wierd