NorthernMan54/rtl_433_ESP #54: Acurite Atlas

beckmx, 2023-01-22:
Hello guys, I noticed some of the decoders include the acurite atlas, which is one of the first decoders on the list, but my setup is not able to decode anything so far, I just see this message:

`Received message : {"model":"unknown","protocol":"signal parsing failed","duration":177000,"rssi":-81,"pulses":807}`

I was wondering if I can share my trace to see if I am seeing another device or what exactly is going on

NorthernMan54, 2023-01-22:
I'm thinking it may be just slightly out of range, "rssi":-81

Have you tried enabling the rtl_433 verbose decode options

```
RTL_DEBUG					    ; Enable RTL_433 device decoder verbose mode for all device decoders ( 0=normal, 1=verbose, 2=verbose decoders, 3=debug decoders, 4=trace decoding. )
or
RTL_VERBOSE=##        ; Enable RTL_433 device decoder verbose mode, ## is the decoder # from the appropriate memcpy line in rtl_433_ESP.cpp
```

It may give future insite into the issue

beckmx, 2023-01-23:
Hello @NorthernMan54 yup, this is the output from my initial startup boot:
```

.ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
N:  
N: ****** setup ******
rtl_433_ESP(6): SX1278 gpio receive pin: 32
rtl_433_ESP(6): SX1278 receive frequency: 433.920013
rtl_433_ESP(6): # of device(s) configured 101
rtl_433_ESP(6): ssizeof(r_device): 108
rtl_433_ESP(6): cfg->devices size: 10908
rtl_433_ESP(6): # of device(s) enabled 100
Registering protocol [2] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [3] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas"
Registering protocol [4] "Acurite 986 Refrigerator / Freezer Thermometer"
Registering protocol [5] "Acurite 606TX Temperature Sensor"
Registering protocol [6] "Acurite 00275rm,00276rm Temp/Humidity with optional probe"
Registering protocol [7] "Acurite 590TX Temperature with optional Humidity"
Registering protocol [8] "Acurite Grill/Meat Thermometer 01185M"
Registering protocol [9] "Akhan 100F14 remote keyless entry"
Registering protocol [10] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [11] "Ambient Weather TX-8300 Temperature/Humidity Sensor"
Registering protocol [13] "Auriol AFT 77 B2 temperature sensor"
Registering protocol [14] "Auriol AFW2A1 temperature/humidity sensor"
Registering protocol [15] "Auriol AHFL temperature/humidity sensor"
Registering protocol [16] "Auriol HG02832, HG05124A-DCF, Rubicson 48957 temperature/humidity sensor"
Registering protocol [18] "Blyss DC5-UK-WH"
Registering protocol [20] "Bresser Thermo-/Hygro-Sensor 3CH"
Registering protocol [22] "Burnhard BBQ thermometer"
Registering protocol [23] "Calibeur RF-104 Sensor"
Registering protocol [24] "Cardin S466-TX2"
Registering protocol [25] "Chuango Security Technology"
Registering protocol [26] "Companion WTR001 Temperature Sensor"
Registering protocol [29] "Ecowitt Wireless Outdoor Thermometer WH53/WH0280/WH0281A"
Registering protocol [30] "Eurochron EFTH-800 temperature and humidity sensor"
Registering protocol [34] "Esperanza EWS"
Registering protocol [36] "Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor"
Registering protocol [37] "Fine Offset Electronics, WH0530 Temperature/Rain Sensor"
Registering protocol [38] "Fine Offset WH1050 Weather Station"
Registering protocol [39] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [41] "FT-004-B Temperature Sensor"
Registering protocol [42] "Generic wireless motion sensor"
Registering protocol [43] "Generic Remote SC226x EV1527"
Registering protocol [44] "Generic temperature sensor 1"
Registering protocol [45] "Govee Water Leak Dectector H5054, Door Contact Sensor B5023"
Registering protocol [46] "Globaltronics QUIGG GT-TMBBQ-05"
Registering protocol [47] "Globaltronics GT-WT-02 Sensor"
Registering protocol [48] "Globaltronics GT-WT-03 Sensor"
Registering protocol [49] "Microchip HCS200 KeeLoq Hopping Encoder based remotes"
Registering protocol [50] "Honeywell ActivLink, Wireless Doorbell"
Registering protocol [51] "HT680 Remote control"
Registering protocol [52] "inFactory, nor-tec, FreeTec NC-3982-913 temperature humidity sensor"
Registering protocol [54] "Interlogix GE UTC Security Devices"
Registering protocol [56] "Kedsum Temperature & Humidity Sensor, Pearl NC-7415"
Registering protocol [57] "Kerui PIR / Contact Sensor"
Registering protocol [58] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [59] "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor"
Registering protocol [60] "LaCrosse/ELV/Conrad WS7000/WS2500 weather sensors"
Registering protocol [61] "LaCrosse WS-2310 / WS-3600 Weather Station"
Registering protocol [63] "Markisol, E-Motion, BOFU, Rollerhouse, BF-30x, BF-415 curtain remote"
[... 219 lines cut ...]
rtl_433_ESP(7): demod(6) - Waveman Switch Transmitter
rtl_433_ESP(7): demod(6) - WG-PB12V1 Temperature Sensor
pulse_demod_pwm(): WG-PB12V1 Temperature Sensor Status: -2
rtl_433_ESP(7): demod(6) - WS2032 weather station
pulse_demod_pwm(): WS2032 weather station Status: -2
rtl_433_ESP(7): demod(5) - Hyundai WS SENZOR Remote Temperature Sensor
pulse_demod_ppm(): Hyundai WS SENZOR Remote Temperature Sensor Status: -1
rtl_433_ESP(7): demod(6) - WT0124 Pool Thermometer
pulse_demod_pwm(): WT0124 Pool Thermometer Status: -1
rtl_433_ESP(7): demod(5) - X10 RF
pulse_demod_ppm(): X10 RF Status: -1
rtl_433_ESP(7): demod(5) - X10 Security
pulse_demod_ppm(): X10 Security Status: -2
rtl_433_ESP(7): demod(5) - Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor
pulse_demod_ppm(): Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor Status: -2
rtl_433_ESP(7): demod(5) - Springfield Temperature and Soil Moisture
pulse_demod_ppm(): Springfield Temperature and Soil Moisture Status: 0
rtl_433_ESP(7): demod(6) - TFA Dostmann 30.3221.02 T/H Outdoor Sensor
pulse_demod_pwm(): TFA Dostmann 30.3221.02 T/H Outdoor Sensor Status: -2
rtl_433_ESP(6): Unparsed Signal length: 177000, Signal RSSI: -80, pulses: 443
[... cut ...] 55-96(-93)+53-75(-93)+97-56(-93)+57-61(-93)+52-59(-92)+68-64(-93)+105-147(-92)+55-69(-92)+63-90(-93)+52-51(-94)+57-54(-92)+57-82(-92)+52-61(-93)+68-207(-93)+55-170(-92)+88-53(-92)+53-75(-92)+56-89(-92)+63-53(-92)+69-55(-92)+118-98(-92)+52-53(-93)+51-67(-93)+63-69(-93)+55-69(-93)+52-53(-93)+84-119(-93)+58-63(-93)+86-53(-93)+63-53(-92) 
N: Received message : {"model":"unknown","protocol":"signal parsing failed","duration":177000,"rssi":-80,"pulses":443}
rtl_433_ESP(6): # of messages decoded 0

```

NorthernMan54, 2023-01-23:
This is from the decode attempt.  The Atlas is message s/b 80 bits long.  Not sure if it is the Atlas

```
acurite_txr_decode: row 0 bits 1, bytes 1 
acurite_txr_decode: row 1 bits 2, bytes 1 
acurite_txr_decode: row 2 bits 1, bytes 1 
acurite_txr_decode: row 3 bits 116, bytes 15 
acurite_txr_decode: skipping wrong len
acurite_txr_decode: row 4 bits 4, bytes 1 
acurite_txr_decode: row 5 bits 116, bytes 15 
acurite_txr_decode: skipping wrong len
acurite_txr_decode: row 6 bits 2, bytes 1 
acurite_txr_decode: row 7 bits 193, bytes 25 
acurite_txr_decode: skipping wrong len
pulse_demod_pwm(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas Status: 0
```

If you also enable `PUBLISH_UNPARSED`, it will log the received signal for further investigation.

beckmx, 2023-01-23:
I just got my first message fully readable, I am so excited 😆 

```
ulse_demod_ppm(): Acurite 606TX Temperature Sensor Status: -2
rtl_433_ESP(7): demod(6) - Acurite 00275rm,00276rm Temp/Humidity with optional probe
rtl_433_ESP(6): data_output {"model":"Acurite-00276rm","subtype":0,"id":5849,"battery_ok":0,"temperature_C":21.2,"humidity":48,"mic":"CRC","protocol":"Acurite 00275rm,00276rm Temp/Humidity with optional probe","rssi":-72,"duration":177000}
N: Received message : {"model":"Acurite-00276rm","subtype":0,"id":5849,"battery_ok":0,"temperature_C":21.2,"humidity":48,"mic":"CRC","protocol":"Acurite 00275rm,00276rm Temp/Humidity with optional probe","rssi":-72,"duration":177000}
pulse_demod_pwm(): Acurite 00275rm,00276rm Temp/Humidity with optional probe
bitbuffer:: Number of rows: 3 
[00] {90} 00 16 9a d9 4b c0 6a 18 00 00 11 40 
[01] {88} 00 16 9a d9 4b c0 6c 30 00 00 45 
[02] {174} 00 0b 4d 6c a5 c0 1b 0c 00 00 11 00 00 00 00 00 00 00 00 00 00 00 
```

The other from the Atlas, I dont know if it is because of the timing or what, but certainly I did read something

beckmx, 2023-01-23:
I just captured the atlas:

```
N: Received message : {"model":"Acurite-Tower","id":520,"channel":"C","battery_ok":0,"temperature_C":516.0,"humidity":33,"mic":"CHECKSUM","protocol":"Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas","rssi":-78,"duration":176000}
acurite_txr_decode: row 7 bits 68, bytes 9 
acurite_txr_decode: skipping wrong len
acurite_txr_decode: row 8 bits 3, bytes 1 
acurite_txr_decode: row 9 bits 2, bytes 1 
acurite_txr_decode: row 10 bits 1, bytes 1 
acurite_txr_decode: row 11 bits 1, bytes 1 
acurite_txr_decode: row 12 bits 208, bytes 26 
acurite_txr_decode: skipping wrong len
pulse_demod_pwm(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas
bitbuffer:: Number of rows: 13 
[00] { 2} 00                                                                            : 00
[01] {89} 00 14 a9 91 49 30 2d 40 30 00 00 00 
[02] {33} 00 00 60 24 80                                                                : 00000000 00000000 01100000 00100100 1
[03] { 3} a0                                                                            : 101
[04] { 1} 80                                                                            : 1
[05] { 1} 00                                                                            : 0
[06] {56} 02 08 a8 21 b0 10 93 
[07] {68} c0 0d 62 32 10 10 08 01 10 
[08] { 3} 20                                                                            : 001
[09] { 2} 00                                                                            : 00
[10] { 1} 00                                                                            : 0
[11] { 1} 00                                                                            : 0
[12] {208} 00 01 04 10 00 28 01 03 08 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
rtl_433_ESP(7): demod(5) - Acurite 986 Refrigerator / Freezer Thermometer
```

daaaamn!

beckmx, 2023-01-25:
I noticed that I stopped receiving messages from the acurite atlas, I saw the message when most flags were activated for debugging, now I havent received it, I left a recording uart file but I dont see it in there, I placed the antenna closer to the sender

NorthernMan54, 2023-01-26:
If you add `PUBLISH_UNPARSED`, it should also log the raw undecoded signal, which may be worth looking at.  To see what is being received.

What's odd is that I have the 'Acurite Tower', with its multiple sensors, which uses the same decoder and it works fine for me.

One comment, is you device positioned with the antenna sticking straight up?  I find that position makes a difference

beckmx, 2023-01-26:
Hello, the antenna is pointing upwards, in my original test I was testing in my table and then moved it to another place where the cable that connects to the antenna is hanging a little down but pointing up, I am going to try leveling everything up again. And about the sensor, I also have another sensor, it might be possible that both sensors (both from AcuRite) are stepping each other message?

beckmx, 2023-01-29:
@NorthernMan54 quick question, if I use a 868 915 MHz can the decoders work? or they really have a dependency against 433mhz? (it's for other sensor I have)

NorthernMan54, 2023-01-29:
Their is no dependency on frequency, you can change it as needed

beckmx, 2023-03-29:
I left the script working for days, turns out that the antenna was not “working” properly I discovered the antenna was the problem when after reassembling my setup all of a sudden I was getting a lot more messages, including the Atlas, serves a lot if you have a device that is pushing info. So everything is working just fine now

closed 2023-03-29 by beckmx

NorthernMan54, 2023-03-29:
Glad to hear that you got this sorted