NorthernMan54/rtl_433_ESP #55: Support for 868MHz ESP32 LoRa ?

MadMacMad, 2023-01-30:
hello,
well by any chance is there a way to support the 868MHz ESP32 LoRa boards for Bresser 6 in 1 sensors ?
the rtl_433 does support the 868MHz ...
many thx

NorthernMan54, 2023-01-30:
The Bresser 6 in 1 uses FSK modulation, which is not supported - https://github.com/merbanan/rtl_433/blob/3597af8221b73e485cdeda0cccfa03fc171e1fad/src/devices/bresser_6in1.c#L243

The 868Mhz ESP32 Lora boards as long as they are the sx127x chipset should be supported.

ristomatti, 2023-02-19:
Should a weather station work with SX127x 868M Lora boards that I've been using with a RTL_433 w/ SDR dongle? It's detected by "Fine Offset Electronics WH1080/WH3080 Weather Station" decoder? I'd test right away unless the weather station was not at our cottage.. :smile:

ristomatti, 2023-02-19:
Answering myself: WH1080 seems to be using either OOK_PWM or FSK_PCM depending on the variant (https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_wh1080.c). I hope ours is using FSK. 🤞

CazYokoyama, 2023-02-20:
> The Bresser 6 in 1 uses FSK modulation, which is not supported - https://github.com/merbanan/rtl_433/blob/3597af8221b73e485cdeda0cccfa03fc171e1fad/src/devices/bresser_6in1.c#L243
> 
> The 868Mhz ESP32 Lora boards as long as they are the sx127x chipset should be supported.

Is there anything I can do to support FSK modulation? I have LaCrosse Technology LTV-WSDTH01 Breeze Pro. rtl_433/src/devices/lacrosse_breezepro.c tells "FSK_PCM on a center frequency of 914.938 MHz". I am glad if you point me where I should look at.

NorthernMan54, 2023-02-22:
FSK Modulated signals are a challenge, as their are multiple FSK modulations, with unique receiver settings for each.  Hence the lack of progress with FSK modulation.  So creating a general receiver for OOK and FSK modulated signals is not feasible. Also on a personal nature I do not have access to a device that uses FSK modulation, so that is also preventing progress.

CazYokoyama, 2023-02-22:
Yes, I found your fsk branch. Let me look at my box where some La Crosse transmitter.

I am modifing example/OOK_Receiver/OOK_Receiver.ino for FSK/915MHz. Is this another good approach? Or working at your fsk branch is better idea?

NorthernMan54, 2023-02-22:
That fsk branch and example are failed attempts to get it working from a few years ago, and I don’t recall the exact status except non functional

CazYokoyama, 2023-03-21:
@NorthernMan54 I created https://github.com/NorthernMan54/rtl_433_ESP/pull/66. It does not ask pull. It is the status report pf FSK 915MHz.

NorthernMan54, 2024-11-30:
I refreshed the device decoders from rtl_433 24.10, maybe this is resolved?

closed 2024-11-30 by NorthernMan54