NorthernMan54/rtl_433_ESP #58: 315MHz pre made lora boards

richms, 2023-02-21:
Has anyone tried the 433MHz lilygo board as a 315MHz reciever, or found another pre-made board that will work for 315MHz generic keychain remotes with this?

NorthernMan54, 2023-02-22:
At 315Mhz I have only used a CC1101.

z-l-p, 2023-10-09:
A few more details: I also couldn't receive 315MHz on a LilyGo Lora32 (T3 v 2.1, also known as v1.6.1). I'm decoding TPMS tire pressure sensors, so FSK mode. 
No errors at compile or runtime, but nothing is ever received. The SX1278 chip supports 137 - 525 MHz but the datasheet specifies 2 bands: 137-175MHZ and 410-525MHz. That's not reflected in Radiolib, where setFrequency() accepts the whole 137-525 range without complaint. Oh well, back to the CC1101!