NorthernMan54/rtl_433_ESP #61: Specify MALLOC_CAP_INTERNAL while calloc()ing heap

SmittyHalibut, 2023-02-27:
https://github.com/NorthernMan54/rtl_433_ESP/blob/master/src/rtl_433_ESP.cpp#L409

I was getting `calloc()` failures, even though there was enough heap reported by `ESP.getFreeHeap()` and `ESP.getMaxAllocHeap()`.

Those two reporting tools however are calling `heap_caps_` functions, and hard coding a capability of `MALLOC_CAP_INTERNAL` to force reporting heap for Internal memory only.

I tested by detecting a `calloc(...)` failure, then trying a `heap_cap_calloc(..., MALLOC_CAP_INTERNAL)` which always succeeded.

So I propose replacing `calloc(..)` with `heap_caps_calloc(..., MALLOC_CAP_INTERNAL)`

Pull request incoming. :-)

NorthernMan54, 2023-02-27:
For this issue, any thoughts on how I could recreate it? During testing I never ran into this.

I won’t be able to merge this for a few weeks, as I’m out of the country, and don’t have access to my test environment

SmittyHalibut, 2023-02-27:
It came up for me when I enabled both ZgatewayBT and ZgatewayRTL_433 at the same time in OpenMQTTGateway. I suspect that BT is also using an inspecific malloc() scheme and it's defaulting to something other than INTERNAL that's smaller than INTERNAL.

SmittyHalibut, 2023-03-13:
Its not confirmed yet, but https://github.com/1technophile/OpenMQTTGateway/issues/1529 MIGHT be the same problem I was running into yet.  [Their](https://github.com/Andyriod) reported symptoms sound similar, after doing the same thing I did.  Soooo, maybe chances are good that ticket is reproduction?

NorthernMan54, 2023-04-03:
This has been implemented

closed 2023-04-03 by NorthernMan54