NorthernMan54/rtl_433_ESP #63: Upload to Wunderground

CazYokoyama, 2023-03-08:
The master branch as of 2023-0307 reports Acurite-5n1 when I power-on Lilygo TTGO LOra32 which has the binary built by esp32_lilygo. Now it is the time to upload the decoded data to yunderground.
Code section is 30% of the flash memory in Lora32. There is enough memory to implement uploading to wunderground.

CazYokoyama, 2023-03-10:
I am working on. Will create pull-request. The code will be in example directory.

NorthernMan54, 2023-03-10:
@CazYokoyama was thinking about this further, and the package is meant to be a library invoked by other programs or routines.  You could make your own project/package and just include this as a library similar to how the integration to OpenMQTTGateway.  Or my earlier suggestion about a OpenMQTTGateway package.

CazYokoyama, 2023-03-11:
I am not familiar with OpenMQTTGateway.
https://www.youtube.com/watch?v=_gdXR1uklaY tells OpenMQTTGateway passes the data from weather station to Internet. Does it mean OpenMQTTGateway already implements what I want except for uploading wunderground? Could you clarify my understandings?

DigiH, 2023-03-11:
OpenMQTTGateway uses different libraries, rf , like rtl_433_ESP, Bluetooth, IR, to capture and decode device transmission/broadcasts to then publish their data via MQTT. 

Subscribing to these MQTT messages then allows different software, like home automation controllers or other projects, to process or pass on these values as desired.

For your Wunderground  example something like this might be an already existing solution

https://github.com/bdemers/mqtt-wunderground-publish

I'm sure with a bit of looking around an searching several others in that direction might turn up.

Hope this helps.

CazYokoyama, 2023-03-14:
Successfully upload the data from Acurite-5n1 to https://www.wunderground.com. Thank you.

NorthernMan54, 2023-03-15:
Very nice

closed 2023-03-15 by CazYokoyama