NorthernMan54/rtl_433_ESP #72: Add support for updated Govee H5054 water leak detector protocol

spectrumjade, 2023-03-25:
I have a handful of these devices that I purchased at different times, and I've found that some are detected by rtl_433_ESP and some are not. I believe the reason for this is because Govee updated the protocol a couple years ago and the new protocol is not yet supported in rtl_433_ESP.

rtl_433 does detect both, however they are registered as separate protocols:
```
# Detected by rtl_433_ESP
time      : 2023-03-25T14:51:11                    Protocol  : 192
model     : Govee-Water  id        : 62878
event     : Water Leak   Raw Code  : 0a610404074a  Integrity : PARITY

# Not detected by rtl_433_ESP
time      : 2023-03-25T14:57:07                    Protocol  : 231
model     : Govee-Water  id        : 16132
event     : Water Leak   Leak Num  : 2             Raw Code  : 3f0432026962  Integrity : CRC
```
I found a PR which recently added support for this upstream: https://github.com/merbanan/rtl_433/pull/2273

Could support for this new protocol be added?

NorthernMan54, 2023-03-26:
I'm in the process of updating the device decoders to the most recent rtl_433 release, and will include this.  Please watch this branch

 https://github.com/NorthernMan54/rtl_433_ESP/tree/rtl_433-22.11

NorthernMan54, 2023-04-01:
Pls see #75

rodpayne, 2023-04-03:
My recently purchased Govee H5054 water detectors are being detected with `version 0.2.0` (as well as the` rtl_433-22.11 branch`).  Thank you for your work.

Hopefully it won't be long before we can get them working in OpenMQTTGateway.  

It seems that there are some breaking changes with this new version.  Do you have any advice on changes that need to be made?  Things like `rtl_433.enableReceiver(-1)`?

NorthernMan54, 2023-04-04:
I'm just letting this simmer for a few days before creating the OMG pull request

This pull request contains the OMG changes and other wonderful things

https://github.com/1technophile/OpenMQTTGateway/pull/1544