NorthernMan54/rtl_433_ESP #74: Nodejs small console

beckmx, 2023-03-30:
I wanted to share with you guys a script I made for displaying in a more comfortable way the incoming signals, I made several so I will try to update this if I make some changes. A possible PR can be done of course.

This script was made for raspberry but you can easily swap the serial port dependencies for other ones.

The console looks like the image attached:

[image]

The JS script makes 2 main things:
1) Connects to the Serial port and starts listening to the output data, it doesnt matter if you configured the verbose, it will try to locate only the json objects
2) setups a WebSocket server so that you can connect to the data feed which only outputs the json objects
3) it displays a table of the objects it has received

```
const WebSocket = require("ws");
const Serial = require('raspi-serial').Serial;
const fs = require("fs");
const Table = require("cli-table3");

const SERIAL_PORT = "/dev/ttyUSB0"; // Change this to your serial port
const BAUD_RATE = 921600;
const WS_PORT = 8080;

const clients = new Set(); // Maintain a set of connected WebSocket clients
let buffer = ""; // Buffer to store incoming serial data
const modelCount = {}; // Object to store the count of each model and last received time

function truncateString(str, maxLength) {
    if (str.length > maxLength) {
      return str.substring(0, maxLength - 3) + "...";
    }
    return str;
  }
  
function updateTable(model, jsonObj) {
const currentDateTime = new Date().toLocaleString();
const lastMessage = truncateString(JSON.stringify(jsonObj), 50);

if (modelCount[model]) {
    modelCount[model].count++;
    modelCount[model].time = currentDateTime;
    modelCount[model].lastMessage = lastMessage;
} else {
    modelCount[model] = { count: 1, time: currentDateTime, lastMessage: lastMessage };
}

const table = new Table({
    head: ["Model", "Total Detected", "Last Time Received", "Last Message"],
});

for (const [key, value] of Object.entries(modelCount)) {
    table.push([key, value.count, value.time, value.lastMessage]);
}

console.clear();
console.log(table.toString());
}
  

const wss = new WebSocket.Server({ host: "0.0.0.0", port: WS_PORT }, () => {
  console.log(`WebSocket server started on ws://0.0.0.0:${WS_PORT}`);
});

const serialPort =  new Serial({portId:SERIAL_PORT,baudRate:BAUD_RATE});
serialPort.open(() => {
  serialPort.on("data", (data) => {
    // process.stdout.write(data);
    buffer += data.toString();
    let startIdx = buffer.indexOf("{");
    let endIdx = buffer.indexOf("}");

    while (startIdx !== -1 && endIdx !== -1) {
      try {
        const jsonStr = buffer.slice(startIdx, endIdx + 1);
        const jsonObj = JSON.parse(jsonStr);

        updateTable(jsonObj.model, jsonObj); // Update and display the ASCII table

        for (const client of clients) {
          if (client.readyState === WebSocket.OPEN) {
            client.send(JSON.stringify(jsonObj));
          }
        }

        buffer = buffer.slice(endIdx + 1); // Remove the processed JSON from the buffer
      } catch (error) {
        break;
      }

      startIdx = buffer.indexOf("{");
      endIdx = buffer.indexOf("}");
    }

    fs.appendFileSync('captures.txt',data, 'utf8');
  });
});

wss.on("connection", (ws) => {
  clients.add(ws); // Add the new client to the set
  console.log("WebSocket connected");

  ws.on("close", () => {
    clients.delete(ws); // Remove the client from the set
    console.log("WebSocket disconnected");
  });
});

```

beckmx, 2023-03-30:
This next script is a python file, basically if you feed it with a txt file with all the data it has recorded from the verbosed output of the serial port, it can give you a table of the objects found:

[image]

The code can be found below:
```
import re
import json
from collections import defaultdict
from prettytable import PrettyTable

def extract_json_objects(text):
    json_objects = []
    open_brackets = 0
    start_pos = -1
    
    for i, char in enumerate(text):
        if char == '{':
            if open_brackets == 0:
                start_pos = i
            open_brackets += 1
        elif char == '}':
            open_brackets -= 1
            if open_brackets == 0:
                json_object = text[start_pos:i + 1]
                print(f"Extracted JSON object: {json_object}")  # Debug print
                json_objects.append(json_object)

    return json_objects

def count_models(text):
    model_count = defaultdict(int)
    json_objects = extract_json_objects(text)

    for obj_str in json_objects:
        try:
            data = json.loads(obj_str)
            model = data["model"]
            model_count[model] += 1
        except (json.JSONDecodeError, KeyError) as e:
            print(f"Error parsing JSON object: {obj_str}, Error: {e}")  # Debug print
            pass
    return model_count

def display_table(model_count):
    table = PrettyTable(["Model", "Total Messages"])
    for model, count in model_count.items():
        table.add_row([model, count])
    print(table)

if __name__ == "__main__":
    file_path = "captures.txt"  # Change this to your file path

    with open(file_path, "r") as file:
        text = file.read()

    model_count = count_models(text)
    display_table(model_count)

```

NorthernMan54, 2023-03-30:
Why not just run rtl_433_ESP inside of openMQTTGateway, and use a mqtt server ?

NorthernMan54, 2023-03-30:
Now that makes perfect sense then.

tks for sharing

closed 2023-04-02 by beckmx