NorthernMan54/rtl_433_ESP #76: Retransmit packages

beckmx, 2023-04-02:
Hello there, I was wondering if is it possible to retransmit packages, I noticed my weather station which is very far from the receiver some times loses reception because of the weather. so I was wondering if I can retransmit, maybe not directly with the library, but if someone points me in the right direction i can go hands on myself

thanx

NorthernMan54, 2023-04-19:
Have you looked at some of the stuff with dual CC1101 stuff ?

https://github.com/joelsernamoreno/EvilCrow-RF

Personally I just ran a long ethernet cable and power to a shed at the back of my property ( I got household approval as I said we needed a light in the shed ), and put a wifi hotspot out there.   As I got approval, we rented a ditch witch and installed dual conduit etc.

beckmx, 2023-04-19:
btw, I was going to send a PR for a heltec 915mhz board (lora stick v3) but didnt work :(

closed 2023-04-19 by beckmx