NorthernMan54/rtl_433_ESP #78: Board heltec_wifi_lora_32

jurkov, 2023-04-12:
I tried to get the older version of the 433mhz lora heltec working. The board name is heltec_wifi_lora_32. I seems not to detect anything. Do you have any idea what i could try?

`[env:esp32_heltec_v1]
board = heltec_wifi_lora_32
build_flags = 
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
  '-DONBOARD_LED=25'          ; Onboard LED is GPIO 25 on the Heltec Board
; *** rtl_433_ESP Options ***
;  '-DRTL_DEBUG=4'           ; rtl_433 verbose mode
  '-DRTL_VERBOSE=58'          ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
;  '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages
;  '-DMEMORY_DEBUG=true'   ; display memory usage information
  '-DDEMOD_DEBUG=true'  ; display signal debug info
;	'-DMY_DEVICES=true'		; subset of devices
  '-DPUBLISH_UNPARSED=true'   ; publish unparsed signal details
  '-DRSSI_THRESHOLD=12'         ; Apply a delta of 12
  '-DOOK_FIXED_THRESHOLD=0x50'  ; Inital OOK Threhold - Only for SX127X
;  '-DAVERAGE_RSSI=5000'     ; Display RSSI floor ( Average of 5000 samples )
  '-DSIGNAL_RSSI=true'             ; Display during signal receive
; *** RF Module Options ***
;  '-DRF_SX1278="SX1278"'   ; Heltec ESP 32 Module - module settings come from heltec_wifi_lora_32_V2/pins_arduino.h
;  '-DRF_MODULE_DIO0=26'    ; SX1276 pin DIO0
;  '-DRF_MODULE_DIO1=35'    ; SX1276 pin DIO1
;  '-DRF_MODULE_DIO2=34'    ; SX1276 pin DIO2
;  '-DRF_MODULE_RST=14'     ; pin to be used as hardware reset
  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
; *** Heltec module requires non-standard SPI Config ***
;  '-DRF_MODULE_CS=18'      ; pin to be used as chip select
;  '-DRF_MODULE_MOSI=27'
;  '-DRF_MODULE_MISO=19'
;  '-DRF_MODULE_SCK=5'
; *** RadioLib Options ***
  '-DRADIOLIB_DEBUG=true'
;  '-DRADIOLIB_VERBOSE=true'
monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 115200
upload_port = /dev/cu.SLAB_USBtoUART
upload_speed = 921600
`

NorthernMan54, 2023-04-12:
When it starts, what is logged ?  It should display an error if it can't reach the transceiver module.

jurkov, 2023-04-13:
Maybe it can reach the transceiver module. Here is what I get:
```
18:27:30.269 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41071, Gap length: 5217612, Signal RSSI: -101, Current RSSI: -125, pulses: 0, noise count: 4
18:27:30.720 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41071, Gap length: 400700, Signal RSSI: -101, Current RSSI: -113, pulses: 0, noise count: 4
18:27:30.816 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41071, Gap length: 56788, Signal RSSI: -101, Current RSSI: -112, pulses: 0, noise count: 4
18:27:31.204 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41071, Gap length: 330875, Signal RSSI: -101, Current RSSI: -117, pulses: 0, noise count: 4
18:27:34.103 -> rtl_433_ESP(6): Ignored Signal length: 46000, Time since last bit length: 40080, Gap length: 2803788, Signal RSSI: -101, Current RSSI: -110, pulses: 0, noise count: 4
18:27:37.484 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41071, Gap length: 3334347, Signal RSSI: -101, Current RSSI: -108, pulses: 0, noise count: 4
18:27:57.932 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41071, Gap length: 20408700, Signal RSSI: -101, Current RSSI: -121, pulses: 0, noise count: 4
18:28:01.668 -> rtl_433_ESP(7): Average RSSI Signal -115 dbm, adjusted RSSI Threshold -103, samples 50000
18:28:05.068 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41079, Gap length: 7105612, Signal RSSI: -102, Current RSSI: -113, pulses: 0, noise count: 4
18:28:05.294 -> rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 41071, Gap length: 152692, Signal RSSI: -101, Current RSSI: -112, pulses: 0, noise count: 4
18:28:07.095 -> rtl_433_ESP(6): Ignored Signal length: 11000, Time since last bit length: 40080, Gap length: 1774788, Signal RSSI: -101, Current RSSI: -107, pulses: 0, noise count: 4
```
This is the product and its antenna: https://www.hackerspace-ffm.de/wiki/index.php?title=Heltec_Wifi_LoRa_32

NorthernMan54, 2023-04-13:
That's a good sign that is connecting to the transceiver module and picking up something.  But it appears to be picking up noise.

Signal RSSI: -102 - is a little bit of noise as no pulses were detected ( pulses: 0 )

I'm thinking that the board may not be setup for 433 Mhz so you are out of luck as a 433 MHz receiver.

Another use case for it, is a bluetooth receiver, OpenMQTTGateway has a preset package for this board with bluetooth decoding

jurkov, 2023-04-13:
Hmmm. It is aleast an 433Mhz device. Here is another serial dump:
```
N:
    N: ****** setup ******
                          rtl_433_ESP(6): SX1278 gpio receive pin: 32
rtl_433_ESP(6): SX1278 receive frequency: 433.920013
rtl_433_ESP(6): # of device(s) configured 157
rtl_433_ESP(6): ssizeof(r_device): 112
rtl_433_ESP(6): cfg->devices size: 17584
M       SX127x
----- SX127x Status -----
RegOpMode: 0x2c
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x2a
RegRxBw: 0x0b
RegAfcBw: 0x02
-------------------------
RegOokPeak: 0x08
RegOokFix: 0x50
RegOokAvg: 0x12
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----- SX127x Status -----
N: ****** setup complete ******
                               ----- SX127x Status -----
RegOpMode: 0x2d
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x2a
RegRxBw: 0x0b
RegAfcBw: 0x02
-------------------------
RegOokPeak: 0x08
RegOokFix: 0x50
RegOokAvg: 0x12
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----- SX127x Status -----
rtl_433_ESP(7): Average RSSI Signal -101 dbm, adjusted RSSI Threshold -89, samples 50000
rtl_433_ESP(6): Ignored Signal length: 6, Time since last bit length: 41158, Gap length: 53069742, Signal RSSI: -87, Current RSSI: -99, pulses: 22, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 41075, Gap length: 3397589, Signal RSSI: -82, Current RSSI: -95, pulses: 22, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 41071, Gap length: 18609784, Signal RSSI: -87, Current RSSI: -101, pulses: 11, noise count: 2
rtl_433_ESP(7): Average RSSI Signal -101 dbm, adjusted RSSI Threshold -89, samples 50000
rtl_433_ESP(7): Average RSSI Signal -101 dbm, adjusted RSSI Threshold -89, samples 50000
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 41079, Gap length: 109576612, Signal RSSI: -88, Current RSSI: -99, pulses: 15, noise count: 2
rtl_433_ESP(7): Average RSSI Signal -102 dbm, adjusted RSSI Threshold -90, samples 50000
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 41079, Gap length: 22430604, Signal RSSI: -88, Current RSSI: -118, pulses: 0, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 41071, Gap length: 1173692, Signal RSSI: -89, Current RSSI: -98, pulses: 1, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 25000, Time since last bit length: 40081, Gap length: 9875, Signal RSSI: -89, Current RSSI: -97, pulses: 5, noise count: 4
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 41071, Gap length: 269786, Signal RSSI: -89, Current RSSI: -101, pulses: 5, noise count: 60
rtl_433_ESP(6): Signal length: 185000, Gap length: 64788, Signal RSSI: -89, train: 0, messageCount: 0, pulses: 21
gasmate_ba1008_decode: Length check fail
pulse_slicer_ppm: Gasmate BA1008 meat thermometer codes [{2}8]
gasmate_ba1008_decode: Length check fail
pulse_slicer_ppm: Gasmate BA1008 meat thermometer codes [{1}8]
gasmate_ba1008_decode: Length check fail
pulse_slicer_ppm: Gasmate BA1008 meat thermometer codes [{1}0]
gasmate_ba1008_decode: Length check fail
pulse_slicer_ppm: Gasmate BA1008 meat thermometer codes [{1}0]
gasmate_ba1008_decode: Length check fail
pulse_slicer_ppm: Gasmate BA1008 meat thermometer codes [{1}8]
rtl_433_ESP(6): Unparsed Signal length: 185000, Signal RSSI: -89, pulses: 22
rtl_433_ESP(6): RAW (185000): +86-9378(-93)+53-30026(-88)+53-17867(-102)+53-1547(-101)+53-53(-86)+54-7893(-86)+53-12374(-96)+53-1547rtl_433_ESP(6): (-100)+53-4534(-104)+53-2026(-95)+54-27999(-93)+53-53(-93)+54-9173(-92)+53-14400(-98)+54-7466(-90)+54-3946(-100)Ignored Signal length: 14001, Time since last bit length: 57897+53, Gap length: 2699-17920(-110)+53-535(-110)+52-11893(-93)+53, Signal RSSI: -89-3894(-88)+53-27520(-94)+53-1920(-94)
, Current RSSNI: -100, pulses: 10, noise count: 96
: Received message : {"model":"unknown","protocol":"signal parsing failed","duration":185000,"rssi":-89,"pulses":22}
                                                                                                                    rtl_433_ESP(6): # of messages decoded 0

```

beckmx, 2023-04-20:
i have that same device, antenna position and contact is very important

NorthernMan54, 2023-04-20:
Did you look at the temperature it reported, "temperature_C":-150.2,  are you in deep space or something ?

closed 2024-11-30 by NorthernMan54