NorthernMan54/rtl_433_ESP #83: Support for WiFi LoRa 32 (V2.1) - 915mhz version?

pdkwork, 2023-05-04:
I have been trying to test this chip set with your example OOk_Receiver.  I converted the OOK_Receiver.ino to a main.cpp file.  I changed the frequency in this file by setting #define RF_MODULE_FREQUENCY 915.0.

Below is the setup in the platformio.ini file and below that is some of the output.  Every signal it detects is ignored.  I see the Signal RSSI is low but it is getting pulses.  I don't see in the log file any reference to devices being enabled as shown in your example log file nor do I see the receiver task being started.  

I have a RTL-SDL dongle that receives signals from my water meter (Neptune R900 protocol), my gas meter (scm protocol) and my electric meter (scm+) protocol.  I assume that I don't have it configured correctly since all I have is the chip powered off of usb, ie no wiring needed.

Does this only work with 433mhz?

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:heltec_wifi_lora_32_V2]
platform = espressif32
board = heltec_wifi_lora_32_V2
framework = arduino
build_flags = 
;  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
  '-DONBOARD_LED=LED_BUILTIN'          ; Onboard LED is GPIO 25 on the Heltec Board
; *** rtl_433_ESP Options ***
;  '-DRTL_DEBUG=4'           ; rtl_433 verbose mode
;  '-DRTL_VERBOSE=58'          ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
;  '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages
;  '-DMEMORY_DEBUG=true'   ; display memory usage information
  '-DDEMOD_DEBUG=true'  ; display signal debug info
;  '-DMY_DEVICES=true'		; subset of devices
  '-DPUBLISH_UNPARSED=true'   ; publish unparsed signal details
;  '-DRSSI_THRESHOLD=12'         ; Apply a delta of 12
;  '-DOOK_FIXED_THRESHOLD=0x50'  ; Inital OOK Threhold - Only for SX127X
;  '-DAVERAGE_RSSI=5000'     ; Display RSSI floor ( Average of 5000 samples )
;  '-DSIGNAL_RSSI=true'             ; Display during signal receive
; *** RF Module Options ***
  '-DRF_SX1278="SX1278"'   ; Heltec ESP 32 Module - module settings come from heltec_wifi_lora_32_V2/pins_arduino.h
  '-DRF_MODULE_DIO0=26'    ; SX1276 pin DIO0
  '-DRF_MODULE_DIO1=35'    ; SX1276 pin DIO1
  '-DRF_MODULE_DIO2=34'    ; SX1276 pin DIO2
  '-DRF_MODULE_RST=14'     ; pin to be used as hardware reset
  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
; *** Heltec module requires non-standard SPI Config ***
  '-DRF_MODULE_CS=18'      ; pin to be used as chip select
  '-DRF_MODULE_MOSI=27'
  '-DRF_MODULE_MISO=19'
  '-DRF_MODULE_SCK=5'
; *** RadioLib Options ***
;  '-DRADIOLIB_DEBUG=true'
  '-DRADIOLIB_VERBOSE=true'
; *** Setting Testing ***
  ;'-DsetBitrate'
  ;'-DsetFrequencyDeviation'
  ;'-DsetRxBW'
monitor_speed = 115200
lib_deps = 
	jgromes/RadioLib@^6.0.0
	thijse/ArduinoLog@^1.1.1
	bblanchon/ArduinoJson@^6.21.2

This is the output:
I:  
I: ****** setup ******
rtl_433_ESP(6): SX1278 gpio receive pin: 34
rtl_433_ESP(6): SX1278 receive frequency: 915.000000
rtl_433_ESP(6): # of device(s) configured 157
rtl_433_ESP(6): ssizeof(r_device): 112
rtl_433_ESP(6): cfg->devices size: 17584
rtl_433_ESP(7): rtl_433_DecoderTask awaiting signal
rtl_433_ESP(6): SX1278 SPI Config SCK: 5, MISO: 19, MOSI: 27, CS: 18
rtl_433_ESP(6): SX1278 radio.begin() - success!
rtl_433_ESP(6): SX1278 setOOK - success!
rtl_433_ESP(6): SX1278 setCrcFiltering - success!
rtl_433_ESP(6): SX1278 setDataShapingOOK - success!
rtl_433_ESP(6): SX1278 OOK Thresh PEAK - success!
rtl_433_ESP(6): SX1278 OOK PEAK Thresh Decrement - success!
rtl_433_ESP(6): SX1278 Ook Peak Threshold Step - success!
rtl_433_ESP(6): SX1278 OokFixedThreshold - success!
rtl_433_ESP(6): SX1278 setBitRate - success!
rtl_433_ESP(6): SX1278 setRxBandwidth - success!
rtl_433_ESP(6): SX1278 RSSI Smoothing - success!
rtl_433_ESP(6): SX1278 preamble detect off - success!
rtl_433_ESP(6): SX1278 setDirectSyncWord - success!
rtl_433_ESP(6): SX1278 disableBitSync - success!
rtl_433_ESP(6): SX1278 receiveDirect - success!
----- SX127x Status -----
RegOpMode: 0x2d
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x2a
RegRxBw: 0x01
RegAfcBw: 0x02
-------------------------
RegOokPeak: 0x08
RegOokFix: 0x0f
RegOokAvg: 0x12
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----- SX127x Status -----
I: ****** setup complete ******
----- SX127x Status -----
RegOpMode: 0x29
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x2a
RegRxBw: 0x01
RegAfcBw: 0x02
-------------------------
RegOokPeak: 0x08
RegOokFix: 0x0f
RegOokAvg: 0x12
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----- SX127x Status -----
rtl_433_ESP(7): Average RSSI Signal -116 dbm, adjusted RSSI Threshold -107, samples 50000
rtl_433_ESP(6): Ignored Signal length: 9, Time since last bit length: 42158, Gap length: 120878787, Signal RSSI: -106, Current RSSI: -114, pulses: 14, noise count: 0
rtl_433_ESP(7): Average RSSI Signal -116 dbm, adjusted RSSI Threshold -107, samples 50000
rtl_433_ESP(6): Ignored Signal length: 6, Time since last bit length: 42087, Gap length: 85628358, Signal RSSI: -106, Current RSSI: -116, pulses: 16, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 42098, Gap length: 33375530, Signal RSSI: -106, Current RSSI: -116, pulses: 19, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42098, Gap length: 19391509, Signal RSSI: -106, Current RSSI: -121, pulses: 18, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42099, Gap length: 10427509, Signal RSSI: -106, Current RSSI: -124, pulses: 20, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42099, Gap length: 25499509, Signal RSSI: -106, Current RSSI: -107, pulses: 20, noise count: 0
rtl_433_ESP(7): Average RSSI Signal -116 dbm, adjusted RSSI Threshold -107, samples 50000
rtl_433_ESP(7): Average RSSI Signal -116 dbm, adjusted RSSI Threshold -107, samples 50000
rtl_433_ESP(6): Ignored Signal length: 5, Time since last bit length: 42090, Gap length: 111889510, Signal RSSI: -106, Current RSSI: -111, pulses: 19, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42098, Gap length: 2757438, Signal RSSI: -106, Current RSSI: -108, pulses: 18, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 1, Time since last bit length: 42098, Gap length: 10753594, Signal RSSI: -106, Current RSSI: -118, pulses: 21, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42098, Gap length: 895509, Signal RSSI: -106, Current RSSI: -116, pulses: 19, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42099, Gap length: 1015684, Signal RSSI: -106, Current RSSI: -114, pulses: 21, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42098, Gap length: 39115593, Signal RSSI: -106, Current RSSI: -112, pulses: 17, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42098, Gap length: 3949509, Signal RSSI: -106, Current RSSI: -119, pulses: 21, noise
 count: 0
rtl_433_ESP(6): Ignored Signal length: 2, Time since last bit length: 42098, Gap length: 23853593, Signal RSSI: -106, Current RSSI: -120, pulses: 19, noise count: 0
rtl_433_ESP(7): Average RSSI Signal -116 dbm, adjusted RSSI Threshold -107, samples 50000

NorthernMan54, 2023-06-05:
Pls update you code and try again with the latest release.  I made some changes in regard to support for 915mhz

pdkwork, 2023-06-05:
I tried the new code.  Still not working.  I get the same output as before.  I will try another chip to see if the chip is bad.

pdkwork, 2023-06-06:
I did try as you suggested.  Below is what I have set in complier definition.  I had to set '-DOOK_MODULATION=false to get any response.  Without it, there was no responses.  With it set, there are messages but they are not decoded.

[env:esp32_heltec_915]
board = heltec_wifi_lora_32_V2
build_flags = 
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
  '-DONBOARD_LED=LED_BUILTIN'          ; Onboard LED is GPIO 25 on the Heltec Board
; *** rtl_433_ESP Options ***
  '-DRF_MODULE_FREQUENCY=915.00' 
  
  '-DOOK_MODULATION=false'       ; False is FSK, True is OOK
;  '-DRF_SX1276="SX1276"'
;  '-DRTL_DEBUG=4'           ; rtl_433 verbose mode
;  '-DRTL_VERBOSE=58'          ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
;  '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages
;  '-DMEMORY_DEBUG=true'   ; display memory usage information
;  '-DDEMOD_DEBUG=true'  ; display signal debug info
;	'-DMY_DEVICES=true'		; subset of devices
  '-DPUBLISH_UNPARSED=true'   ; publish unparsed signal details
;  '-DRSSI_THRESHOLD=12'         ; Apply a delta of 12
;  '-DOOK_FIXED_THRESHOLD=0x50'  ; Inital OOK Threhold - Only for SX127X
;  '-DAVERAGE_RSSI=5000'     ; Display RSSI floor ( Average of 5000 samples )
;  '-DSIGNAL_RSSI=true'             ; Display during signal receive
; *** RF Module Options ***
;  '-DRF_SX1278="SX1278"'   ; Heltec ESP 32 Module - module settings come from heltec_wifi_lora_32_V2/pins_arduino.h
  '-DRF_MODULE_DIO0=26'    ; SX1276 pin DIO0
  '-DRF_MODULE_DIO1=35'    ; SX1276 pin DIO1
  '-DRF_MODULE_DIO2=34'    ; SX1276 pin DIO2
  '-DRF_MODULE_RST=14'     ; pin to be used as hardware reset
  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
; *** Heltec module requires non-standard SPI Config ***
  '-DRF_MODULE_CS=18'      ; pin to be used as chip select
  '-DRF_MODULE_MOSI=27'
  '-DRF_MODULE_MISO=19'
  '-DRF_MODULE_SCK=5'
; *** RadioLib Options ***
;  '-DRADIOLIB_DEBUG=true'
  '-DRADIOLIB_VERBOSE=true'

NorthernMan54, 2023-06-06:
; '-DDEMOD_DEBUG=true' ; display signal debug info

This should enable raw signal debug

pdkwork, 2023-06-06:
I did enable this and all signals that are not fsk are Ignored.  With fsk enabled there are Ignored messages and some received but unparsed messages.  Some of the output is below.
----------- FSK --------------
FDEV_MSB: 0x02
FDEV_LSB: 0x8f
----- SX127x Status -----
rtl_433_ESP(7): Average RSSI Signal -115 dbm, adjusted RSSI Threshold -106, samples 50000
rtl_433_ESP(6): RAW (5001): +144-786+66-424+77-51+77-117+66-1595+65-52+69-1207+62-1082+63-417+190-55+58-121+65-308+55-186+55-59+186-62+238-241+59-179+66-186+62-55+121-307+55-121+66-186+62-304+186-180+182-122+120-69+62-58
rtl_433_ESP(6): Unparsed Signal length: 5001, Signal RSSI: -97, pulses: 27
N: Received message : {"model":"undecoded signal","protocol":"signal parsing failed","duration":5001,"rssi":-97,"pulses":27}
rtl_433_ESP(6): RAW (5001): rtl_433_ESP(6): RAW (5001): +42-656+368-55+124-245+55-58+66-55+242-127+180-117+128-245+62-124+121-62+245-125+127-64+61-62+66-62+120-63+62-55+66-124+179-73+183-124+179-66+183-183+55-428+62-117+190-362+121-62+62-114+117-65+180-66+179-58+59-66+117-66+116-187+121-175+128-114+179-69+245-117+62-121+117-66+124-176+69-121+55-117+67-178+55-113
rtl_433_ESP(6): Unparsed Signal length: 5001, Signal RSSI: -103, pulses: 43
N: Received message : {"model":"undecoded signal","protocol":"signal parsing failed","duration":5001,"rssi":-103,"pulses":43}
rtl_433_ESP(6): RAW (5001): rtl_433_ESP(7): Average RSSI Signal -115 dbm, adjusted RSSI Threshold -106, samples 50000
rtl_433_ESP(7): Average RSSI Signal -115 dbm, adjusted RSSI Threshold -106, samples 50000
rtl_433_ESP(6): RAW (7000): +5225-66+1854-59+54-59+113-59+117-65+56-120+58-300+176-110+58-59+234-55+120-66+117-51+59-289+58-55+59-230+55-113+55-55+55-399+171-59+55-55+113-285+55-58+67-113+58-59+113-58
rtl_433_ESP(6): Unparsed Signal length: 7000, Signal RSSI: -92, pulses: 25
N: Received message : {"model":"undecoded signal","protocol":"signal parsing failed","duration":7000,"rssi":-92,"pulses":25}
rtl_433_ESP(6): RAW (7000): rtl_433_ESP(7): Average RSSI Signal -115 dbm, adjusted RSSI Threshold -106, samples 50000

antamy, 2023-06-21:
@pdkwork did you make any progress on this?   I'm about to go down this path.  I'm not entirely clear what I should be compiling though.  Did you stay with the renamed ook_receiver (main.cpp) and upload that to the board?

NorthernMan54, 2023-06-26:
Sorry, was out of the country for a few weeks and now have time to look at this further.

Let me take one of my 915 Mhz boards and try it in OOK mode to see what I can receive.

pdkwork, 2023-06-26:
> Your logs are similar to what I am seeing (see issue #87<https://github.com/NorthernMan54/rtl_433_ESP/issues/87>). I also have RSSI around -100dbm, lots of single digit length ignored messages, then the occasional 4 digit length message. The latter are always round 1000s (2000, 3000, 4000) etc. I've yet to pick up my FSK / PCM encoded stream from my temperature sensor. What antenna are you using?

I am using the standard antenna that came with the Heltek chip.

pdkwork, 2023-06-26:
I am using the antenna that came with the chip.  I have also tried using a freq that my SDR reports instead of 915.00

'-DRF_MODULE_FREQUENCY=912.600155

 

The reported pulse count does seem to go up.

 

I have 3 transmitters that I am trying to receive: Neptune R900 water meter using R900 protocol, Gas Meter using ERT SCM protocol, and Electric meter using SCM+ protocol.  I don’t believe that SCM+ is enabled in this software but the others appear to be listed.  I believe that the SDR samples at a higher frequency and I don’t know if the sample frequency is adjusted for the higher transmitter freq of 915 vs 433.

antamy, 2023-06-26:
I should play around with the frequency.  I used RTL_433 with my SDR and pick up the two devices I am interested it at 915MHz.  The sample rate is a lot higher than with the LoRa radio:
```

[SDR] Using device 0: Generic, RTL2832U, SN: 77771111153705700, "Generic RTL2832U"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-06-21 20:37:35
model     : SCMplus      id        : 100480017
Protocol_ID: 0x1E        Endpoint_Type: 0xBC       Endpoint_ID: 100480017    Consumption: 818640       Tamper    : 0x0808        crc       : 0x0A88        Meter_Type: Gas           Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-06-21 20:38:06
model     : Fineoffset-WH32                        ID        : 18
Battery   : 1            Temperature: 23.5 C       Humidity  : 52 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

NorthernMan54, 2023-06-27:
@antamy For testing of FSK Reception I used a Fineoffset WH51 @ 915 Mhz and had enabled FSK mode.  When testing it was about 5 meters away.

antamy, 2023-06-27:
> @antamy For testing of FSK Reception I used a Fineoffset WH51 @ 915 Mhz and had enabled FSK mode. When testing it was about 5 meters away.

I'll double check my code changes.  I had to apply a change to` fineoffset.c` from the rtl_433 issues to get it to work with rtl_433.  Maybe my device is bad, there's just no way to tell.

NorthernMan54, 2023-06-27:
When I was doing testing I used a rtl_sdr and rtl_433 to confirm the device signal etc

antamy, 2023-06-27:
> When I was doing testing I used a rtl_sdr and rtl_433 to confirm the device signal etc

Switching back to my issue to continue this

Entropy512, 2024-06-25:
> I believe that the SDR samples at a higher frequency and I don’t know if the sample frequency is adjusted for the higher transmitter freq of 915 vs 433.

Yeah.  The SDR samples fast enough to receive multiple channels (2.36 MHz bandwidth, channel spacing is 125 kHz, so it can monitor 18 channels simultaneously.  This project instead uses a dedicated RF hardware chain in the receiver chipset, with a fixed bandwidth intended for reception of only one channel at a time.

For something like the CC1101 or SX127X, it can only monitor one (maaaaaybe two with severely reduced performance) channel at a time.  So you'll only see one message every 700 seconds (as that's the length of a full hop sequence), even if the meter is broadcasting every 14 seconds or so.  I don't know the exact broadcast interval, as it's "at least" 50 hops in a sequence but I don't know the exact number.

Entropy512, 2024-06-25:
I think I may have figured out why I'm not seeing the preamble:

In the interrupt handler, it ignores pulses as noise if "receiveMode" is not true.

receiveMode is set to true when the RSSI crosses an appropriate threshold.  The problem with this is that the RSSI is polled, and for messages with short duration (such as a 32.768 kbit/second message), this means that a significant part of the preamble and possibly the syncword are ignored.

Options I can think of:

- If the SX127x can set an RSSI interrupt GPIO and there's a free GPIO, use that to immediately start receiveMode
- Instead of immediately dismissing a receive interrupt outside of receiveMode as noise, put it in a "pending" bucket.  If the next receiver task cycle sees an appropriate RSSI - empty the bucket but don't add it as noise.  If it doesn't see an appropriate RSSI, empty the bucket into the noise counter and also reset the active pulse buffer

Also, I think the following will need to move from being hardcoded to overridable defines:

- Bitrate setting
- RX bandwidth setting
- Minimum pulse length
- Minimum signal length
- Whether or not to use the bit synchronizer (the pulses are really ragged without it for a 32.768 kbit/sec signal/30 microsecond pulse length

closed 2024-11-30 by NorthernMan54