NorthernMan54/rtl_433_ESP #85: CC1101 stops working- hardware-defect?

Dattel, 2023-06-14:
Hi... i'm using the rtl_433_esp lib through OpenMqttGateway and a esp32-devboard but suddenly my cc1101 stops receiving anything. I reflashed the ESP, changed back and forth between versions without any change.

Before i'm buying a replacement cc1101 i would like to ask for help :-/
Maybe i'm not unterstanding these modules right, but as far as i know, they stores it's own register settings on it's own eeprom, right?
Is there a way, to reset all registers on the eeprom to it's default using these lib (if not, maybe you have any suggestions, how)? I remember from earlier projects (signalesp) that there was some cc_reset procedure for these job. 

Many thanks in advance.

NorthernMan54, 2023-06-14:
The transceiver module does not store settings between power cycles, but relies on the code flashed to the esp32 to set the desired settings.

Dattel, 2023-06-14:
got a dumb guess..... 
Additional to my ESP-CC1101 combo i have some older ESP-FS1000-MX-RM combination.
Seems like the older once crashed due to a week power supply some days ago and falls into a state, where it constant mumbles 433 signals all the time, which seems to floods the cc1101 and avoids and data..
Does that sound possible?

It was more like a coincidence, that i recognize the online-absence of the esp and repowered it... in the meantime my CC1101 seems to wake up and starts receiving signals.

closed 2023-06-14 by Dattel