NorthernMan54/rtl_433_ESP #86: How to...

habub, 2023-06-19:
Hi,
Looks like the OOK example is working to some extent (ESP32+CC1101). I have some basic usage questions. Much appreciated.

1) Which coding formats are supported? Not clear if it's just PWM and PPM, or others also, specifically PMC and MS (for OOK).
2) What's the difference between the devices in the _rtl_433/devices_ folder and the _contrib_ folder?
3) What's the process for adding/removing devices? Do I need to run _update_rtl_433_devices.sh_? Or is it executed automatically in the build process? Do I add/remove the device,c file and update _rtl_433_devices.h_ manually (in the contrib or devices folders)?
4) When transimtting my specific device I get this message in the Terminal. What does it mean?

rtl_433_ESP(6): Ignored Signal length: 327000, Time since last bit length: 40058, Gap length: 1216404, Signal RSSI: -35, Current RSSI: -89, pulses: 0, noise count: 0
rtl_433_ESP(6): Ignored Signal length: 326000, Time since last bit length: 40058, Gap length: 1217404, Signal RSSI: -33, Current RSSI: -89, pulses: 0, noise count: 0
rtl_433_ESP(7): Average RSSI Signal -86 dbm, adjusted RSSI Threshold -77, samples 50000

Thanks.

NorthernMan54, 2023-06-20:
1 - All ook formats should be supported, including pmc and mc.  As I only only have access to a few devices, I have no method of determining that they work, so ymmv.

2 - Contrib is overlaid to devices when I do an update from rtl_433.  So that any specific changes to a decoder are not lost.

3 - You should not need to add or remove devices, as included all that did not cause an exception or issue.

4 - That message is the receiver logic indicating that a signal carrier was detected ( signal rssi ), that the carrier was heard for how long ( signal length ), and how many bits/pulses received.

it appears that it is detecting a signal, but not seeing any bits/pulses.  Could be a transmitter issue, or a wiring issue.

habub, 2023-06-20:
Thanks.
But I do want to test a new device locally. How can I add it to the devices list?
Can I print the actual bits or raw signal the receiver is receiving? to understand where the problem is?

NorthernMan54, 2023-06-23:
With the hardware/software wiring, look at the gdo pins.  The code watches a gdo pin for pulses when a signal is detected, and the log indicates that no pulses are being detected.

habub, 2023-07-02:
Thanks.
Looks like it's working now.
I have a questions regarding cc1101 register settings. Where can I find them in the code? Also, I assume you are using Asynchronous serial mode?

NorthernMan54, 2023-07-02:
Register settings are in rtl_433_ESP.cpp

And you are correct, I’m using asynchronous serial mode to receive the raw signal for decoding in software

habub, 2023-07-12:
Hi
I have a few clarifications of the asynchronous serial mode if you don't mind:

In this mode, which pins are used for Data (Rx) and Clock?
Who supplies the clock signal? Does the ESP still work as an SPI master and supply the clock on CLK pin?
Does the receiver use the IRQ and CS pins? Or is data just spilled out from the CC1101?
Is it at all an SPI protocol in asynch mode?

Thanks in advance.

NorthernMan54, 2023-07-13:
Please double check the datasheet, Asynchronous serial mode only uses a single GDO pin for signal data, and no clock pin.

habub, 2023-07-17:
Thanks.
Do you know if I can define 2 RF modules (for example cc1101) in parallel for a single rtl_433_esp instance? I think it's possible in RadioLib but I'm not sure regarding your code...
_#    define RADIO_LIB_MODULE \
      new Module(RF_MODULE_CS, RF_MODULE_GDO0, RADIOLIB_NC, RF_MODULE_GDO2, newSPI)_

NorthernMan54, 2023-07-17:
The code was designed around a single receiver module instance with interrupt driven pulse receiver loop, so using dual modules is not supported.

Why dual when a second ESP32 is dirt cheap or even easier a LilyGo board with no soldering required

NorthernMan54, 2023-07-17:
You are correct, every bit-pulse generates an ESP32 interrupt, that records the time since the previous bit-pulse.

Start and end of signal is detected by monitoring the RSSI value.

habub, 2023-07-19:
Thanks, I got it now....sorry I'm a newbie...

Another question, I want to edit the output message format and also send to MQTT. I couldn't locate the code file that creates/prints the output messages:
_N: Received message : {"model":"Acurite-Tower","id":2043,"channel...._
(In the original RTL_433 code it's in output_file.c)

habub, 2023-07-19:
And another one regarding OpenMqttGateway:
1) Where do I define the RTL_433_ESP pinouts and parameters for OMG? Does it inherit the parameters I set in the original repo (ini+ino files)? Or is there a config_xx.h file to update?
2) I'm currently using a standard Esspressif ESP32 dev board and ths set the OMG environment to _default_envs = esp32dev-rtl_433_. If I want to switch to an Adafruit Huzzah Feather board, can I keep the same environment? Or do I need a different one?

NorthernMan54, 2023-07-20:
If you have customized the rtl_433_ESP library, then you need to tell platformio to use your local copy, besides that it should work.  Their may be some minor DIRECTIVE differences, but majority just carry thru

```
[libraries]
rtl_433_esp = symlink://../rtl_433_ESP  ; Builds library from source directory
```

1 - How to specify pins
```
; *** RF Module Options ***
  '-DRF_CC1101="CC1101"'  ; CC1101 Transceiver Module
  '-DRF_MODULE_CS=5'      ; pin to be used as chip select
  '-DRF_MODULE_GDO0=12'   ; CC1101 pin GDO0
  '-DRF_MODULE_GDO2=27'   ; CC1101 pin GDO2
```

2 - I think #1 is causing your issue #2

closed 2023-12-20 by habub