NorthernMan54/rtl_433_ESP #88: rtl_433_ESP + ESPHome

mag1024, 2023-06-25:
Greetings! Thank you for writing this useful library!
I'm trying to get it to work as a custom component in ESPHome (for reasons I won't go into here) using a lilygo v2.1 (aka T3 1.6.1) board.

I go it to the point where it's receiving signals... but they appear to be garbage.

> rtl_433_ESP(6): RAW (846017): +0-991+186-1814+186-1814+186-1814+186-1814+186-1814+...
> pulse_slicer_pwm: Govee Water Leak Detector H5054 codes [{1}8, {1}8, {1}8, {1}8, {1}8, {1}8, {1}8, {1}8, ....
> rtl_433_ESP(6): Unparsed Signal length: 846017, Signal RSSI: -53, pulses: 444
> rtl_433_ESP(6): RAW (846017): [D][custom:085]: Received msg: {"model":"undecoded signal","protocol":"signal parsing failed","duration":846017,"rssi":-53,"pulses":444}

Do you have any hints about what might cause this particular failure?

Comparing the output of OOK_Receiver and my code running side-by-side, I noticed that the esphome version reports RegOpMode: 0x29 (which I think might be standby?) whereas OOK_Receiver reports RegOpMode: 0x2d. The chip init seems to succeed, and it starts off in the right mode

> [I][app:029]: Running through setup()...
> [D][custom:075]: Receiver pin: 32, freq: 433.920000
> [D][custom:077]: DIO0: 26, DIO1: 33
> rtl_433_ESP(6): SX1278 radio.begin() - success!
> rtl_433_ESP(6): SX1278 setOOK - success!
> rtl_433_ESP(6): SX1278 setCrcFiltering - success!
> rtl_433_ESP(6): SX1278 setDataShapingOOK - success!
> rtl_433_ESP(6): SX1278 OOK Thresh PEAK - success!
> rtl_433_ESP(6): SX1278 OOK PEAK Thresh Decrement - success!
> rtl_433_ESP(6): SX1278 Ook Peak Threshold Step - success!
> rtl_433_ESP(6): SX1278 OokFixedThreshold - success!
> rtl_433_ESP(6): SX1278 setBitRate - success!
> rtl_433_ESP(6): SX1278 setRxBandwidth - success!
> rtl_433_ESP(6): SX1278 RSSI Smoothing - success!
> rtl_433_ESP(6): SX1278 preamble detect off - success!
> rtl_433_ESP(6): SX1278 setDirectSyncWord - success!
> rtl_433_ESP(6): SX1278 disableBitSync - success!
> rtl_433_ESP(6): SX1278 receiveDirect - success!
> ----- SX127x Status -----
> RegOpMode: 0x2d

But almost immediately resets to 0x29 -- but I'm at a loss for what could be causing this.

I do realize that this is way outside your main supported use case, but I definitely appreciate any pointers for how to approach debugging this.

NorthernMan54, 2023-06-26:
Double check the other register settings, it sounds like the receiver is not in continuous receive mode, and is in packet mode.  And is going to sleep after receiving a packet.

mag1024, 2023-06-27:
Thanks for the response!
After some absurd amount of fiddling, I finally have it working.
The trigger was the difference in RadioLib versions -- RadioLib 6.x.x from git does not work, but the default version that platformio pulls (5.7.0) does. Using that fixed my esphome component. As for the cause 🤷‍♂️

closed 2023-06-27 by mag1024