NorthernMan54/rtl_433_ESP #91: OOK_Receiver example does not with recent RadioLib versions (6.0.0+)

mag1024, 2023-07-01:
Following up on #88, I went back and tested unmodified OOK_Receiver with RadioLib 6.1.0 (and also 6.0.0) -- neither seems to work with my lilygo v2.1. Similar symptoms I was originally seeing with esphome: opmode is not the steady 0x2d, but instead 0x29, 0x2c, 0x01 or 0x00.

Based on my investigation \w esphome, I don't think the issue is the lack of contiguous RX -- the chip notices/triggers when I run the transmitter, it just seems to receive garbage.

Have you tried recent radiolib versions? Does it work for you?

```
N: ****** setup ******
rtl_433_ESP(6): SX1278 radio.begin() - success!
rtl_433_ESP(6): SX1278 setOOK - success!
rtl_433_ESP(6): SX1278 setCrcFiltering - success!
rtl_433_ESP(6): SX1278 setDataShapingOOK - success!
rtl_433_ESP(6): SX1278 OOK Thresh PEAK - success!
rtl_433_ESP(6): SX1278 OOK PEAK Thresh Decrement - success!
rtl_433_ESP(6): SX1278 Ook Peak Threshold Step - success!
rtl_433_ESP(6): SX1278 OokFixedThreshold - success!
rtl_433_ESP(6): SX1278 setBitRate - success!
rtl_433_ESP(6): SX1278 setRxBandwidth - success!
rtl_433_ESP(6): SX1278 RSSI Smoothing - success!
rtl_433_ESP(6): SX1278 preamble detect off - success!
rtl_433_ESP(6): SX1278 setDirectSyncWord - success!
rtl_433_ESP(6): SX1278 disableBitSync - success!
rtl_433_ESP(6): SX1278 receiveDirect - success!
----- SX127x Status -----
RegOpMode: 0x2c
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x2a
RegRxBw: 0x01
RegAfcBw: 0x02
-------------------------
RegOokPeak: 0x08
RegOokFix: 0x0f
RegOokAvg: 0x12
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----- SX127x Status -----
N: ****** setup complete ******

rtl_433_ESP(6): Status Message: Gap length: 1050706, Modulation: OOK, Signal RSSI: 0, train: 0, messageCount: 0, totalSignals: 0, signalRatio: 0, ignoredSignals: 0, unparsedSignals: 0, _enabledReceiver: 1, receiveMode: 0, currentRssi: -95, rssiThreshold: -82, StackHWM: 6220, RTL_HWM: 908, DCD_HWM: 9480, pulses: 0
----- SX127x Status -----
RegOpMode: 0x29
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x00
RegRxBw: 0x01
RegAfcBw: 0x00
-------------------------
RegOokPeak: 0x08
RegOokFix: 0x0f
RegOokAvg: 0x12
-------------------------
RegLna: 0x20
RegRxConfig: 0x00
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----- SX127x Status -----
N: Received message : {"model":"status","protocol":"rtl_433_ESP status message","modulation":"OOK","RTLRssi":-95,"RTLAVGRssi":0,"RTLRssiThresh":-82,"signalRssi":0,"train":0,"RTLCnt":0,"totalSignals":0,"signalRatio":0,"ignoredSignals":0,"unparsedSignals":0,"StackHWM":6220,"RTL_HWM":908,"DCD_HWM":9480,"freeMem":255392,"_enabledReceiver":1,"receiveMode":0}
----- SX127x Status -----
RegOpMode: 0x00
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x2a
RegRxBw: 0x01
RegAfcBw: 0x02
```

NorthernMan54, 2023-07-01:
I have not looked at the updated version of RadioLib yet, maybe over the winter.

Nealtron, 2023-07-01:
@mag1024, would you mind sharing your esphome config?

mag1024, 2023-07-02:
@Nealtron Sure, uploaded to [mag1024/esphome-rtl433](https://github.com/mag1024/esphome-rtl433)

PR #92 (2023-07-04): Fix disabling of messing with the receive mode in getRSSI call for RadioLib >5.7.0
[Culprit commit](https://github.com/jgromes/RadioLib/commit/09ab5f8073f7d9b2ca20492b2aa78fa9c3aa35a1#diff-2f43a1ab977b88f0d25a8ea1816d16a29440707f54f80d644a9c2e71c4c66c88R260)

fixes #91
closed 2023-07-04 by NorthernMan54