NorthernMan54/rtl_433_ESP #93: PackageException: Can not create a symbolic link for `../../../rtl_433_ESP`, not a directory

maho24at, 2023-07-07:
I cant open or bulid the project, I always get the error 

PackageException: Can not create a symbolic link for `../../../rtl_433_ESP`, not a directory

[image]

PIO 6.1.9 / 3.4.4

and if I try to inspect the project 

PIO Core Call Error: "Processing esp32_lilygo (board: ttgo-lora32-v21; framework: arduino; platform: espressif32@6.1.0)\r\n--------------------------------------------------------------------------------\r\nLibrary Manager: Installing symlink://../../../rtl_433_ESP\r\n\n\nPackageException: Can not create a symbolic link for `../../../rtl_433_ESP`, not a directory"

maho24at, 2023-07-08:
win10 - I tried to google the error but I couldnt find anything 

but before you go nuts on it, I was never using PIO before, usually I use the Arduino IDE maybe I did something wrong.
I did a new install of the PIO plugin in VSC 
I tried to uninstall the plugin and reinstall it

I extra bought the LILYGO  T3 1.6.1 which you used just to be fine - and now I run into an issue with PIO - do you have discord? maybe we come around pio anyways, I'm looking for the code to receive data from my weather station WS69 on 868mhz and the libraries / my weather station was shown as "fine offset WH65B" in rtl_433

NorthernMan54, 2023-07-08:
I use a Mac and symlink is a Mac/Unix thing.  It probably needs slightly different syntax with Win10.

If you change it to `rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.3.0` it should work.  This will download the library for the build, rather than your local copy.

maho24at, 2023-07-10:
ok that solved the first problem but reveiled another problems,.. 

to be sure I'm doin right here and not wasting your time by being clueless...

first I open the OOK_Receiver ino - then I go to Project tasks and choose in my case the esp32_lilygo - then click built - right? 

that results in many errors... basically all related to the missing library SPI.h

In file included from .pio/libdeps/esp32_lilygo/RadioLib/src/Module.cpp:1:
.pio/libdeps/esp32_lilygo/RadioLib/src/Module.h:7:12: fatal error: SPI.h: No such file or directory

I'm not sure how to add the library  to the project,.. I guess in  the libraries configuration tab,...

NorthernMan54, 2023-07-10:
The challenges are more in align with the differences between running Visual Studio / Platformio on Mac / Linux versus Win10.  I find that on Mac this issue does not occur, and it just finds the SPI library.

maho24at, 2023-07-11:
do you see a possibility to make libraries and code for the arduino IDE? I just have issues with platformio, not even the blink sketch will compile correctly,.. maybe my VSC is already messed up because I worked a lot with it with many extensions and plugins,

NorthernMan54, 2023-07-11:
The Arduino IDE is not something I'm looking at working with.  I used it years ago, then switched to VSC/Platformio when I realized how much easier it was to work with.

Why not just re-install VSC/Platformio

seb821, 2023-08-01:
I had similar issues. I moved the project outside of "My Documents" and made sure there is no space or special character in the path. Then I was able to build properly.

NorthernMan54, 2023-08-09:
@seb821 Tks for the tip for Windows