NorthernMan54/rtl_433_ESP #95: instructions on how to compile and flash

sgoudelis, 2023-08-02:
Good evening, 

I apologize for asking this but I have been trying to understand how to compile this and flash it in a TTGO lora32 (with OLED) and I can't seem to make this work. Does the code on this repo enough for that? Should I use the openmqttgateway repo ? If so, does it require a compile to enable options for that board and an FSK based weather station transmitting on 868mgz? I tried one of the pre-built images (or openmqttgateway), the board booted and I configured it, but how to enable the rtl_433 logic for 868mhz with FSK support? I can't seem to find any documentation that.

I appreciate any hint, thank you

1technophile, 2023-08-02:
Good evening,
You may find help by reading this:
https://community.home-assistant.io/t/heltec-bresser-weatherstation/444967/8?u=1technophile

sgoudelis, 2023-08-02:
This works perfectly! Thank you