NorthernMan54/rtl_433_ESP #96: Terminal decoding not working

bgmcode, 2023-08-06:
I'm able to compile and upload to a LilyGo TTGO LoRa32 V2.1_1.6, and the onboard LED flickers when a 433 signal is detected. However, I am not able to make sense of the serial output in the VS Code Serial Monitor.  I'm probably missing something obvious...  Below is a snippet of the serial window.  The question mark symbols appear periodically and correspond to the LED.  Thanks!

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4

�}?����u����H��H�%%%��8�.����������u�O�������������UUO������������UuϜ�������������}gd��������8�u��c��'�����������������������c���������������9�������������O�������������?,����������d�����c�������������s������d������݄���������
                                                                                                  ��՟�����-�������Us�����k�����������c�Y��������������   �����������g�����������Χ���������c���������Us�����Z��������������=���������O�����=�������՟��V�������������������� �
                                 ���c��������f��pu��c����������U��O����������
                                                                             |�}gd������������Us���i��������uկ�����������
                                                                                                                         ��݄c#�����c���������������[���������Ws��������������}g���������������O������������Յשּׂ��,������
                                                                                      Ju]שּׂ����������յ������֬�������c�l�������k�݄c���������

NorthernMan54, 2023-08-06:
You have an incorrect monitor speed setting

https://github.com/NorthernMan54/rtl_433_ESP/blob/0b9b378bd390c0dc53751c34816b1f5d5cbb14c1/example/OOK_Receiver/platformio.ini#L65

bgmcode, 2023-08-06:
Thanks. I just tried reverted to the master branch and only updated my port info.  Possible hardware issue?

---- Opened the serial port COM4 ----
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
���u�����u�)!���%%%�ƌ�������������

NorthernMan54, 2023-08-07:
No, you have an incorrect monitor speed setting

bgmcode, 2023-08-22:
I had to add 921600 in the VS Code serial monitor as an option for baud rate. That fixed it. Thanks!

closed 2023-08-22 by bgmcode