NorthernMan54/rtl_433_ESP #97: support for Misol WH2900 wireless weather station

abhaysbharadwaj, 2023-08-11:
Hello,
I had brought this WH2900 on amazon for my small farm. It has an outdoor weather station unit and an indoor display unit. I wanted to monitor this from my home and came across this repo and was figuring out a way to use ESP32 to read and send the data to a cloud server.

I am not able to get any data decoded. 
Tried with a Heltec LoRa V2 (having a RF98 chip). Tried with a DOIT ESP32 + RA02 and a DOIT ESP32 + Seeed RFM96W.

Any help in this regard will be highly appreciated.

abhaysbharadwaj, 2023-08-11:
The output I get on serial is something like this:

`Received message : {"model":"undecoded signal","protocol":"signal parsing failed","duration":5001,"rssi":-97,"pulses":27}`

NorthernMan54, 2023-08-11:
This codebase only supports the cc1101 and SX127x transceiver chipsets.

Also I googled "Misol WH2900 rtl_433" and I don't see the device being mentioned so I don't think a signal decoder is available.

closed 2023-09-28 by NorthernMan54