NorthernMan54/rtl_433_ESP #99: esp32S3 crashing

tablatronix, 2023-08-16:
```
Guru Meditation Error: Core  0 panic’ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (rtl_433_Receive) 
(Stacktrace)
```
stack size?

Anyone know what this is before I look more into it.

tablatronix, 2023-08-16:
```
Rebooting…
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403775a4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[  1534][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
rtl_433_ESP(6): CC1101 gpio receive pin: 7
rtl_433_ESP(6): CC1101 receive frequency: 433.950012
rtl_433_ESP(6): # of device(s) configured 157
rtl_433_ESP(6): ssizeof(r_device): 112
rtl_433_ESP(6): cfg->devices size: 17584
rtl_433_ESP(7): Process rtl_433_DecoderTask resource hit Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor, deltaStack: 0, stack: 6768, deltaHeap: 208, heap: 176272
rtl_433_ESP(7): Process rtl_433_DecoderTask resource hit HT680 Remote control, deltaStack: 0, stack: 6768, deltaHeap: 248, heap: 173592
rtl_433_ESP(7): Process rtl_433_DecoderTask resource hit Proove / Nexa / KlikAanKlikUit Wireless Switch, deltaStack: 0, stack: 6768, deltaHeap: 308, heap: 169572
rtl_433_ESP(7): Process rtl_433_DecoderTask resource hit Yale HSA (Home Security Alarm), YES-Alarmkit, deltaStack: 0, stack: 6768, deltaHeap: 396, heap: 163544
rtl_433_ESP(6): rtl_433_Decoder_Stack 10000
rtl_433_ESP(6): CC1101 SPI Config SCK: 13, MISO: 12, MOSI: 11, CS: 10
rtl_433_ESP(6): rtl_433_ReceiverTask_Stack 1800
—— CC1101 Status ——
CC1101_MDMCFG1: 0x02
CC1101_MDMCFG2: 0x30
CC1101_MDMCFG3: 0x93
CC1101_MDMCFG4: 0x07
...
CC1101_PARTNUM: 0x00
CC1101_VERSION: 0x14
CC1101_MARCSTATE: 0x0d
CC1101_PKTSTATUS: 0xb0
CC1101_RXBYTES: 0x00
—— CC1101 Status ——
Guru Meditation Error: Core  0 panic’ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (rtl_433_Receive) 
Core  0 register dump:
PC      : 0x40380333  PS      : 0x00060836  A0      : 0x8037f78b  A1      : 0x3fcc6570  
A2      : 0x3fc97150  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060823  
A6      : 0x00060823  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x3fcf08d8  A12     : 0x00000080  A13     : 0x3fc959a8  
A14     : 0x02c97150  A15     : 0x00ffffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40380330:0x3fcc6570 0x4037f788:0x3fcc65b0 0x4037d666:0x3fcc65d0 0x4037db0a:0x3fcc65f0 0x4037dc6d:0x3fcc6630 0x403790f7:0x3fcc6650 0x4037919d:0x3fcc6680 0x4206c609:0x3fcc66a0 0x4206b639:0x3fcc66c0 0x4206ac75:0x3fcc66e0 0x420e74b5:0x3fcc6700 0x420e506a:0x3fcc6720 0x420e50f2:0x3fcc6740 0x420e5e62:0x3fcc6770 0x420f5ff5:0x3fcc67a0 0x420ef1be:0x3fcc67c0 0x420e6b2d:0x3fcc6ad0 0x4202866d:0x3fcc6b20

```

NorthernMan54, 2023-08-16:
Try increasing this a little bit, maybe to 2000

https://github.com/NorthernMan54/rtl_433_ESP/blob/0b9b378bd390c0dc53751c34816b1f5d5cbb14c1/src/rtl_433_ESP.cpp#L58

NorthernMan54, 2023-08-16:
Also of interest is this output from the status message

alogprintf(LOG_INFO, ", RTL_HWM: %d", uxTaskGetStackHighWaterMark(rtl_433_ReceiverHandle));

After increasing the value, this will show how much is available.  It is a high water mark, and is best collected after running for a while.

tablatronix, 2023-08-16:
Tried that no luck also tried bumping arduino loop size.

NorthernMan54, 2023-08-16:
Increase it larger, then you can tune it back by watching RTL_HWM

And your build included the tweak?  ie this changed 

rtl_433_ESP(6): rtl_433_ReceiverTask_Stack 1800

tablatronix, 2023-08-17:
That works! I was editing `rtl_433_Decoder_Stack` my bad.
its at 2000 now

`rtl_433_ESP(6): rtl_433_ReceiverTask uxTaskGetStackHighWaterMark: 80`

tablatronix, 2023-08-17:
I get a crash now and then but thats another issue,  its working

```
Guru Meditation Error: Core  0 panic’ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (wifi) 
Core  0 register dump:
PC      : 0x4037ebbe  PS      : 0x00060736  A0      : 0x4037d08b  A1      : 0x3fcc6210  
A2      : 0x3fca1af4  A3      : 0x3fca1b1c  A4      : 0x3fca1b1c  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000017  A8      : 0x3fca1afc  A9      : 0x3fcc61f0  
A10     : 0x3fcf7d1c  A11     : 0x3fcf7d1c  A12     : 0x00000014  A13     : 0x00000000  
A14     : 0x3fcf7d14  A15     : 0x80000001  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x4037ebbb:0x3fcc6210 0x4037d088:0x3fcc6240 0x4037d07e:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: 2fe5d7535ad5127d

Rebooting…```

closed 2024-11-30 by NorthernMan54