NorthernMan54/rtl_433_ESP #100: HELP? Too much noise on cc1101?

tablatronix, 2023-08-19:
I am having nothing but issues with the C1101, I have 3 modules across esp32 various boards.
I get this all the time, where is this coming from? I sniff GD0 and there is nothing there, I see my accurite 5in1 data stream now and then, but this noise is always being reported ( I assume from miso interface? )

[image]

What could cause this, what am I doing wrong? Is there really this much noise around me on 433?

Is this what everyone deals with, is this normal. I can only receive and decode if I am very close to device.

Any debugging tips without an SDR ?

tablatronix, 2023-08-19:
I had to DDISABLERSSITHRESHOLD to receive, I think it doesn't work at all in my environment, I think maybe there is a single high power pulse near me that is causing issues with rssi tracking maybe, not sure.

tablatronix, 2023-08-21:
Replaced antenna with a tuned one, these "433" ones are terrible, turned off autorssithreshhold

Getting ~100% now at 100ft , I will try to figure out the rssi averaging and noise later when I have time

[image]

NorthernMan54, 2023-08-22:
That is very good results,  can you share a link to the antenna ?

tablatronix, 2023-08-22:
I made a 1/4 whip, need the db to get above noise floor. Still no idea why its so high, could be the strong 433 beacon I hear, and also another weather station slight farther than mine.. Could be all these cc1101 are fake or rejects ( i hear there is also some silicon bugs in these and workarounds) I have not checked rtl_433 to see the specifics or if they are handled just saw some notes on some forums.
[image]

jasiek, 2024-01-16:
Can you share which CC1101 module you were using?

tablatronix, 2024-01-17:
I used both these
[image]

i gave up and moved to a lora32 sx1278 its a much more complex chip

closed 2024-11-30 by NorthernMan54