NorthernMan54/rtl_433_ESP #101: Simple HW question

Crsarmv7l, 2023-09-14:
### Current Situation

I am not seeing any difference between the paxcounter and the disaster radio for the ttgo board aside from the fw.

 Just wanted to ensure that was the case.
[image]

### Logs

```markdown
N/a
```

### Configuration

```json
N/a
```

### Environment

N/a

### Process Supervisor

not applicable

NorthernMan54, 2023-09-14:
The difference is the factory installed 'default firmware'

The SX1278 and SX1276 Module is supported

Crsarmv7l, 2023-09-14:
> The difference is the factory installed 'default firmware'
> 
> The SX1278 and SX1276 Module is supported

Awesome, thanks. That is what I thought. Last question: how deeply integrated is openMQTTGateway?

I would like to use this as basically a standalone rtl_433 without needing a computer (maybe logging to a file like rtl_433 with json if that is supported)

NorthernMan54, 2023-09-14:
I run it within OpenMQTTGateway as it provides a framework for the library to run within, including WiFi and MQTT integration.

Logging to a file on a ESP32 is problematic due to the usage a flash memory on the device.  I send the via MQTT, then listen to the topic to utilize the data.

Crsarmv7l, 2023-09-14:
Cool thanks.

closed 2023-09-14 by Crsarmv7l