NorthernMan54/rtl_433_ESP #102: Support for WH65B station

tguozden, 2023-09-16:
### Current Situation

Hello, thanks very much for this repository. We are working in a students project to acquire data from several weather stations and this is really useful.

We started with a fine offset WH65b, which we can see with a rtl-sdr device and rtl_433 for linux.

Using the example posted here with OOK_modulation=false (FSK) there is no information parsed through to the monitor. Although we can see output with each transmitted message when turning on -DRTL_DEBUG=1

Using openmqttgateway and home assistant didn't work either.

Is device supported?

We use a Heltec 915 Lora device. Unfortunately I don't have the monitor log here when setting DEBUG=1 of 4, will promptly send it if needed.

### Logs

```markdown
Unfortunately I don't have the monitor log here when setting DEBUG=1 of 4, will promptly send it if needed.
```

### Configuration

```json
[env:esp32_heltec_915]
board = heltec_wifi_lora_32_V2
build_flags = 
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
  '-DONBOARD_LED=LED_BUILTIN'          ; Onboard LED is GPIO 25 on the Heltec Board
; *** rtl_433_ESP Options ***
;  '-DRF_MODULE_FREQUENCY=915.00'
  '-DOOK_MODULATION=false'       ; False is FSK, True is OOK
;  '-DRF_SX1276="SX1276"'
  '-DRTL_DEBUG=1'           ; rtl_433 verbose mode
 ;'-DRTL_VERBOSE=58'          ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
;  '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages
;  '-DMEMORY_DEBUG=true'   ; display memory usage information
;  '-DDEMOD_DEBUG=true'  ; display signal debug info
;	'-DMY_DEVICES=true'		; subset of devices
;  '-DPUBLISH_UNPARSED=true'   ; publish unparsed signal details
;  '-DRSSI_THRESHOLD=12'         ; Apply a delta of 12
;  '-DOOK_FIXED_THRESHOLD=0x50'  ; Inital OOK Threhold - Only for SX127X
;  '-DAVERAGE_RSSI=5000'     ; Display RSSI floor ( Average of 5000 samples )
;  '-DSIGNAL_RSSI=true'             ; Display during signal receive
; *** RF Module Options ***
;  '-DRF_SX1278="SX1278"'   ; Heltec ESP 32 Module - module settings come from heltec_wifi_lora_32_V2/pins_arduino.h
;  '-DRF_MODULE_DIO0=26'    ; SX1276 pin DIO0
;  '-DRF_MODULE_DIO1=35'    ; SX1276 pin DIO1
;  '-DRF_MODULE_DIO2=34'    ; SX1276 pin DIO2
;  '-DRF_MODULE_RST=14'     ; pin to be used as hardware reset
;  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
; *** Heltec module requires non-standard SPI Config ***
;  '-DRF_MODULE_CS=18'      ; pin to be used as chip select
;  '-DRF_MODULE_MOSI=27'
;  '-DRF_MODULE_MISO=19'
;  '-DRF_MODULE_SCK=5'
; *** RadioLib Options ***
;  '-DRADIOLIB_DEBUG=true'
;  '-DRADIOLIB_VERBOSE=true'
; *** Setting Testing ***
  ;'-DsetBitrate'
  ;'-DsetFreqDev'
  ;'-DsetRxBW'
monitor_port = /dev/ttyUSB0
monitor_speed = 921600
upload_port = /dev/ttyUSB0
upload_speed = 921600
```

### Environment

- OS: Ubuntu 22.04
- Software: Platform Ino
- Node: Heltec Lora 915
- npm:

### Process Supervisor

hb-service

tguozden, 2023-09-18:
Hello again, I also comment that the recption strength is poor, miss the signal in 5 meter distance.

Here is the monitor's parsed output for each station's transmision: (RTL_debug=1)
```
```
ambientweather_whx_decode: WH31E/WH31B/WH40 detected, buffer is 135 bits length
ambientweather_whx_decode: unknown message type 24 (expected 0x30/0x40/0x68)
fineoffset_WH51_callback: Msg family unknown: 24
fineoffset_wh1080_callback: Fine Offset WH1080 data  codes {80}ff245480024028000000
fineoffset_wh31l_decode: WH31L detected, buffer is 120 bits length
fineoffset_wn34_decode:  codes {72}245480024028000000
honeywell_cm921_decode:  codes {284}b3333333333333333333033cfccc0186066618000000618000330000000000000000000
honeywell_cm921_decode: preamble_start=284 start=314 len=-30
honeywell_cm921_decode:  codes {226}c01f800fcf0fcc303381f3c1ff019861800018033f0f019803fe61f0c
honeywell_cm921_decode: preamble_start=226 start=256 len=-30
inkbird_ith20r_callback: CRC check failed (0xAFF6 != 0x523C)
lacrosse_breezepro_decode: Sync word not found
lacrosse_r1_decode: Packet too long: 274 bits
lacrosse_th_decode: packet length: 274
lacrosse_th_decode: Sync word not found
lacrosse_wr1_decode: Packet too long: 274 bits
``` 

And this is the output with PUBLISH_UNPARSED=true uncommented:
```
rtl_433_ESP(6): Unparsed Signal length: 9999, Signal RSSI: -110, pulses: 61
rtl_433_ESP(6): RAW (9999): +40-59+58-55+59-58+59-55+58-59+54-59+58-59+59-54+59-58+59-58+59-55+58-176+59-54+117-59+172-58+55-59+58-234+59-113+60-174+55-59+58-59+58-117+59-58+59-54+59-58+117-348+58-114+58-465+58-59+58-3880+59-175+114-113+59-58+59-54+118-58+176-172+117-171+59-457+110-171+118-58+172-59+58-172+58-114+58-59+117-230+59-58+235-65+176-58+59-113+59-175+66-62+238-66+55-175+59-124+230-66+468-66+114-117+65-63+117-124+128-55+0-0 
N: Received message : {"model":"undecoded signal","protocol":"signal parsing failed","duration":9999,"rssi":-110,"pulses":61}
rtl_433_ESP(7): Average RSSI Signal -121 dbm, adjusted RSSI Threshold -112, samples 50000
```

tguozden, 2023-09-22:
Hello, also tried to listen to a temperature-humidity sensos Fine Offset WH2 (OOK modulation?) but couldn't either. I tried OOK and FSK  mode and almost every other posible setting combination but there was no response.

I could only get some information parsed with each message when using same settings as with WH65: FSK modulation and debug > 0.

Also check again using openmqtt and home assistant, again nothing happened. (wifi and mqtt connected)

Please any advice would be very welcome.

NorthernMan54, 2023-09-22:
Take a read of this [issue](https://github.com/NorthernMan54/rtl_433_ESP/issues/87), they recommended a change to MINIMUM_SIGNAL_LENGTH that resolved a reception issue.

tguozden, 2023-09-22:
Thanks! Got it working, but with poor signal reception: rssi lower than -80 with the heltec and the antenna alongside the station.

```
--- Terminal on /dev/ttyUSB0 | 921600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
N: Received message : {"model":"Fineoffset-WH65B","id":84,"battery_ok":1,"temperature_C":19.3,"humidity":44,"wind_dir_deg":110,"wind_avg_m_s":0,"wind_max_m_s":0,"rain_mm":0,"uv":0,"uvi":0,"light_lux":19.0,"mic":"CRC","protocol":"Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000, Misol WS2320 Temperature/Humidity/Pressure Sensor","rssi":-81,"duration":11000}
```

Still I cannot decode the WH2 sensor. Any ideas? The sensor is OOK. If it is useful,  the only way I could obtain some parsed information was in FSK mode, debug = 1 and the older value of MINIMUM_SIGNAL_LENGTH

Thanks again

closed 2023-09-22 by tguozden