NorthernMan54/rtl_433_ESP #104: How to access raw data bytes

dirkbeer, 2023-09-21:
### Current Situation

Currently, the rtl_433_Callback() in the examples receives an already-decoded message as JSON. I'd like to be able to receive the sensor data as a string that's the raw hex bytes (e.g. 7 bytes of data for the FineOffset WS80) for efficient retransmission via another transmitter. Currently, I can log a very long hex string that probably includes the data bytes, but can't figure out how to access it or determine which bytes are the right ones.

Any suggestions on the best way to do that? I'd like to try to do it myself, but have failed so far after a couple of days of effort.

### Logs

```markdown
pulse_slicer_pcm: Exact bit width (in us) is 57.92 vs 58.00, 38 bit preamble
N: Received message : {"model":"Fineoffset-WS80","id":11672,"battery_ok":0.98,"battery_mV":2980,"temperature_C":23.7,"humidity":64,"wind_dir_deg":222,"wind_avg_m_s":0.9,"wind_max_m_s":0.9,"uvi":0,"light_lux":60.0,"flags":10,"mic":"CRC","protocol":"Fine Offset Electronics WS80 weather station","rssi":-70,"duration":12000}
pulse_slicer_pcm: Fine Offset Electronics WS80 weather station codes [{219}555555555516ea400016cc00034a853ea004ef04801fffead5001e6]
```

### Configuration

```json
.
```

### Environment

ESP32

### Process Supervisor

not applicable

dirkbeer, 2023-09-21:
Best I've been able to do so far is to put this in each individual sensor's decoder: 
```
    // Convert 'b' to a hex string
    char hexString[18 * 2 + 1];
    for (unsigned i = 0; i < 18; ++i) {
        sprintf(&hexString[i * 2], "%02x", b[i]);
    }
    hexString[18 * 2] = '\0';
    
    /* clang-format off */
    data_t *data = data_make(
            "model",            "",                 DATA_STRING, "Fineoffset-WS80",
            "id",               "ID",               DATA_FORMAT, "%06x", DATA_INT,    id,
            "battery_ok",       "Battery",          DATA_DOUBLE, battery_lvl * 0.01f,
            "battery_mV",       "Battery Voltage",  DATA_FORMAT, "%d mV", DATA_INT,    battery_mv,
            "temperature_C",    "Temperature",      DATA_COND, temp_raw != 0x3ff,   DATA_FORMAT, "%.1f C",   DATA_DOUBLE, temp_c,
            "humidity",         "Humidity",         DATA_COND, humidity != 0xff,    DATA_FORMAT, "%u %%",    DATA_INT, humidity,
            "wind_dir_deg",     "Wind direction",   DATA_COND, wind_dir != 0x1ff,   DATA_INT, wind_dir,
            "wind_avg_m_s",     "Wind speed",       DATA_COND, wind_avg != 0x1ff,   DATA_FORMAT, "%.1f m/s", DATA_DOUBLE, wind_avg * 0.1f,
            "wind_max_m_s",     "Gust speed",       DATA_COND, wind_max != 0x1ff,   DATA_FORMAT, "%.1f m/s", DATA_DOUBLE, wind_max * 0.1f,
            "uvi",              "UVI",              DATA_COND, uv_index != 0xff,    DATA_FORMAT, "%.1f",     DATA_DOUBLE, uv_index * 0.1f,
            "light_lux",        "Light",            DATA_COND, light_raw != 0xffff, DATA_FORMAT, "%.1f lux", DATA_DOUBLE, (double)light_lux,
            "flags",            "Flags",            DATA_FORMAT, "%02x", DATA_INT, flags,
            "unknown",          "Unknown",          DATA_COND, unknown != 0x3fff, DATA_INT, unknown,
            "mic",              "Integrity",        DATA_STRING, "CRC",
            "raw bytes",        "Raw bytes",        DATA_STRING, hexString,       
            NULL);
    /* clang-format on */
```

NorthernMan54, 2023-09-22:
My rule of thumb for the library, is to **not** change the device decoders.  Makes ongoing code base maintenance painful.

But if you want/need the data for the sensor only, your suggestion makes sense.

dirkbeer, 2023-09-22:
Thanks. Was hoping not to change the device decoders, but would then have to write an efficient *encoder* for each sensor. I made a fork with my changes, will think about how I can minimize the maintenance pain.

closed 2023-09-22 by dirkbeer

cgmckeever-orbis, 2024-06-01:
@dirkbeer wondering if you are still doing re-transmit of the signal? Trying to do something similiar are just running into walls. Appreciate it

dirkbeer, 2024-06-02:
@cgmckeever-orbis yes, I am - doing it by modifying the library like I mentioned above. Check out  https://github.com/dirkbeer/Helium-Bridge
It uses a modified version of NorthernMan's library (https://github.com/dirkbeer/rtl_433_ESP) to retransmit the raw byte sensor payload.

cgmckeever-orbis, 2024-06-02:
oooh, lots to look through. Thank you

dirkbeer, 2024-06-02:
lmk if I can point you to something specific, not sure what exactly you are looking for

Sent with [Proton Mail](https://proton.me/) secure email.