NorthernMan54/rtl_433_ESP #107: How add a decoder as in rtl_433.conf file

giacomitti, 2023-10-05:
### Current Situation

What´s the best approach to add a decoder like in rtl_433.conf file? 

Ex: 
decoder n=EV1527-Remote,m=OOK_PWM,s=369,l=1072,g=1400,r=12840,bits>=24,repeats>=3,invert,get=@0:{20}:id,get=@20:{4}:button:[1:D 2:C 3:CD 4:B 5:BD 6:BC 7:BCD 8:A 9:AD 10:AC 11:ACD 12:AB 13:ABD 14:ABC 15:ABCD],unique

### Logs

```markdown
NA.
```

### Configuration

```json
NA.
```

### Environment

- OS:
- Software:
- Node:
- npm:

### Process Supervisor

not applicable

NorthernMan54, 2023-10-06:
Unfortunately the flex decoder is not supported due to resource issues, so you would need to code your own device decoder.

giacomitti, 2023-10-06:
Thank you! Another question, what´s the correct syntax for the build_flags of MY_DEVICES?

- '-DMY_DEVICES=DECL(tpms_truck),DECL(tpms_kia)'

NorthernMan54, 2023-10-06:
MY_DEVICES is a compiler directive boolean in the codebase to show how to configure a subset of device decoders to be included with the code base.

closed 2023-10-07 by giacomitti