NorthernMan54/rtl_433_ESP #110: ESP32 with WH65B

pcfritz, 2023-10-15:
### Current Situation

Hello, thank you first of all for the great work!

I am trying to receive and decode the weather data from the WH65B, unfortunately I am not succeeding in either. I have a cc1101 connected to my ESP32, but unfortunately I can't receive anything. In addition, I have tried a door contact switch, which works. 

Am I making a mistake? Thanks in advance!

### Logs

```markdown
N: ****** setup complete ******
----- CC1101 Status -----
CC1101_MDMCFG1: 0x02
CC1101_MDMCFG2: 0x00
CC1101_MDMCFG3: 0x5b
CC1101_MDMCFG4: 0x69
-------------------------
CC1101_DEVIATN: 0x44
CC1101_AGCCTRL0: 0x91
CC1101_AGCCTRL1: 0x40
CC1101_AGCCTRL2: 0x03
-------------------------
CC1101_IOCFG0: 0x0d
CC1101_IOCFG1: 0x2e
CC1101_IOCFG2: 0x29
-------------------------
CC1101_FIFOTHR: 0x07
CC1101_SYNC0: 0xad
CC1101_SYNC1: 0x12
-------------------------
CC1101_PKTLEN: 0xff
CC1101_PKTCTRL0: 0x32
CC1101_PKTCTRL1: 0x04
-------------------------
CC1101_ADDR: 0x00
CC1101_CHANNR: 0x00
CC1101_FSCTRL0: 0x00
CC1101_FSCTRL1: 0x0f
-------------------------
CC1101_FREQ0: 0x71
CC1101_FREQ1: 0xb0
CC1101_FREQ2: 0x10
-------------------------
CC1101_MCSM0: 0x14
CC1101_MCSM1: 0x30
CC1101_MCSM2: 0x07
-------------------------
CC1101_FOCCFG: 0x76
CC1101_BSCFG: 0x6c
CC1101_WOREVT0: 0x6b
CC1101_WOREVT1: 0x87
CC1101_WORCTRL: 0xf8
CC1101_FREND0: 0x10
CC1101_FREND1: 0x56
-------------------------
CC1101_FSCAL0: 0x0d
CC1101_FSCAL1: 0x13
CC1101_FSCAL2: 0x2c
CC1101_FSCAL3: 0xaa
-------------------------
CC1101_RCCTRL0: 0x00
CC1101_RCCTRL1: 0x41
-------------------------
CC1101_PARTNUM: 0x00
CC1101_VERSION: 0x14
CC1101_MARCSTATE: 0x0d
CC1101_PKTSTATUS: 0xb1
CC1101_RXBYTES: 0x00
----- CC1101 Status -----
rtl_433_ESP(7): Average RSSI Signal -106 dbm, adjusted RSSI Threshold -97, samples 50000
rtl_433_ESP(7): Average RSSI Signal -107 dbm, adjusted RSSI Threshold -98, samples 50000
rtl_433_ESP(7): Average RSSI Signal -107 dbm, adjusted RSSI Threshold -98, samples 50000
rtl_433_ESP(6): Ignored Signal length: 5, Time since last bit length: 1107, Gap length: 162109353, Signal RSSI: -76, Current RSSI: -109, pulses: 0, noise count: 0
```

### Configuration

```json
[env:esp32_cc1101]
card = esp32dev
build_flags = 
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
  '-DONBOARD_LED=2' ; My ESP32 board had this wiring
; *** rtl_433_ESP options ***
; '-DRF_MODULE_FREQUENCY=915.00'
  '-DOOK_MODULATION=false' ; False is FSK, True is OOK
  '-DRTL_DEBUG=4' ; rtl_433 detailed mode
; '-DRTL_VERBOSE=58' ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth Sensor
; '-DRAW_SIGNAL_DEBUG=true' ; Display of received raw data
'-DMEMORY_DEBUG=true' ; Display of information on memory usage
  '-DDEMOD_DEBUG=true' ; Display signal debug information
'-DMY_DEVICES=true' ; Subset of devices
'-DPUBLISH_UNPARSED=true' ; Publishing unparsed signal details
  '-DMINRSSI=-82'
'-DRSSI_THRESHOLD=12' ; apply a delta of 12 to the average RSSI level
'-DAVERAGE_RSSI=5000' ; Display RSSI floor (average of 5000 samples)
; '-DSIGNAL_RSSI=true' ; Display on signal reception
; '-DOOK_MODULATION=false' ; False is FSK, True is OOK
*** RF module options ***
  '-DRF_CC1101="CC1101"' ; CC1101 transceiver module
'-DRF_MODULE_CS=5' ; Pin is used as chip select
  '-DRF_MODULE_GDO0=13' ; CC1101 pin GDO0 - breadboard is soldered on 22 versus 13
  '-DRF_MODULE_GDO2=4' ; CC1101 pin GDO2
; '-DRF_MODULE_INIT_STATUS=true' ; Show transceiver configuration during start-up
; *** RadioLib options ***
; '-DRADIOLIB_DEBUG=true' ; '-DRADIOLIB_DEBUG
; '-DRADIOLIB_VERBOSE=true' ; '-DRADIOLIB_DEBUG=true
  *** FSK setting test ***
  ; '-DsetBitrate'
  ;'-DsetFreqDev'
  ;'-DsetRxBW'
```

### Environment

MacOS

### Process Supervisor

not applicable

NorthernMan54, 2023-10-15:
Did you see this in the closed issues - https://github.com/NorthernMan54/rtl_433_ESP/issues/102

pcfritz, 2023-10-16:
Thank you, I have. However, I had understood that the fix only applies to SX1278, I use the CC1101 chip.

NorthernMan54, 2023-10-16:
It applies to both

DigiH, 2024-04-05:
Hi @pcfritz 

Not sure if you have already solved your issue, and this can be closed, but be aware that if your WH65B is a 868MHZ version you need to set the frequency to 868**.35**MHZ, e. g.
```
'-DOOK_MODULATION=false' ; FSK modulation activated
'-DRF_FREQUENCY=868.35'
```

pcfritz, 2024-06-17:
Hi, @NorthernMan54, a long time has now passed, and the project has briefly faded into the background. However, the problem as such still exists: I am currently testing with a Bresser 5in1 station, which works wonderfully with the Lilygo on 868.30: 

```
----- SX127x Status -----
RegOpMode: 0x0c
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x07
RegBitrateLsb: 0x40
RegRxBw: 0x12
RegAfcBw: 0x02
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----------- FSK --------------
FDEV_MSB: 0x02
FDEV_LSB: 0x8f
----- SX127x Status -----
N: Received message : {"model":"Bresser-6in1","id":42370050,"channel":0,"sensor_type":1,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":338,"rain_mm":28.8,"startup":1,"flags":1,"mic":"CRC","protocol":"Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A","rssi":-63,"duration":32996}
```

Now I wanted to try the same with CC1101 868 mhz receiver on an ESP32. I use this config: 
```
[env:esp32_cc1101]
board = esp32dev
build_flags = 
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
  '-DONBOARD_LED=13'          ; My ESP32 board had this wiring
; *** rtl_433_ESP Options ***
;  '-DRF_MODULE_FREQUENCY=915.00'
;  '-DOOK_MODULATION=false'       ; False is FSK, True is OOK
  '-DRTL_DEBUG=4'           ; rtl_433 verbose mode
;  '-DRTL_VERBOSE=58'          ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
;  '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages
;  '-DMEMORY_DEBUG=true'   ; display memory usage information
  '-DDEMOD_DEBUG=true'  ; display signal debug info
;	'-DMY_DEVICES=true'		; subset of devices
;  '-DPUBLISH_UNPARSED=true'   ; publish unparsed signal details
  '-DMINRSSI=-82'
;  '-DRSSI_THRESHOLD=12'         ; Apply a delta of 12 to average RSSI level
;  '-DAVERAGE_RSSI=5000'     ; Display RSSI floor ( Average of 5000 samples )
;  '-DSIGNAL_RSSI=true'             ; Display during signal receive
;  '-DOOK_MODULATION=false'       ; False is FSK, True is OOK
; *** RF Module Options ***
  '-DRF_CC1101="CC1101"'  ; CC1101 Transceiver Module
;  '-DRF_MODULE_CS=5'      ; pin to be used as chip select
  '-DRF_MODULE_GDO0=22'   ; CC1101 pin GDO0 - Breadboard is 22 versus 13 on soldered
  '-DRF_MODULE_GDO2=4'    ; CC1101 pin GDO2
; '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
; *** radio settings ***
  '-DOOK_MODULATION=false'
  '-DCC1101_FREQUENCY=868.30'
; *** RadioLib Options ***
;  '-DRADIOLIB_DEBUG=true'
;  '-DRADIOLIB_VERBOSE=true'
  ; *** FSK Setting Testing ***
  ;'-DsetBitrate'
  ;'-DsetFreqDev'
  ;'-DsetRxBW'
monitor_port = /dev/cu.usbserial-537E0451031
monitor_speed = 921600
upload_port = /dev/cu.usbserial-537E0451031
upload_speed = 460800
```

Unfortunately, I only get the result: 
```
pulse_slicer_pcm: Bresser Weather Center 5-in-1 codes [{93}ff69c91484100400005230e8]
bresser_6in1_decode: bit_per_row 93 out of range
pulse_slicer_pcm: Bresser Weather Center 6-in-1, 7-in-1 indoor, soil, new 5-in-1, 3-in-1 wind gauge, Froggit WH6000, Ventus C8488A codes [{93}ff69c91484100400005230e8]
bresser_7in1_decode: to few bits (93)
```
Have I made a mistake somewhere?

closed 2024-11-30 by NorthernMan54