NorthernMan54/rtl_433_ESP #111: Fine Offset Electronics WS90 weather station support

Oleg-Sob, 2023-10-24:
### Current Situation

Is it possible to create support for Fine Offset Electronics WS90 weather station?

### Proposed Change

Is it possible to create support for Fine Offset Electronics WS90 weather station?

NorthernMan54, 2023-11-16:
@Oleg-Sob About every 6 to 12 months I do refresh of the device decoders, and this will be included then. You can close this here, and track it via this issue - https://github.com/NorthernMan54/rtl_433_ESP/issues/111

balaboooot, 2024-05-22:
Please update😭

ruimarinho, 2024-09-05:
Would love to see support for WS90 too!

claudobahn, 2024-11-24:
@Oleg-Sob can you re-open this issue? I think @NorthernMan54 intended for this issue to remain open to track progress, his comment above links to this very same issue. The issue he asked you to close was actually the one over in the upstream rtl_433 repo.

NorthernMan54, 2024-11-30:
I refreshed the device decoders from rtl_433 24.10, maybe this is resolved?

Oleg-Sob, 2024-11-30:
There is no way to check yet.

Oleg-Sob, 2025-02-12:
> Я обновил декодеры устройств из rtl_433 24.10, может быть это решено?

I still don't see the WS-90 device.

Oleg-Sob, 2025-02-12:
> I refreshed the device decoders from rtl_433 24.10, maybe this is resolved?

I still don't see the WS-90 device.

DigiH, 2025-02-12:
@Oleg-Sob and all - howe are you using the latest rtl_433_ESP, and which frequency are your WS90s?

Oleg-Sob, 2025-02-12:
I use WS-90 at 433 MHz

DigiH, 2025-02-12:
You need to test with the development version, only which currently includes the new decoders rtl_433 24.10, mentioned by @NorthernMan54 above, as OpenMQTTGateway 1.8.1 does not actually have the WS90 decoder included yet.

Make sure to instal the `lilygo-rtl_433-fsk` binary on the development web upload page.

https://docs.openmqttgateway.com/dev/upload/web-install.html

Oleg-Sob, 2025-02-12:
Installed lilygo-rtl_433-fsk, Fineoffset-WS90 appeared, but does not see other sensors. What did I do wrong?

DigiH, 2025-02-12:
> What did I do wrong?

Nothing, the WS90 is an FSK encoded sensor, that is why you had to install the `lilygo-rtl_433-fsk` binary, while your other sensors, WT-0124 Pool and Nexus-TH sensors, are OOK encoded sensors, which require the OOK build `lilygo-rtl_433` binary.

I also have two LilyGos for my OOK and FSK sensors.

If this is even possible as programatically switchable sometime in the future I really don't know, as also mentioned here

https://github.com/NorthernMan54/rtl_433_ESP/issues/167

Oleg-Sob, 2025-02-12:
I see, thanks for the information.